openai/mujoco-py #170: Error while Installing

rajathkmp, 2017-11-15:
what is the gcc version that is required?
   
```
 /tmp/pip-build-u92nshqf/mujoco-py/mujoco_py/gl/eglshim.c: In function 'closeOpenGL':
    /tmp/pip-build-u92nshqf/mujoco-py/mujoco_py/gl/eglshim.c:159:5: error: 'for' loop initial declarations are only allowed in C99 mode
         for (int device_id=0; device_id<MAX_DEVICES; device_id++) {
         ^
    /tmp/pip-build-u92nshqf/mujoco-py/mujoco_py/gl/eglshim.c:159:5: note: use option -std=c99 or -std=gnu99 to compile your code
    error: command 'gcc' failed with exit status 1
```

Kaixhin, 2017-11-26:
I've had the same issue by trying to install this from source on the current commit (`36f3011` at the moment), but by checking out the commit from the last release (`git checkout 1.50.1.0`) and then installing (`python setup.py install`) I didn't encounter this issue. So whatever created this error has been introduced since then (at least for me).

liyaxi1996, 2018-03-18:
I meet the same problem. Did you figure it our ?

iSaran, 2018-05-03:
I had the same issue in Ubuntu 14.04, virtual environment, and gcc version 4.8.4. It was solved by checking out the tag 1.50.1.0, as mentioned before.

arpit15, 2018-05-10:
The error is caused by using the an old syntax in for loop as mentioned in the error. I was able to get around by 2 options. 
* Make the change in eglshim.c:159 as done here. 
* Add a compiler flag `-std=c99` in builder.py to allow the old syntax as done.
 I have suggested the 1st change in this pull request as it won't conflict with other compiler flags.