openai/mujoco-py #172: GlfwError: Failed to initialize GLFW

aravind0706, 2017-11-20:
Hi, 

I am trying to use the MjViewer on a remote instance. Get the following traceback on initializing the MjViewer. (This is even if I use the -X server while doing the ssh). Any pointers on how to use headless rendering with MjViewer?

`GLFW error (code %d): %s 65544 b'X11: The DISPLAY environment variable is missing'
GLFW error (code %d): %s 65544 b'X11: The DISPLAY environment variable is missing'
---------------------------------------------------------------------------
GlfwError                                 Traceback (most recent call last)
<ipython-input-5-38372fa41077> in <module>()
----> 1 viewer = MjViewer(sim)

/usr/local/lib/python3.5/dist-packages/mujoco_py/mjviewer.py in __init__(self, sim)
    131 
    132     def __init__(self, sim):
--> 133         super().__init__(sim)
    134 
    135         self._ncam = sim.model.ncam

/usr/local/lib/python3.5/dist-packages/mujoco_py/mjviewer.py in __init__(self, sim)
     24 
     25     def __init__(self, sim):
---> 26         super().__init__(sim)
     27 
     28         self._gui_lock = Lock()

/usr/local/lib/python3.5/dist-packages/mujoco_py/mjrendercontext.pyx in mujoco_py.cymj.MjRenderContextWindow.__init__()

/usr/local/lib/python3.5/dist-packages/mujoco_py/mjrendercontext.pyx in mujoco_py.cymj.MjRenderContext.__init__()

/usr/local/lib/python3.5/dist-packages/mujoco_py/mjrendercontext.pyx in mujoco_py.cymj.MjRenderContext._setup_opengl_context()

/usr/local/lib/python3.5/dist-packages/mujoco_py/opengl_context.pyx in mujoco_py.cymj.GlfwContext.__init__()

/usr/local/lib/python3.5/dist-packages/mujoco_py/opengl_context.pyx in mujoco_py.cymj.GlfwContext._init_glfw()

GlfwError: Failed to initialize GLFW`

aravind0706, 2017-11-20:
On using the Dockerfile with the Xdummy entry point, I get this:
`
viewer` = MjViewer(sim)
GLFW error (code %d): %s 65544 b'X11: RandR gamma ramp support seems broken'
GLFW error (code %d): %s 65544 b'Linux: Failed to watch for joystick connections in /dev/input: No such file or directory'
GLFW error (code %d): %s 65544 b'Linux: Failed to open joystick device directory /dev/input: No such file or directory'
libEGL warning: DRI2: failed to authenticate
Creating window glfw
`
Guess it does create the glfw window now... Is this the right way?

closed 2018-01-09 by aravind0706

HuiwenZhang, 2018-02-09:
@aravind0706 Hi, did you fix this problem?

BTickell, 2018-02-15:
@aravind0706 Also having this issue.

kirk86, 2018-03-14:
Same issue here

BTickell, 2018-03-16:
I fixed this by using the latest bindings. You can call sim.render() with desired image size and mujoco_py will initialize a an offscreen render context. sim.render() will return image data as an np array that can be converted to an img with the likes of PIL. Doesn't fix the animation issue directly however.

kirk86, 2018-03-16:
@BTickell would mind describing the process in a bit more detail. What did you actually do differently if you could possibly outline the steps that would be great.

BTickell, 2018-03-16:
@kirk86 Instead of creating a new viewer with MjViewer(sim), call sim.render(width, height) which will return an np array of image data. Subsequent calls to render should use the context which you can find in sim.render_contexts I believe.

kirk86, 2018-03-16:
@BTickell thank you for your answers. I believe that a minimal working example would be ideal and helpful to a lot of people landing in this issue. For instance let's take the `mujoco_py/examples/body_interaction.py` as a concrete example.

In the last couple of lines that file looks like:
```
 model = load_model_from_xml(MODEL_XML)
 sim = MjSim(model)
 viewer = MjViewer(sim)
```

I've changed `viewer = MjViewer(sim)` as per your suggestion `viewer = sim.render(600, 600)` but I still get an error. Additionally I added in the end `sim.render_contexts(viewer)` but still I get an error. What am I doing wrong here?

Thanks!

BTickell, 2018-03-16:
Ok sim.render(...) creates an offscreen render_context but returns the image data. So if you want to get images do
```
model = loda_model_from_xml(MODEL_XML)
sim = MjSim(model)
img = sim.render(600, 600)
```
img will have a [600, 600] np array with image data that you can now use. There will be an offscreen render context inside of sim.render_contexts but this is an array and you shouldn't need this to grab images. Just make additional calls to render(...). So for a basic example, once you have sim and model set up.
```
imgs = []
for i in range(20):
    sim.step()
    imgs.append(sim.render(600, 600))
```
Should give you image data for those 20 steps.

wookayin, 2020-07-15:
This is a bug of glfw3.2. Try installing glfw 3.3+.

dhruvramani, 2020-08-26:
Run 
```
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so 
xvfb-run -a -s "-screen 0 1400x900x24" bash
```
Or just paste them in your `.bashsrc`.

a-z-e-r-i-l-a, 2021-01-11:
@wookayin how do you suggest installing glfw 3.3+ ? especially in a Linux server

a-z-e-r-i-l-a, 2021-01-11:
> @btickell thank you for your answers. I believe that a minimal working example would be ideal and helpful to a lot of people landing in this issue. For instance let's take the [...]

for me it says 

`ERROR: GLEW initalization error: Missing GL version`

btickell, 2021-01-13:
Everyone who begins to use the python bindings for mjp runs into this exact issue. 😄 It should almost be in the installation notes at this point.

PluteW, 2023-02-15:
> This is a bug of glfw3.2. Try installing glfw 3.3+.

This helps me a lot:
We can follow this：
download file here: https://www.glfw.org/
unzip it where you want, cd in this file and make a new dir "build"
cd build
install this dependencies:
`sudo apt-get install libxrandr-dev
sudo apt-get install libxinerama-dev
sudo apt-gey install libxcursor-dev 
sudo apt-gey install libxi-dev
sudo apt install doxygen`

insatll this:
sudo cmake ../
sudo make 
sudo make install

I have ubuntu 18 and mujoco200 with mujoco_py 2.0.2.13, that helps

VikasChidananda, 2025-07-25:
> Everyone who begins to use the python bindings for mjp runs into this exact issue. 😄 It should almost be in the installation notes at this point.

I agree!