openai/mujoco-py #174: Error: "Process finished with exit code 132 (interrupted by signal 4: SIGILL)" OR "Illegal instruction (core dumped)"

Superchicken1, 2017-11-25:
Hi
Like many people I am trying to install mujoco_py 1.5 with mujoco 151. I had mujoco_py 0.5.7 with mujoco 1.31 running but now i need the new camera features.
I do have the key and mujoco 151 in ~/.mujoco. I am using a conda env "py35" with python version 3.5.4.

I am trying it like [here](https://github.com/openai/mujoco-py/issues/47). #47 

This is what I did:
```
$ git clone https://github.com/openai/mujoco-py.git
$ cd mujoco-py/
$ python setup.py install
```
```
running install
running bdist_egg
running egg_info
creating mujoco_py.egg-info
writing dependency_links to mujoco_py.egg-info/dependency_links.txt
writing top-level names to mujoco_py.egg-info/top_level.txt
writing requirements to mujoco_py.egg-info/requires.txt
writing mujoco_py.egg-info/PKG-INFO
writing manifest file 'mujoco_py.egg-info/SOURCES.txt'
reading manifest file 'mujoco_py.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'mujoco_py.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib
creating build/lib/mujoco_py
copying mujoco_py/modder.py -> build/lib/mujoco_py
copying mujoco_py/version.py -> build/lib/mujoco_py
copying mujoco_py/mjrenderpool.py -> build/lib/mujoco_py
copying mujoco_py/__init__.py -> build/lib/mujoco_py
copying mujoco_py/builder.py -> build/lib/mujoco_py
copying mujoco_py/mjviewer.py -> build/lib/mujoco_py
copying mujoco_py/utils.py -> build/lib/mujoco_py
creating build/lib/mujoco_py/generated
copying mujoco_py/generated/__init__.py -> build/lib/mujoco_py/generated
copying mujoco_py/generated/const.py -> build/lib/mujoco_py/generated
creating build/lib/mujoco_py/gl
copying mujoco_py/gl/__init__.py -> build/lib/mujoco_py/gl
creating build/lib/mujoco_py/tests
copying mujoco_py/tests/test_builder.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/test_render_pool.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/test_substep.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/test_vfs.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/test_gen_wrappers.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/__init__.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/test_viewer.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/test_modder.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/test_examples.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/test_cymj.py -> build/lib/mujoco_py/tests
copying mujoco_py/tests/utils.py -> build/lib/mujoco_py/tests
creating build/lib/mujoco_py/pxd
copying mujoco_py/pxd/__init__.py -> build/lib/mujoco_py/pxd
copying mujoco_py/cymj.pyx -> build/lib/mujoco_py
copying mujoco_py/mjbatchrenderer.pyx -> build/lib/mujoco_py
copying mujoco_py/mjrendercontext.pyx -> build/lib/mujoco_py
copying mujoco_py/mjsim.pyx -> build/lib/mujoco_py
copying mujoco_py/mjsimpool.pyx -> build/lib/mujoco_py
copying mujoco_py/mjsimstate.pyx -> build/lib/mujoco_py
copying mujoco_py/opengl_context.pyx -> build/lib/mujoco_py
copying mujoco_py/generated/wrappers.pxi -> build/lib/mujoco_py/generated
copying mujoco_py/gl/dummyshim.c -> build/lib/mujoco_py/gl
copying mujoco_py/gl/egl.h -> build/lib/mujoco_py/gl
copying mujoco_py/gl/eglext.h -> build/lib/mujoco_py/gl
copying mujoco_py/gl/eglplatform.h -> build/lib/mujoco_py/gl
copying mujoco_py/gl/eglshim.c -> build/lib/mujoco_py/gl
[... 159 lines cut ...]
[1/1] Cythonizing /home/chris/python3_ws/mujoco-py/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/chris
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/chris/python3_ws
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/chris/python3_ws/mujoco-py
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/chris/python3_ws/mujoco-py/mujoco_py
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/chris/python3_ws/mujoco-py/mujoco_py/gl
gcc -pthread -B /home/chris/anaconda3/envs/py35/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/chris/python3_ws/mujoco-py/mujoco_py -I/home/chris/.mujoco/mjpro150/include -I/home/chris/anaconda3/envs/py35/lib/python3.5/site-packages/numpy/core/include -I/home/chris/anaconda3/envs/py35/include/python3.5m -c /home/chris/python3_ws/mujoco-py/mujoco_py/cymj.c -o /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/chris/python3_ws/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/chris/anaconda3/envs/py35/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/chris/python3_ws/mujoco-py/mujoco_py -I/home/chris/.mujoco/mjpro150/include -I/home/chris/anaconda3/envs/py35/lib/python3.5/site-packages/numpy/core/include -I/home/chris/anaconda3/envs/py35/include/python3.5m -c /home/chris/python3_ws/mujoco-py/mujoco_py/gl/osmesashim.c -o /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/chris/python3_ws/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/lib.linux-x86_64-3.5
creating /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py
gcc -pthread -shared -L/home/chris/anaconda3/envs/py35/lib -B /home/chris/anaconda3/envs/py35/compiler_compat -Wl,-rpath=/home/chris/anaconda3/envs/py35/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/chris/python3_ws/mujoco-py/mujoco_py/cymj.o /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/chris/python3_ws/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/chris/.mujoco/mjpro150/bin -L/home/chris/anaconda3/envs/py35/lib -Wl,--enable-new-dtags,-R/home/chris/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -lpython3.5m -o /home/chris/python3_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
>>>
>>> model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +"/xmls/claw.xml")
Illegal instruction (core dumped)
```

I am getting a `Illegal instruction (core dumped)`-error.
If I try to load the model in PyCharm I get an `Process finished with exit code 132 (interrupted by signal 4: SIGILL)`-error.

Thx for the help in advance

jmichaux, 2018-01-09:
@Superchicken1 Were you able to solve this problem?

Superchicken1, 2018-02-04:
@jmichaux no I was not able to solve it.
I after some weeks i am now running into it again.
I really want to use the new update of gym v0.9.6 with mujoco_py v1.50.1.0.

Superchicken1, 2018-02-06:
I probably found the problem.
In [link](https://github.com/openai/mujoco-py/issues/47#issuecomment-339561609) 

> The illegal instruction might mean your CPU is missing AVX instructions. Unfortunately AVX is required for mujoco 1.50, though it's only likely to be on processors from 2010 or later.

I have a i5-760 CPU with Nehalem (Lynnfield) from Q3 2010.
Looks like AVX is available in SandyBridge, IvyBridge. So my CPU does not have AVX.

Can someone confirm that this is the problem?

BlGene, 2018-06-13:
@Superchicken1 missing AVX was also the problem for me.

It would be nice if this would be checked at some point instead `Illegal instruction (core dumped)` is a bit ambiguous.

closed 2018-06-29 by Superchicken1

amini2nt, 2020-07-03:
Did anyone solve this? 

How do you fix the AVX problem?

himakarbavikaty, 2020-08-14:
Try by changing python interpreter. It solved for me