openai/mujoco-py #176: Failed building wheel for mujoco-py on macOS High Sierra due to wrong architecture

ferreirafabio, 2017-12-13:
Hi,

when running

`LD_LIBRARY_PATH=$HOME/.mujoco/mjpro150/bin pip3 install mujoco-py`

I receive the following erros:

```
mujoco-py/mujoco_py/cymj.o -fopenmp -w
  FATAL:/opt/local/bin/../libexec/as/i386/as: I don't understand 'm' flag!
  error: command '/usr/local/bin/gcc-7' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for mujoco-py
```

I will post the full log below. Despite the error, I can execute ./simulate and load models. When I want to run code, I receive the following error:

```
ImportError: dlopen(/Users/user/virtualenvs/mujoco/lib/python3.5/site-packages/mujoco_py-1.50.1.34-py3.5.egg/mujoco_py/generated/cymj_macextensionbuilder.so, 2): no suitable image found.  Did find:
	/Users/user/virtualenvs/mujoco/lib/python3.5/site-packages/mujoco_py-1.50.1.34-py3.5.egg/mujoco_py/generated/cymj_macextensionbuilder.so: mach-o, but wrong architecture
	/Users/user/virtualenvs/mujoco/lib/python3.5/site-packages/mujoco_py-1.50.1.34-py3.5.egg/mujoco_py/generated/cymj_macextensionbuilder.so: mach-o, but wrong architecture
```
`uname -m` shows me the architecture x86_64 but `lipo info` on the .so file above shows me I'm set up for i386 although I have never specified it.

I'm running a virtual environment with Python 3.5.2 and site-packages.

ferreirafabio, 2017-12-13:
```
(mujoco) scc-wkit-clxa-4-232:planar_robot_experiment fabioferreira$ pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
Collecting mujoco-py<1.50.2,>=1.50.1
  Using cached mujoco-py-1.50.1.34.tar.gz
Requirement already up-to-date: glfw>=1.4.0 in /Users/fabioferreira/virtualenvs/mujoco/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: numpy>=1.11 in /Users/fabioferreira/virtualenvs/mujoco/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: Cython>=0.27.2 in /Users/fabioferreira/virtualenvs/mujoco/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: imageio>=2.1.2 in /Users/fabioferreira/virtualenvs/mujoco/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: cffi>=1.10 in /Users/fabioferreira/virtualenvs/mujoco/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: pillow in /Users/fabioferreira/virtualenvs/mujoco/lib/python3.5/site-packages (from imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: pycparser in /Users/fabioferreira/virtualenvs/mujoco/lib/python3.5/site-packages (from cffi>=1.10->mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: olefile in /Users/fabioferreira/virtualenvs/mujoco/lib/python3.5/site-packages (from pillow->imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /Users/fabioferreira/virtualenvs/mujoco/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/tmphgmfmr3ipip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/q5
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py
  creating /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/gl
  /usr/local/bin/gcc-7 -fno-strict-aliasing -Wsign-compare -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -g -arch i386 -DONMAC -Imujoco_py -I/private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py -I/Users/fabioferreira/.mujoco/mjpro150/include -I/Users/fabioferreira/virtualenvs/mujoco/lib/python3.5/site-packages/numpy/core/include -I/Library/Frameworks/Python.framework/Versions/3.5/include/python3.5m -c /private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/q5/xcsnx_ [... cut ...]
    error: command '/usr/local/bin/gcc-7' failed with exit status 1
    
    ----------------------------------------
Command "/Users/fabioferreira/virtualenvs/mujoco/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-build-c9xpzg3m/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/q5/xcsnx_45509cq7057vljrmkc0000gn/T/pip-rr1iregy-record/install-record.txt 
```

ferreirafabio, 2017-12-17:
Put differently: how can I ensure the gcc understands the 'm' flag?

`FATAL:/opt/local/bin/../libexec/as/x86_64/as: I don't understand 'm' flag!`

scarlettDiMa, 2018-03-09:
Hi, sorry to interrupt. I have exactly same error as you do: Mac os system, gcc 7. Did you solve this issue? Any help is appreciated!

ferreirafabio, 2018-03-09:
no, unfortunately

azmosavi, 2018-04-05:
have similar issue with installing mujoco as well.
For me every things is installed and located under the python virtual environment.
The pip install mujoco-py==0.5.7 works very well. But when is the turn for the
pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
I get the error of:
Exception: Please add path to mujoco library to your .bashrc:
export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/Users/user/.mujoco/mjpro150/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/user.mujoco/mjpro150/bin
While I have in my bashrc the path to the virtual py:
export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/Users/user/.pyenv/versions/3.6.1/envs/py_env_azi/lib/python3.6/site-packages/gym/envs/mujoco/mjpro150/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/user/.pyenv/versions/3.6.1/envs/py_env_azi/lib/python3.6/site-packages/gym/envs/mujoco/mjpro150/bin

So, I dont know why it can not figure it out the location of binary files although I have the correct path under my bashrc!

episodeyang, 2019-09-21:
> Hi, sorry to interrupt. I have exactly same error as you do: Mac os system, gcc 7. Did you solve this issue? Any help is appreciated!

It worked for me after downgrading to `brew install gcc@6`. See below:

Most critically, I made a mistake of including      `/bin` in the `MJPRO_PATH` for `mujoco-py`. See third-line below:

```bash
export LD_LIBRARY_PATH=$HOME/.mujoco/mujoco200/bin:$LD_LIBRARY_PATH
export MUJOCO_PY_MJPRO_PATH=$HOME/.mujoco/mujoco200/
export MUJOCO_PY_MJKEY_PATH=$HOME/.mujoco/mjkey.txt
```

worked. In addition I had to do
```bash
brew uninstall gcc
brew uninstall gcc@8
brew uninstall gcc@9

brew install gcc@6
```

Only the gcc@6 worked with this installation. OSX version is 10.14.6 (18G87)

Note this is also true on Linux. It seems that `mujoco-py` likes the path without the platform post-fix, where as DeepMind `dm_controlsuite` likes it with those post-fixes.