openai/mujoco-py #177: Make use of mjlib.mj_fullM()

MahanFathi, 2017-12-17:
Apparently, as Emo Todorov mentioned [here](http://www.mujoco.org/forum/index.php?threads/joint-space-inertia-matrix-not-symmetric.3332/), the only way around accessing the dense (!sparse) Joint-Space Inertia Matrix is `mj_fullM()`. However, It is not clear what kind of arguments it takes. One can find the following in `mujoco_py/mjlib.py`:
```
#mjlib.mj_fullM.argtypes = [POINTER(MJMODEL), POINTER(c_double), POINTER(c_double)]
#mjlib.mj_fullM.restype = None
```
I have uncommented these lines, so I can access `mj_fullM()` by importing `mujoco_py.mjlib`. 
I don't know what he meant by `buffer` but when I pass `model.buffer` or `model.data.buffer` an argument error is raised: 
```
ctypes.ArgumentError: argument 2: <class 'TypeError'>: expected LP_c_double instance instead of memoryview
```
 So what's a buffer? How do I pass the right type and what's LP_c_double?
Thanks in advance.

MahanFathi, 2017-12-20:
I used `ctypes` to cast my numpy array to the right type. 

```
mjlib.mj_fullM(self.model.ptr, M.ctypes.data_as(ctypes.POINTER(ctypes.c_double)), self.model.data.qM.ctypes.data_as(ctypes.POINTER(ctypes.c_double)))
```

closed 2017-12-20 by MahanFathi

kracon7, 2019-07-14:
Any work around accessing the dense Joint-space Inertia Matrix now? Since I couldn't find mjlib.py in the current version of mujoco-py, I have no idea how to do similar tricks like @MahanFathi mentioned above. Really appreciate it if anyone can offer some help on this issue. Thanks!