openai/mujoco-py #180: Error: sudo pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'

dialogchat, 2017-12-23:
on  ubuntu 16.04,

Error for : `sudo` pip3 install -U `'mujoco-py<1.50.2,>=1.50.1'`

tmp/pip-build-n1scx1m9/mujoco-py/mujoco_py/gl/eglshim.c:4:21: fatal error: GL/glew.h: No such file or directory
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

  Rolling back uninstall of mujoco-py
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-1drv2pah/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-mgp577n3-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-1drv2pah/mujoco-py/

blackvitriol, 2018-01-29:
sudo apt-get install libglew-dev
fixed it for me...

phaniram-sayapaneni, 2018-09-12:
What about in mac? I installed glew but still getting the following error:

 pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
Collecting mujoco-py<1.50.2,>=1.50.1
  Could not fetch URL https://pypi.python.org/simple/mujoco-py/: There was a problem confirming the ssl certificate: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:645) - skipping
  Could not find a version that satisfies the requirement mujoco-py<1.50.2,>=1.50.1 (from versions: )
No matching distribution found for mujoco-py<1.50.2,>=1.50.1

ecada, 2018-09-25:
Hello I'm also getting the “fatal error: GL/glew.h: No such file or directory compilation terminated." error for mujoco 1.5.

I'm using a multi-user shared ubuntu server. Other people are using the same server for their ml projects so I don't want to corrupt the system and I should be extra careful with the sudo commands. Considering that may I use the below command without corrupting the system? 

sudo apt-get -f install libglew-dev

Without the f arguement I'm getting the error below:
libjpeg-turbo8-dev : Depends: libjpeg-turbo8 (= 1.4.2-0ubuntu3.1) but 1.4.2-0ubuntu3 is to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

HACLANG, 2021-11-25:
i cpoy “GL”  documents to   `/root/anaconda3/envs/env1/include`  form `/usr/include/` ， Successfully fixed the problem,

Check if your installation path is correct. I tested it in conda. There may be some problems with this version of conda.