openai/mujoco-py #181: python 2.7 import error, how eventhough ex-path is set

dialogchat, 2017-12-23:
I set the export path in .bashrc as follows,
`export LD_LIBRARY_PATH=~/mujoco/mjpro150/bin`

having the mjkey.txt and .so files both in that directory. ` ./simulate model.xml ` is running without a problem, but when I try to run this code with python test_mu.py it still cannot import the library??

```
from mujoco_py import load_model_from_xml, MjSim, MjViewer
import math
import os
```

Traceback (most recent call last):
  File "test_mu_1.py", line 1, in <module>
    from mujoco_py import load_model_from_xml, MjSim, MjViewer
  File "/usr/local/lib/python2.7/dist-packages/mujoco_py/__init__.py", line 2, in <module>
    init_config()
  File "/usr/local/lib/python2.7/dist-packages/mujoco_py/config.py", line 33, in init_config
    raise error.MujocoDependencyError('To use MuJoCo, you need to either populate ~/.mujoco/mjkey.txt and ~/.mujoco/mjpro131, or set the MUJOCO_PY_MJKEY_PATH and MUJOCO_PY_MJPRO_PATH environment variables appropriately. Follow the instructions on https://github.com/openai/mujoco-py for where to obtain these.')
mujoco_py.error.MujocoDependencyError: To use MuJoCo, you need to either populate ~/.mujoco/mjkey.txt and ~/.mujoco/mjpro131, or set the MUJOCO_PY_MJKEY_PATH and MUJOCO_PY_MJPRO_PATH environment variables appropriately. Follow the instructions on https://github.com/openai/mujoco-py for where to obtain these.

dialogchat, 2017-12-23:
installing mjpro131 with mujoco-py==0.5.7 is okay
https://github.com/openai/mujoco-py/issues/66

ykim104, 2017-12-27:
I have same problem. Have you fixed it?

ImportError: cannot import name 'load_model_from_xml'

ykim104, 2017-12-27:
I honestly think it's because that those files don't exist. I don't know why.

JMarceloCasanova, 2018-03-14:
Any updates on this?

Having the same problem.

strinkaj, 2018-04-25:
I am having the same issue, has anyone fixed this?

dineshresearch, 2018-06-06:
even im facing the same issue..any fix?

kracon7, 2019-02-19:
I had similar issue. 
I'm using Mujoco_py 0.5.7 with python 2.7 on Unbuntu 16.04 and mujoco 150

I added this two lines into .bashrc
```
export LD_LIBRARY_PATH=/home/jc/mjpro150/bin
export PATH="$LD_LIBRARY_PATH:$PATH"
```

mjkey.txt is in both ./bin and the root directory, Anyone have a fix to this? Thanks!
```
jc@jc-XPS-15:~/mjpro150$ python 
Python 2.7.12 (default, Nov 12 2018, 14:36:49) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jc/.local/lib/python2.7/site-packages/mujoco_py/__init__.py", line 2, in <module>
    init_config()
  File "/home/jc/.local/lib/python2.7/site-packages/mujoco_py/config.py", line 33, in init_config
    raise error.MujocoDependencyError('To use MuJoCo, you need to either populate ~/.mujoco/mjkey.txt and ~/.mujoco/mjpro131, or set the MUJOCO_PY_MJKEY_PATH and MUJOCO_PY_MJPRO_PATH environment variables appropriately. Follow the instructions on https://github.com/openai/mujoco-py for where to obtain these.')
mujoco_py.error.MujocoDependencyError: To use MuJoCo, you need to either populate ~/.mujoco/mjkey.txt and ~/.mujoco/mjpro131, or set the MUJOCO_PY_MJKEY_PATH and MUJOCO_PY_MJPRO_PATH environment variables appropriately. Follow the instructions on https://github.com/openai/mujoco-py for where to obtain these.
>>> 
[2]+  Stopped                 python
jc@jc-XPS-15:~/mjpro150$ ls
bin  doc  include  mjkey.txt  model  sample
jc@jc-XPS-15:~/mjpro150$ cd bin/
jc@jc-XPS-15:~/mjpro150/bin$ ls
basic       libglewegl.so     libglfw.so.3         mjkey.txt  test
compile     libglewosmesa.so  libmujoco150nogl.so  record
derivative  libglew.so        libmujoco150.so      simulate

```

HareshMiriyala, 2020-05-20:
installing gym 0.9.1 fixes this issue

```python -m pip install gym==0.9.1```