openai/mujoco-py #182: pip install fails with the following error

yasiemir, 2017-12-24:
``` Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-iabusc5q/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-22z90kii-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-iabusc5q/mujoco-py/```

Shaofanl, 2018-01-10:
Same problem on ubuntu 16.04

nazir-ust, 2018-01-28:
@yasiemir @Shaofanl Experiencing same problem. Did you find a resolve?

yasiemir, 2018-01-28:
Temporary resolve is to use the older version ` mujoco-py  0.5.7` which apparently runs just fine.
`pip install mujoco-py==0.5.7`

cardwing, 2018-02-08:
Right solution. @yasiemir