openai/mujoco-py #187: Failed to initialize OpenGL

richardrl, 2018-01-29:
I want to use raw image observations, so I am trying to call the _read_pixels_as_in_window(self) in mjviewer.py. However, getting this error:

```
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-384/libGL.so python trpo_run_mujoco.py --env 'Reacher-Pixel-v0'
Logging to /tmp/openai-2018-01-28-23-59-19-815938
2018-01-28 23:59:19.816385: I tensorflow/core/platform/cpu_feature_guard.cc:137] Your CPU supports instructions that this TensorFlow binary was not compiled to use: SSE4.1 SSE4.2 AVX AVX2 FMA
Device mapping: no known devices.
2018-01-28 23:59:19.816722: I tensorflow/core/common_runtime/direct_session.cc:297] Device mapping:

[2018-01-28 23:59:19,816] Making new env: Reacher-Pixel-v0
Compiling /home/richard/lib/mujoco-py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/richard/lib/mujoco-py/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/richard
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/richard/lib
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/richard/lib/mujoco-py
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/richard/lib/mujoco-py/mujoco_py
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/richard/lib/mujoco-py/mujoco_py/gl
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/richard/lib/mujoco-py/mujoco_py -I/home/richard/.mujoco/mjpro150/include -I/home/richard/anaconda3/envs/py35/lib/python3.5/site-packages/numpy/core/include -I/home/richard/lib/mujoco-py/mujoco_py/vendor/egl -I/home/richard/anaconda3/envs/py35/include/python3.5m -c /home/richard/lib/mujoco-py/mujoco_py/cymj.c -o /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/richard/lib/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/richard/lib/mujoco-py/mujoco_py -I/home/richard/.mujoco/mjpro150/include -I/home/richard/anaconda3/envs/py35/lib/python3.5/site-packages/numpy/core/include -I/home/richard/lib/mujoco-py/mujoco_py/vendor/egl -I/home/richard/anaconda3/envs/py35/include/python3.5m -c /home/richard/lib/mujoco-py/mujoco_py/gl/eglshim.c -o /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/richard/lib/mujoco-py/mujoco_py/gl/eglshim.o -fopenmp -w
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/lib.linux-x86_64-3.5
creating /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py
gcc -pthread -shared -L/home/richard/anaconda3/envs/py35/lib -Wl,-rpath=/home/richard/anaconda3/envs/py35/lib,--no-as-needed /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/richard/lib/mujoco-py/mujoco_py/cymj.o /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/richard/lib/mujoco-py/mujoco_py/gl/eglshim.o -L/home/richard/.mujoco/mjpro150/bin -L/home/richard/anaconda3/envs/py35/lib -Wl,--enable-new-dtags,-R/home/richard/.mujoco/mjpro150/bin -lmujoco150 -lglewegl -lpython3.5m -o /home/richard/lib/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
get_obs
Creating window glfw
Found 1 GPUs for rendering. Using device 0.
Could not make EGL context current
Traceback (most recent call last):
  File "trpo_run_mujoco.py", line 35, in <module>
    main()
  File "trpo_run_mujoco.py", line 31, in main
    train(args.env, num_timesteps=args.num_timesteps, seed=args.seed)
  File "trpo_run_mujoco.py", line 21, in train
    env = make_mujoco_env(env_id, workerseed)
  File "/home/richard/baselines/baselines/common/cmd_util.py", line 34, in make_mujoco_env
    env = gym.make(env_id)
  File "/home/richard/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/registration.py", line 164, in make
    return registry.make(id)
  File "/home/richard/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/registration.py", line 122, in make
    env = spec.make()
  File "/home/richard/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/registration.py", line 89, in make
    env = cls(**self._kwargs)
  File "/home/richard/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/mujoco/reacher_pixel.py", line 10, in __init__
    mujoco_env.MujocoEnv.__init__(self, 'reacher.xml', 2)
  File "/home/richard/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/mujoco/mujoco_env.py", line 39, in __init__
    observation, _reward, done, _info = self._step(np.zeros(self.model.nu))
  File "/home/richard/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/mujoco/reacher_pixel.py", line 18, in _step
    ob = self._get_obs()
  File "/home/richard/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/mujoco/reacher_pixel.py", line 40, in _get_obs
    print(np.shape(self._get_viewer()._read_pixels_as_in_window()))
  File "/home/richard/lib/mujoco-py/mujoco_py/mjviewer.py", line 218, in _read_pixels_as_in_window
    self.sim.render(resolution[0], resolution[1])
  File "mjsim.pyx", line 149, in mujoco_py.cymj.MjSim.render
  File "mjsim.pyx", line 151, in mujoco_py.cymj.MjSim.render
  File "mjrendercontext.pyx", line 43, in mujoco_py.cymj.MjRenderContext.__init__
  File "mjrendercontext.pyx", line 99, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "opengl_context.pyx", line 128, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL
```

My setup works for training and rendering video of the Mujoco environment on my screen, but this '_read_pixels_as_in_window(self)', which I am using to access raw image data, fails. And actually even this method was working until I tried installing CUDA (which overwrote my nvidia-384 driver with nvidia-390') and installing CUDA broke everything, so now I think I rolled everything back to nvidia-384 properly and yet this read_pixels is still not working.

febert, 2018-02-26:
Are there any updates on this?

adonishong, 2018-03-05:
met same issue here, track into this file: ./mujoco_py/gl/eglshim.c
call eglGetError() with the place this issue happens(after eglMakeCurrent(eglDpy, EGL_NO_SURFACE, EGL_NO_SURFACE, eglCtx)), eglGetError returns 0x3002, EGL_BAD_ACCESS, no idea from this place...

HailinRen, 2018-04-16:
Did anyone solve this problem?

vitchyr, 2018-04-27:
I've found that if I want to call `env.render()` then I need to set `LD_PRELOAD` to `/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-384/libGL.so`.

However, if I want to call `env.sim.render(w, h)`, then I need to **not** `LD_PRELOAD` (e.g. run `unset LD_PRELOAD`).

lutein, 2018-10-31:
@vitchyr Thx a lot!

chenyiwen97, 2019-03-11:
I got the same problem,too.How to fix it？

vitchyr, 2019-03-11:
@chenyiwen97 have you tried the solution posted above?

chenyiwen97, 2019-03-13:
> @chenyiwen97 have you tried the solution posted above?

Yes.I add the 'export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-415/libGL.so' to the bashrc, but it still doesn't work.

chenyiwen97, 2019-03-13:
> @chenyiwen97 have you tried the solution posted above?

Sorry, I misunderstood what  you said. Now it works.Thanks a lot.

beark007, 2019-05-17:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-384
This works for me.

beark007, 2019-05-17:
# when training, the following has to be commented, unless do testing.
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so

kracon7, 2019-05-28:
Is there a solution to this yet? 
I'm experiencing the same issue. 
Basically there is no way to call ```env.render()```  and ```env.sim.render(w, h)``` in the same environment because the first one requires ```export LD_PRELOAD=/usr/lib/...../libGLEW.so``` but the second one requires this line commented.

luizmarao, 2019-06-14:
After lots of trying and lots of reading, with no succes, I tried (for no reason at all) something that worked!
Using the mode as rgb_array in env.sim.render(), I received a warning that mode should be offscreen or window. Offscreen was exactly the mode in that the error was happening... so I changed to window.
At first, the only change was that, immediately after loading the env, a render screen has shown up, without the error from offscreen mode. Then, without closing the render window, I decided to run the command env.sim.render() with offscreen mode on the python shell, and it worked!
I received the image without any error!
So I included in my code a line calling env.sim.render() with mode window before the line with mode offscreen, which gets the camera image.

my code lines (inside my gym environment):
self.sim.render(mode='window', camera_name='first-person', width=16, height=16, depth=False)
img = self.sim.render(mode='offscreen', camera_name='first-person', width=16, height=16, depth=False)

huangjiancong1, 2020-04-23:
The most import is to check the usage inside env.render():
```
viewer =mujoco_py.MjRenderContextOffscreen(sim, dveice_id=)
```
or
```
viewer = mujoco_py.MjViewer(sim)
```
The second one should has a `LD_PRELOAD=` as mentioned in [https://github.com/openai/mujoco-py/issues/187#issuecomment-384905400](url)

madhan001, 2020-07-02:
undsetting LD_PRELOAD does't work for me, is there a workaround that doesn't involve it?

arthur801031, 2021-01-11:
I also had this problem, but it had to do with incorrect Ubuntu version. I had this error message using `env.sim.render` with Ubuntu 20.04, but once I reinstalled my machine with Ubuntu 18.04, the program ran perfectly.

drozzy, 2021-06-08:
> I've found that if I want to call `env.render()` then I need to set `LD_PRELOAD` to `/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-384/libGL.so`.
> 
> However, if I want to call `env.sim.render(w, h)`, then I need to **not** set `LD_PRELOAD` (e.g. run `unset LD_PRELOAD`).

This works, but I refuse to believe it's not magic since it makes no sense to me. I came to have this issue after fixing another one: https://github.com/openai/mujoco-py/issues/408

atlas-jj, 2021-06-17:
For anyone using Ubuntu 20.04, latest NVIDIA driver version, but still struggling with this issue: if you find the above methods all not working (just like me), **update your Nvidia driver to the latest version 460** will solve the issue.

geyang, 2022-01-03:
Here is a full-writing targeting our HPC cluster: https://github.com/geyang/jaynes-starter-kit/07_supercloud_setup

axmav, 2022-05-03:
I don't even have a /usr/lib/nvidia-384 folder. Only nvidia folder without any .so files inside. But actually CUDA is installed and working properly.
Normal rendering in widow is working but in offscreen context not.
it fails on: `data = self.sim.render(width=width, height=height, camera_name=camera_name)`

QUIlToT, 2022-05-12:
> 

Hi! I have the same issue as you, have you solved this problem please?

QUIlToT, 2022-05-12:
> I don't even have a /usr/lib/nvidia-384 folder. Only nvidia folder without any .so files inside. But actually CUDA is installed and working properly. Normal rendering in widow is working but in offscreen context not. it fails on: `data = self.sim.render(width=width, height=height, camera_name=camera_name)` `unset LD_PRELOAD ` helped

Hi! I have the same issue as you, have you solved this problem please?

axmav, 2022-05-12:
@QUIlToT as far as I understand you can not use the gym for rendering in window as well as in background at the same time on Ubuntu. You need to switch between using `LD_PRELOAD` and not.

QUIlToT, 2022-05-12:
> @QUIlToT as far as I understand you can not use the gym for rendering in window as well as in background at the same time on Ubuntu. You need to switch between using `LD_PRELOAD` and not.

Thanks for your reply!

After `unset LD_PRELOAD`, the `RuntimeError: Failed to initialize OpenGL` does disappear, but I get a new error `ERROR: GLEW initialization error: Missing GL version`.  It seems that the solution to this new error is contradictory to `unset LD_PRELOAD`.

Here is my test code

```python
import mujoco_py

MODEL_XML = """
<?xml version="1.0" ?>
<mujoco>
    <worldbody>
    <camera name="cam0" pos="-0.1 0.1 0.0" quat="0.707 0.707 0 0" />
        <body name="box" pos="0 0 0.2">
            <geom size="0.15 0.15 0.15" type="box"/>
            <joint axis="1 0 0" name="box:x" type="slide"/>
            <joint axis="0 1 0" name="box:y" type="slide"/>
        </body>
        <body name="floor" pos="0 0 0.025">
            <geom size="1.0 1.0 0.02" rgba="0 1 0 1" type="box"/>
        </body>
    </worldbody>
</mujoco>
"""
model = mujoco_py.load_model_from_xml(MODEL_XML)
sim = mujoco_py.MjSim(model)
viewer = mujoco_py.MjViewer(sim)
image = sim.render(width='300', height = '300', camera_name='cam0', depth=False)
print(image)

```

I'm new to this, and I'm not sure which line renders another window besides `image = sim.render(width='300', height = '300', camera_name='cam0', depth=False).

Thx! 
`

QUIlToT, 2022-05-12:
> @QUIlToT as far as I understand you can not use the gym for rendering in window as well as in background at the same time on Ubuntu. You need to switch between using `LD_PRELOAD` and not.

Thx! You made my day!

Irismoon, 2022-07-06:
> I don't even have a /usr/lib/nvidia-384 folder. Only nvidia folder without any .so files inside. But actually CUDA is installed and working properly. Normal rendering in widow is working but in [...]

Hi, 
I'm running code below, 
[image]

which throws RunTimeError at `env.render(mode='rgb_array')`  like the figure below:
[image]

Would you like to give any advice?

Thanks,
Mango

geng-haoran, 2024-01-04:
I was rendering on a remote server and met the same problem. I tried to open an VNC server with port 9 and `export DISPLAY=:9` in the terminal, and then the problem was fixed.

albee, 2024-03-30:
@huangjiancong1's answer guided me to the right solution on an ARM M1 Mac running an Ubuntu 20.04 VM. The NVIDIA drivers don't exist on this system (which has an Apple GPU), but making libGLX_mesa available for linking fixed things for me.

`export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libGLEW.so:/usr/lib/aarch64-linux-gnu/libGLX_mesa.so.0`

VincentWanghh, 2025-03-19:
> For rgb array mode, this code works for me:对于 RGB 数组模式，此代码对我有效：
> 
> `self.viewer = mujoco_py.MjRenderContextOffscreen(self.sim, None, device_id=-1) `
> 
> `img = self.viewer.read_pixels(512, 512, depth=False)[::-1,:,:]`
> 
> by default my cuda visible device is 0.默认情况下，我的 CUDA 可见设备是 0。

It works! 
You made my day! Thanks!