openai/mujoco-py #188: Error when making envronment

gungui98, 2018-01-29:
I tried to install mujoco_py 0.5.7 with mojoco 131,and when i run gym.make('Humanoid-v2'),it yielded an error

```

> [2018-01-29 03:06:00,896] Making new env: Humanoid-v2
> Traceback (most recent call last):
>   File "a.py", line 2, in <module>
>     env = gym.make('Humanoid-v2')
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 164, in make
>     return registry.make(id)
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 122, in make
>     env = spec.make()
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 89, in make
>     env = cls(**self._kwargs)
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/envs/mujoco/humanoid.py", line 12, in __init__
>     mujoco_env.MujocoEnv.__init__(self, 'humanoid.xml', 5)
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/envs/mujoco/mujoco_env.py", line 27, in __init__
>     self.model = mujoco_py.load_model_from_path(fullpath)
> AttributeError: module 'mujoco_py' has no attribute 'load_model_from_path'
> Error in atexit._run_exitfuncs:
> Traceback (most recent call last):
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/utils/closer.py", line 67, in close
>     closeable.close()
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/core.py", line 164, in close
>     self.render(close=True)
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/core.py", line 150, in render
>     return self._render(mode=mode, close=close)
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/envs/mujoco/mujoco_env.py", line 104, in _render
>     if self.viewer is not None:
> AttributeError: 'HumanoidEnv' object has no attribute 'viewer'
> Exception ignored in: <bound method Env.__del__ of <gym.envs.mujoco.humanoid.HumanoidEnv object at 0x7fb71587a7f0>>
> Traceback (most recent call last):
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/core.py", line 203, in __del__
>     self.close()
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/core.py", line 164, in close
>     self.render(close=True)
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/core.py", line 150, in render
>     return self._render(mode=mode, close=close)
>   File "/home/gungui/.local/lib/python3.5/site-packages/gym/envs/mujoco/mujoco_env.py", line 104, in _render
>     if self.viewer is not None:
> AttributeError: 'HumanoidEnv' object has no attribute 'viewer'
> 

```

gungui98, 2018-02-01:
the problem can be fixed by installing gym version 0.7.4!

closed 2018-02-01 by gungui98

inspirepassion, 2018-04-17:
pip3 uninstall gym
pip3 install gym==0.7.3

This works for me. Thanks @gungui98

ppalcx, 2022-08-18:
I have correctly installed mujoco in my system but its giving me this error while I run my code 

Traceback (most recent call last):
  File "test_simple.py", line 9, in <module>
    env = SingleLegEnv()
  File "/home/pramod/To train_128/mujoco_single_leg_training/env/single_leg_env.py", line 66, in __init__
    self.action_space = spaces.Box(
TypeError: __init__() got an unexpected keyword argument 'dtype'
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/home/pramod/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/gym/utils/closer.py", line 67, in close
    closeable.close()
  File "/home/pramod/To train_128/mujoco_single_leg_training/env/mujoco_env.py", line 161, in close
    if self.viewer is not None:
AttributeError: 'SingleLegEnv' object has no attribute 'viewer'
Exception ignored in: <function Env.__del__ at 0x7fad0b677550>
Traceback (most recent call last):
  File "/home/pramod/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/gym/core.py", line 252, in __del__
    self.close()
  File "/home/pramod/To train_128/mujoco_single_leg_training/env/mujoco_env.py", line 161, in close
    if self.viewer is not None:
AttributeError: 'SingleLegEnv' object has no attribute 'viewer'