openai/mujoco-py #190: share success way to install mujoco-py on linux

Lu-Yi-Hsun, 2018-02-03:
my computer's os is linux manjaro

`
git clone https://github.com/openai/mujoco-py 
cd mujoco-py pip install -e . --no-cache 
`

tegg89, 2018-02-05:
Thanks for sharing the way. In my case, just `pip ~` without `sudo` works.

DanielTakeshi, 2018-03-03:
@Lu-Yi-Hsun @tegg89 

Did you confirm that you can actually import it in python? I'm getting the errors here: https://github.com/openai/mujoco-py/issues/204

Rowing0914, 2019-04-11:
> my computer's os is linux manjaro
> if i install by` pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'`it not work
> 
> work way
> `git clone https://github.com/openai/mujoco-py `
> `cd mujoco-py `
> `sudo pip install -e . --no-cache`

Thank you for sharing!!
It worked on my machine!

## Env
- Python: 3.6.6
- OS: check below
- Anaconda?: NO!

```shell
me@me:~/Desktop$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial
```