openai/mujoco-py #192: OpenAI gym Humanoid: Error with load_model_from_path

Suman7495, 2018-02-07:
When I try running OpenAI's sample code for the humanoid robot:
```
import gym
env = gym.make('Humanoid-v2')
env.reset()
env.render()
```

I get the error:
```
Traceback (most recent call last):
  File "humanoid.py", line 2, in <module>
    env = gym.make('Humanoid-v2')
  File "/home/suman/gym/gym/envs/registration.py", line 163, in make
    return registry.make(id)
  File "/home/suman/gym/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/suman/gym/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/home/suman/gym/gym/envs/mujoco/humanoid.py", line 12, in __init__
    mujoco_env.MujocoEnv.__init__(self, 'humanoid.xml', 5)
  File "/home/suman/gym/gym/envs/mujoco/mujoco_env.py", line 27, in __init__
    self.model = mujoco_py.load_model_from_path(fullpath)
AttributeError: 'module' object has no attribute 'load_model_from_path'
```

closed 2018-02-14 by Suman7495

yeazes, 2018-02-22:
Why is this closed? Did you find the answer? 
I am having the same issue using https://github.com/openai/multiagent-competition

Suman7495, 2018-02-22:
I worked around this issue by installing OpenAI Gym and Mujoco by making an Conda environment.
```
conda create -n test python=3.5.2
sudo activate test
cd ~/.mujoco
git clone https://github.com/openai/mujoco-py
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:$HOME/.mujoco/mjpro150/bin"
pip install -e ./mujoco-py
cd ~
git clone https://github.com/openai/gym.git
cd gym
pip install -e .
```

The following lines need to be added in .bashrc located in the home folder.
`LD\_LIBRARY\_PATH="\$LD\_LIBRARY\_PATH:\$HOME/.mujoco/mjpro150/bin pip install mujoco-py"`

To use Mujoco-py and OpenAI Gym:
`source activate test`

To finish: `source deactivate`

HuiwenZhang, 2018-03-14:
Is it because the wrong version of mujoco-py?

kirk86, 2018-03-14:
The problem here is not only the fact that `mujoco-py` is old or wrong version how about `python=3.5.2` isn't that considered old these days? Why not support 3.6.x out of the box?

yeazes, 2018-03-14:
It actually did work for me with python 3.6, mujoco 0.5.7 with the mjpro131 package installed and gym 0.9.1

itstartstosnow, 2018-06-18:
I am also faced with this problem! But I used python 3.6.2, mujoco_py 0.5.7 and  installed mjpro131. I used a virtual environment in Anaconda on windows 10. Does anyone have any idea?

simonkm1, 2018-07-14:
where did you gym 0.9.1 ? , i can see only 0.9.5 and 0.9.7 :(

thanks

itstartstosnow, 2018-07-16:
@simonkm1  I think using `pip install gym==0.9.1` works fine. ;-)

lewisKit, 2018-08-01:
I have the same issue, but it works for me to upgrade both gym and mujoco_py to the newest version.

superggb, 2019-10-12:
I have gym 0.9.1 mjpro131 mujoco_py 0.5.7 and python 3.6.8.
But still have the issue (I can use Humanoid-v1)
anybody knows why?