openai/mujoco-py #195: libmujoco150.so: cannot open shared object file

jake-sanvito, 2018-02-09:
I've wasted two days trying to get gym up and running with mujoco. Mujoco works fine. But I can't get mujoco-py to work either with or without gym. I have tried copying the files, linking them, etc so I don't know what is wrong.

Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-z0gyoqe4/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpla7h9igppip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-build-z0gyoqe4/mujoco-py/setup.py", line 51, in <module>
      cmdclass={'build': Build},
    File "/usr/lib/python3.5/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/usr/local/lib/python3.5/dist-packages/wheel/bdist_wheel.py", line 204, in run
      self.run_command('build')
    File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/tmp/pip-build-z0gyoqe4/mujoco-py/setup.py", line 19, in run
      spec.loader.exec_module(builder)
    File "<frozen importlib._bootstrap_external>", line 665, in exec_module
    File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
    File "/tmp/pip-build-z0gyoqe4/mujoco-py/mujoco_py/builder.py", line 22, in <module>
      from mujoco_py.utils import discover_mujoco
    File "/tmp/pip-build-z0gyoqe4/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
      from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
    File "/tmp/pip-build-z0gyoqe4/mujoco-py/mujoco_py/builder.py", line 502, in <module>
      cymj = load_cython_ext(mjpro_path)
    File "/tmp/pip-build-z0gyoqe4/mujoco-py/mujoco_py/builder.py", line 82, in load_cython_ext
      return load_dynamic_ext('cymj', cext_so_path)
    File "/tmp/pip-build-z0gyoqe4/mujoco-py/mujoco_py/builder.py", line 88, in load_dynamic_ext
      return loader.load_module()
  ImportError: libmujoco150.so: cannot open shared object file: No such file or directory

HuiwenZhang, 2018-02-09:
Have you tried to add the path of your DLL file to the LOAD_LIB_PATH env variable?

jake-sanvito, 2018-02-10:
Yes. That worked for the NVIDIA files which were not in
/usr/local/nvidia/...

Initially I had everything in the .mujoco directory. After this didn't
work, I downloaded another copy of mujoco into my home directory and then
put one in the gym directory and finally copied the file directly to the
gym directory. I figure it has to be able to read it somewhere in those
four locations. But no luck. It would have been an ugly solution anyway but
I wanted to get it working and sort out why not later. I don't remember if
I actually tried adding the location to LOAD_LIB_PATH. But I was getting
tired at the end of a long day (Italy time).

This isn't that big a deal, most of the stuff I am trying to do is
with/will be with live video feeds or videos and signals/feedback as i
build my robot. I was trying to use gym as a sandbox, er, well, a gym I
guess but I can always train like Rocky instead of Drago. Since I have to
use video capture and synchronize it with pwm signals anyway, there isn't
much difference in programming except for actively controlling the game, no?

Speaking of that, do you have canned sequences to download?, like a dataset
of video frames sequenced with actions and rewards? It would be interesting
to see how much you could learn just by being a passenger instead of a
driver. Given the right sequences, I think it would train just as well and
would really cut down on the overhead in industrial applications.

Also, since I am up early and seemingly have time, aren't y'all coming out
with a new gym and universe?

Thanks for the response and sorry to have been so verbose,

John

codealphago, 2018-02-13:
I have the same problem.

codealphago, 2018-02-13:
I have the same problem:
"ImportError: libmujoco150.so: cannot open shared object file: No such file or directory"

khoa-ngo, 2018-02-13:
Hi @jake-sanvito and @codealphago,

What is your output for
```
echo $LD_LIBRARY_PATH
```

Also, can you make sure you've used the correct [mujoco binary](https://www.roboti.us/index.html) for you OS?

jake-sanvito, 2018-02-15:
echo $LD_LIBRARY_PATH
/usr/local/cuda-8.0/lib64:

jake-sanvito, 2018-02-15:
I tried cloning and then running this from the base mujoco-py directory:

bash -c "python setup.py install"

Using that approach I made it through the instructions on the mujoco website:
$ pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
$ python3
import mujoco_py
from os.path import dirname
model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +"/xmls/claw.xml")
sim = mujoco_py.MjSim(model)

print(sim.data.qpos)
# [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]

sim.step()
print(sim.data.qpos)
# [  2.09217903e-06  -1.82329050e-12  -1.16711384e-07  -4.69613872e-11
#   -1.43931860e-05   4.73350204e-10  -3.23749942e-05  -1.19854057e-13
#   -2.39251380e-08  -4.46750545e-07   1.78771599e-09  -1.04232280e-08]

MusulmonLolayev, 2023-05-22:
Following this [guideline](https://pytorch.org/rl/reference/generated/knowledge_base/MUJOCO_INSTALLATION.html), I can achieve to run successfully while others don't help