openai/mujoco-py #197: Install error: image not found

backpropper, 2018-02-11:
(Anaconda) Python 3.6 on macOS (10.13.2)
  ImportError: dlopen(/private/var/folders/mw/y_8kthx55m76s7y1fssxt01c0000gn/T/pip-build-mdylqfer/mujoco-py/mujoco_py/generated/cymj_macextensionbuilder.so, 2): Library not loaded: /Users/woj/.mujoco/mjpro150/bin/libmujoco150.dylib
    Referenced from: /private/var/folders/mw/y_8kthx55m76s7y1fssxt01c0000gn/T/pip-build-mdylqfer/mujoco-py/mujoco_py/generated/cymj_macextensionbuilder.so
    Reason: image not found

I don't get why it refers to /Users/woj/ even though my username is not woj

DZ9, 2018-02-11:
I got this problem too. It seems that the developer who updated this repo named woj. So I think there must be something wrong with his last submission.

backpropper, 2018-02-11:
So were you able to get around the problem? 
And yeah my guess was the same, the shared object seems to be built on the developer's machine.

DZ9, 2018-02-12:
Nope. I reviewed some recent commits, but failed to find where's wrong.

eliasdc, 2018-02-12:
The versions from 1.50.1.37 to 1.50.1.39 have this issue.

backpropper, 2018-02-13:
So when is it going to be fixed? Is there any workaround?

JamesOwers, 2018-02-13:
+1 - same issue, can't install