openai/mujoco-py #198: Unable to install mujoco-py

ravijo, 2018-02-11:
Hi,
I have installed [mujoco](http://www.mujoco.org/) and placed the license file as per the documentation. `LD_LIBRARY_PATH` is set to `$HOME/.mujoco/mjpro150/bin` and I am able to run demo such as `./simulate ../model/humanoid.xml`.

Now I am trying to install `mujoco-py`. I am using `conda`  package management system.

Environment details
---------------------------
- Python 3.5.4 |Anaconda, Inc.| (default, Nov 20 2017, 18:44:38) 
- conda 4.4.9
- Ubuntu 14.04.5 LTS
- Linux lab 3.13.0-125-generic #174-Ubuntu SMP Mon Jul 10 18:51:24 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
- Cuda compilation tools, release 7.5, V7.5.17
- 01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM206 [GeForce GTX 960] [10de:1401] (rev a1) (prog-if 00 [VGA controller])

Installation Steps
----------------------
### Attempt 1
1. Activate Python 3.5 Environment i.e., `source activate python35env`
1. `cd ~/.mujoco`
1. `git clone https://github.com/openai/mujoco-py.git`
1. `pip install -r requirements.txt`
1. `pip install -r requirements.dev.txt`
1. `python setup.py install`

_No error were reported till here_

Use mujoco-py
-------------------
```
$ python3
import mujoco_py
```
I tried to import `mujoco_py` as shown above. However, the computer starts compiling it and returned after a while. The compilation was failed and following error was reported at terminal-
```
File "/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/python35env/lib/python3.5/site-packages/mujoco_py-1.50.1.36-py3.5.egg/mujoco_py/builder.py", line 127, in caching_compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
File "/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/python35env/lib/python3.5/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'gcc' failed with exit status 1
```
### Attempt 2
I tried `pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'` but didn't succeed. See the errors below-
```
[11/11] Cythonizing /tmp/easy_install-9xftm166/Cython-0.27.3/Cython/Tempita/_tempita.py
Unable to find pgen, not compiling formal grammar.
warning: no files found matching '2to3-fixers.txt'
warning: no files found matching 'Doc/*'
warning: no files found matching '*.pyx' under directory 'Cython/Debugger/Tests'

distutils.errors.DistutilsError: Setup script exited with error: Command "x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -Ibuild/src.linux-x86_64-3.4/numpy/core/src/npymath -Inumpy/core/include -Ibuild/src.linux-x86_64-3.4/numpy/core/include/numpy -I/usr/include/python3.4m -Ibuild/src.linux-x86_64-3.4/numpy/core/src/private -Ibuild/src.linux-x86_64-3.4/numpy/core/src/npymath -Ibuild/src.linux-x86_64-3.4/numpy/core/src/private -Ibuild/src.linux-x86_64-3.4/numpy/core/src/npymath -Ibuild/src.linux-x86_64-3.4/numpy/core/src/private -Ibuild/src.linux-x86_64-3.4/numpy/core/src/npymath -c numpy/core/src/npymath/npy_math.c -o build/temp.linux-x86_64-3.4/numpy/core/src/npymath/npy_math.o -MMD -MF build/temp.linux-x86_64-3.4/numpy/core/src/npymath/npy_math.o.d" failed with exit status 1
In file included from numpy/core/src/npymath/npy_math.c:9:0:
numpy/core/src/npymath/npy_math_internal.h.src:56:30: fatal error: npy_math_private.h: No such file or directory
 #include "npy_math_private.h"
                              ^
compilation terminated.
In file included from numpy/core/src/npymath/npy_math.c:9:0:
numpy/core/src/npymath/npy_math_internal.h.src:56:30: fatal error: npy_math_private.h: No such file or directory
 #include "npy_math_private.h"
                              ^
compilation terminated.
----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_ravi/mujoco-py
Storing debug log for failure in /home/ravi/.pip/pip.log
```

Additional Information
------------------------------
Since above error is related to numpy, so I verified it. I am using numpy 1.13.3 as shown below-
```
(python35env) ravi@lab:~$ conda list | grep numpy
numpy                     1.13.3           py35h3dfced4_2  
numpydoc                  0.7.0                     <pip>
```
### Attempt 3
I also tried `pip install mujoco-py` but no success. See the errors below-
```
File "/tmp/pip-build-8wxj6ah8/mujoco-py/setup.py", line 19, in run
  spec.loader.exec_module(builder)
File "<frozen importlib._bootstrap_external>", line 697, in exec_module
File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
File "/tmp/pip-build-8wxj6ah8/mujoco-py/mujoco_py/builder.py", line 22, in <module>
  from mujoco_py.utils import discover_mujoco
File "/tmp/pip-build-8wxj6ah8/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
  from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
File "/tmp/pip-build-8wxj6ah8/mujoco-py/mujoco_py/builder.py", line 502, in <module>
  cymj = load_cython_ext(mjpro_path)
File "/tmp/pip-build-8wxj6ah8/mujoco-py/mujoco_py/builder.py", line 82, in load_cython_ext
  return load_dynamic_ext('cymj', cext_so_path)
File "/tmp/pip-build-8wxj6ah8/mujoco-py/mujoco_py/builder.py", line 88, in load_dynamic_ext
  return loader.load_module()
ImportError: /usr/local/nvidia/lib64/libEGL.so.1: cannot open shared object file: No such file or directory

----------------------------------------
Command "/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/python35env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-8wxj6ah8/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-bdwvwc34-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-8wxj6ah8/mujoco-py/
```
Additional Information
------------------------------
This time, it seems that the error is related to nvidia. Unfortunately, there is no such file `/usr/local/nvidia/lib64/libEGL.so.1`

```
ravi@lab:~$ ls -la /usr/local/nvidia/lib64/libEGL.so.1
ls: cannot access /usr/local/nvidia/lib64/libEGL.so.1: No such file or directory
ravi@lab:~$ ls -la /usr/local/nvidia
ls: cannot access /usr/local/nvidia: No such file or directory
```
I am confused!!! What is the problem exactly? Any workaround, please?

eliasdc, 2018-02-12:
I think python 3.5.4 is not supported and you should use 3.5.2

ravijo, 2018-02-12:
@eliasdc: Thank you very much. I created new `conda` environment with `python 3.5.4` and tried installing it again but didn't succeed. Below is the error-
```
    raise DistutilsError("Setup script exited with %s" % (v.args[0],))
distutils.errors.DistutilsError: Setup script exited with error: Command "x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -Ibuild/src.linux-x86_64-3.4/numpy/core/src/npymath -Inumpy/core/include -Ibuild/src.linux-x86_64-3.4/numpy/core/include/numpy -I/usr/include/python3.4m -Ibuild/src.linux-x86_64-3.4/numpy/core/src/private -Ibuild/src.linux-x86_64-3.4/numpy/core/src/npymath -Ibuild/src.linux-x86_64-3.4/numpy/core/src/private -Ibuild/src.linux-x86_64-3.4/numpy/core/src/npymath -Ibuild/src.linux-x86_64-3.4/numpy/core/src/private -Ibuild/src.linux-x86_64-3.4/numpy/core/src/npymath -c build/src.linux-x86_64-3.4/numpy/core/src/npymath/npy_math_complex.c -o build/temp.linux-x86_64-3.4/build/src.linux-x86_64-3.4/numpy/core/src/npymath/npy_math_complex.o -MMD -MF build/temp.linux-x86_64-3.4/build/src.linux-x86_64-3.4/numpy/core/src/npymath/npy_math_complex.o.d" failed with exit status 1
numpy/core/src/npymath/npy_math_complex.c.src:34:29: fatal error: npy_math_common.h: No such file or directory
 #include "npy_math_common.h"
                             ^
compilation terminated.
numpy/core/src/npymath/npy_math_complex.c.src:34:29: fatal error: npy_math_common.h: No such file or directory
 #include "npy_math_common.h"
                             ^
compilation terminated.
----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_ravi/mujoco-py
Storing debug log for failure in /home/ravi/.pip/pip.log
```
Following packages are installed in this environment-
```
(mujocoenv) ravi@lab:~/.mujoco/mujoco-py$ conda list
# packages in environment at /home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/mujocoenv:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.10                    <pip>
attrs                     17.4.0                    <pip>
Babel                     2.5.3                     <pip>
ca-certificates           2017.08.26           h1d4fec5_0  
certifi                   2018.1.18                py35_0  
cffi                      1.11.4                    <pip>
chardet                   3.0.4                     <pip>
Cython                    0.27.3                    <pip>
decorator                 4.2.1                     <pip>
docutils                  0.14                      <pip>
glfw                      1.5.1                     <pip>
idna                      2.6                       <pip>
ImageHash                 4.0                       <pip>
imageio                   2.2.0                     <pip>
imagesize                 0.7.1                     <pip>
intel-openmp              2018.0.0             hc7b2577_8  
ipdb                      0.10.3                    <pip>
ipython                   6.2.1                     <pip>
ipython-genutils          0.2.0                     <pip>
jedi                      0.11.1                    <pip>
Jinja2                    2.10                      <pip>
libgcc-ng                 7.2.0                h7cc24e2_2  
libgfortran-ng            7.2.0                h9f7466a_2  
libstdcxx-ng              7.2.0                h7a57d05_2  
MarkupSafe                1.0                       <pip>
mkl                       2018.0.1             h19d6760_4  
numpy                     1.14.0           py35h3dfced4_1  
numpydoc                  0.7.0                     <pip>
openssl                   1.0.2n               hb7f436b_0  
packaging                 16.8                      <pip>
parso                     0.1.1                     <pip>
pexpect                   4.4.0                     <pip>
pickleshare               0.7.4                     <pip>
Pillow                    5.0.0                     <pip>
pip                       9.0.1            py35h7e7da9d_4  
pluggy                    0.6.0                     <pip>
prompt-toolkit            1.0.15                    <pip>
ptyprocess                0.5.2                     <pip>
py                        1.5.2                     <pip>
pycparser                 2.18                      <pip>
Pygments                  2.2.0                     <pip>
pyparsing                 2.2.0                     <pip>
pytest                    3.4.0                     <pip>
pytest-instafail          0.3.0                     <pip>
python                    3.5.2                         0  
pytz                      2018.3                    <pip>
PyWavelets                0.5.2                     <pip>
readline                  6.2                           2  
requests                  2.18.4                    <pip>
scipy                     1.0.0            py35hcbbe4a2_0  
setuptools                38.4.0                   py35_0  
simplegeneric             0.8.1                     <pip>
six                       1.11.0                    <pip>
snowballstemmer           1.2.1                     <pip>
Sphinx                    1.7.0                     <pip>
sphinx-rtd-theme          0.2.4                     <pip>
sphinxcontrib-websupport  1.0.1                     <pip>
sqlite                    3.13.0                        0  
tk                        8.5.18                        0  
traitlets                 4.3.2                     <pip>
urllib3                   1.22                      <pip>
wcwidth                   0.1.7                     <pip>
wheel                     0.30.0           py35hd3883cf_1  
xz                        5.2.3                h55aa19d_2  
zlib                      1.2.11               ha838bed_2  
```

CrazyHeex, 2018-02-12:
try set your to $LD_LIBRARY_PATH:$HOME/.mujoco/mjpro150/bin

export LD_LIBRARY_PATH=“$LD_LIBRARY_PATH:$HOME/.mujoco/mjpro150/bin“

If you do 
export LD_LIBRARY_PATH=“$HOME/.mujoco/mjpro150/bin@

It will overwrite the previous ld library path, that’s why you can not compile.

ravijo, 2018-02-12:
@CrazyHeex: I set `LD_LIBRARY_PATH` inside `.bashrc` file as following-
```
# mujoco configurations
LD_LIBRARY_PATH=${HOME}/.mujoco/mjpro150/bin:${LD_LIBRARY_PATH}
export LD_LIBRARY_PATH
```

CrazyHeex, 2018-02-12:
I get to my laptop and did following command with no problem:
conda create -n test python=3.5.2
git clone https://github.com/openai/mujoco-py
pip install -e ./mujoco-py
python
>> import mujoco_py

(test) ➜  ~ conda list
# packages in environment at /home/zyang/anaconda3/envs/test:
#
ca-certificates           2017.08.26           h1d4fec5_0  
certifi                   2018.1.18                py35_0  
cffi                      1.11.4                    <pip>
Cython                    0.27.3                    <pip>
glfw                      1.5.1                     <pip>
imageio                   2.2.0                     <pip>
libgcc-ng                 7.2.0                h7cc24e2_2  
numpy                     1.14.0                    <pip>
openssl                   1.0.2n               hb7f436b_0  
Pillow                    5.0.0                     <pip>
pip                       9.0.1            py35h7e7da9d_4  
pycparser                 2.18                      <pip>
python                    3.5.2                         0  
readline                  6.2                           2  
setuptools                38.4.0                   py35_0  
sqlite                    3.13.0                        0  
tk                        8.5.18                        0  
wheel                     0.30.0           py35hd3883cf_1  
xz                        5.2.3                h55aa19d_2  
zlib                      1.2.11               ha838bed_2

ravijo, 2018-02-12:
@CrazyHeex: Still no success. Just copied your steps as following-

1. `conda create -n test python=3.5.2`
1. `source activate test`
1. `cd ~/.mujoco`
1. `git clone https://github.com/openai/mujoco-py`
1. `export LD_LIBRARY_PATH=“$LD_LIBRARY_PATH:$HOME/.mujoco/mjpro150/bin“` // although it is in `.bashrc` file too
1. `pip install -e ./mujoco-py`

See below the output in terminal-
```
(test) ravi@lab:~/.mujoco$ pip install -e ./mujoco-py
Obtaining file:///home/ravi/.mujoco/mujoco-py
Collecting glfw>=1.4.0 (from mujoco-py==1.50.1.36)
Collecting numpy>=1.11 (from mujoco-py==1.50.1.36)
  Downloading numpy-1.14.0-cp35-cp35m-manylinux1_x86_64.whl (17.1MB)
    100% |████████████████████████████████| 17.1MB 111kB/s 
Collecting Cython>=0.27.2 (from mujoco-py==1.50.1.36)
  Using cached Cython-0.27.3-cp35-cp35m-manylinux1_x86_64.whl
Collecting imageio>=2.1.2 (from mujoco-py==1.50.1.36)
Collecting cffi>=1.10 (from mujoco-py==1.50.1.36)
  Using cached cffi-1.11.4-cp35-cp35m-manylinux1_x86_64.whl
Collecting pillow (from imageio>=2.1.2->mujoco-py==1.50.1.36)
  Using cached Pillow-5.0.0-cp35-cp35m-manylinux1_x86_64.whl
Collecting pycparser (from cffi>=1.10->mujoco-py==1.50.1.36)
Installing collected packages: glfw, numpy, Cython, pillow, imageio, pycparser, cffi, mujoco-py
  Running setup.py develop for mujoco-py
Successfully installed Cython-0.27.3 cffi-1.11.4 glfw-1.5.1 imageio-2.2.0 mujoco-py numpy-1.14.0 pillow-5.0.0 pycparser-2.18
```
Now see the errors while trying `import mujoco_py` below-
```
(test) ravi@lab:~/.mujoco$ python
Python 3.5.2 |Continuum Analytics, Inc.| (default, Jul  2 2016, 17:53:06) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Compiling /home/ravi/.mujoco/mujoco-py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/ravi/.mujoco/mujoco-py/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder
creating /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5
creating /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home
creating /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/ravi
creating /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/ravi/.mujoco
creating /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/ravi/.mujoco/mujoco-py
creating /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/ravi/.mujoco/mujoco-py/mujoco_py
creating /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/ravi/.mujoco/mujoco-py/mujoco_py/gl
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco-py/mujoco_py -I/home/ravi/.mujoco/mujoco-py/mujoco_py -I/home/ravi/.mujoco/mjpro150/include -I/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/test/lib/python3.5/site-packages/numpy/core/include -I/home/ravi/.mujoco/mujoco-py/mujoco_py/vendor/egl -I/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/test/include/python3.5m -c /home/ravi/.mujoco/mujoco-py/mujoco_py/cymj.c -o /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/ravi/.mujoco/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco-py/mujoco_py -I/home/ravi/.mujoco/mujoco-py/mujoco_py -I/home/ravi/.mujoco/mjpro150/include -I/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/test/lib/python3.5/site-packages/numpy/core/include -I/home/ravi/.mujoco/mujoco-py/mujoco_py/vendor/egl -I/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/test/include/python3.5m -c /home/ravi/.mujoco/mujoco-py/mujoco_py/gl/eglshim.c -o /home/ravi/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/ravi/.mujoco/mujoco-py/mujoco_py/gl/eglshim.o -fopenmp -w
/home/ravi/.mujoco/mujoco-py/mujoco_py/gl/eglshim.c: In function ‘closeOpenGL’:
/home/ravi/.mujoco/mujoco-py/mujoco_py/gl/eglshim.c:159:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int device_id=0; device_id<MAX_DEVICES; device_id++) {
     ^
/home/ravi/.mujoco/mujoco-py/mujoco_py/gl/eglshim.c:159:5: note: use option -std=c99 or -std=gnu99 to compile your code
Traceback (most recent call last):
  File "/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/test/lib/python3.5/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/test/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/test/lib/python3.5/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/ravi/.pyenv/versions/anaconda3-5.0.1/envs/test/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
[... 36 lines cut ...]
[... cut ...]
pycparser                 2.18                      <pip>
python                    3.5.2                         0  
readline                  6.2                           2  
setuptools                38.4.0                   py35_0  
sqlite                    3.13.0                        0  
tk                        8.5.18                        0  
wheel                     0.30.0           py35hd3883cf_1  
xz                        5.2.3                h55aa19d_2  
zlib                      1.2.11               ha838bed_2  
```

CrazyHeex, 2018-02-12:
try to move the int out of the for loop in the code:/home/ravi/.mujoco/mujoco-py/mujoco_py/gl/eglshim.c:159:5

for (int device_id=0; device_id<MAX_DEVICES; device_id++){

to 

int device_id;
for (;device_id<device_id<MAX_DEVICES; device_id++){

not sure, I never saw this before.

ravijo, 2018-02-13:
```
int device_id;
for (device_id=0; device_id<MAX_DEVICES; device_id++) 
```
It means that the compiler is not using `-std=c++11`. Isn't it suspicious? Also the above code resides inside OpenGL function. **Is the error related to OpenGL?**

The above works fine and I am able to use `import mujoco_py` as shown below-
```
>>> import mujoco_py
>>> from os.path import dirname
>>> model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +"/xmls/claw.xml")
>>> sim = mujoco_py.MjSim(model)
>>> print(sim.data.qpos)
[0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
```
However, when I treid to use `gym`, I got following error `ERROR: GLEW initalization error: Missing GL version`
```
(test) ravi@lab:~/.mujoco/mjpro150/bin$ python3
Python 3.5.2 |Continuum Analytics, Inc.| (default, Jul  2 2016, 17:53:06) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gym
>>> env = gym.make('Hopper-v2')
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
>>> print(env.observation_space)
Box(11,)
>>> print(env.action_space)
Box(3,)
>>> env.reset()
array([ 1.24700988e+00, -4.12321754e-03,  3.67075481e-03, -2.96215107e-03,
       -1.26082879e-03,  5.48675500e-04, -4.46060702e-03,  2.28999331e-03,
       -2.64810316e-03, -2.33473989e-03, -4.64960736e-03])
>>> for _ in range(100):
...     env.render()
...     env.step(env.action_space.sample()) # take a random action    
... 
Creating window glfw
ERROR: GLEW initalization error: Missing GL version
Press Enter to exit ...Killed
```
Whats' wrong now? Quick search is showing me that the error can be because of [libglewosmesa](https://github.com/openai/mujoco-py/issues/44#issuecomment-336313803). Should I remove `libglewosmesa` and try again?

Please note that I am still able to use `./simulate ../model/humanoid.xml` and can see the humanoid simulation but when I close this window, it shows `Segmentation fault`. See Below-
```
(test) ravi@lab:~/.mujoco/mjpro150/bin$ ./simulate ../model/humanoid.xmlMuJoCo Pro library version 1.50
Segmentation fault
```

JamesOwers, 2018-02-13:
Did you sort this? I'm having the same problem.

eliasdc, 2018-02-15:
I was having the same problems and decided to use the older mujoco-py in combination with an older openai gym version.
gym==0.9.3
mujoco-py==0.5.7
mujoco 131 from their website

ravijo, 2018-02-15:
@eliasdc: As per your suggestion, I will give it a try. However, I am worried about the license file. Does the same license file work for mujoco v131 too? I have the license file for mujoco v150.

eliasdc, 2018-02-15:
@ravijo: The same license file can be used.

ravijo, 2018-02-16:
As a workaround suggest by @eliasdc, I found that following combination works fine-

- gym==0.9.3
- mujoco-py==0.5.7
- mujoco 131

Below is the list of packages present in the environment while testing-
```
# Name                    Version                   Build  Channel
ca-certificates           2017.08.26           h1d4fec5_0  
certifi                   2018.1.18                py27_0  
chardet                   3.0.4                     <pip>
future                    0.16.0                    <pip>
gym                       0.9.3                     <pip>
idna                      2.6                       <pip>
libedit                   3.1                  heed3624_0  
libffi                    3.2.1                hd88cf55_4  
libgcc-ng                 7.2.0                h7cc24e2_2  
libstdcxx-ng              7.2.0                h7a57d05_2  
mujoco-py                 0.5.7                     <pip>
ncurses                   6.0                  h9df7e31_2  
numpy                     1.14.0                    <pip>
openssl                   1.0.2n               hb7f436b_0  
pip                       9.0.1            py27ha730c48_4  
pyglet                    1.3.1                     <pip>
PyOpenGL                  3.1.0                     <pip>
python                    2.7.14              h1571d57_29  
readline                  7.0                  ha6073c6_4  
requests                  2.18.4                    <pip>
setuptools                38.4.0                   py27_0  
six                       1.11.0                    <pip>
sqlite                    3.22.0               h1bed415_0  
tk                        8.6.7                hc745277_3  
urllib3                   1.22                      <pip>
wheel                     0.30.0           py27h2bc6bb2_1  
zlib                      1.2.11               ha838bed_2 
```

ahsteven, 2018-03-02:
How do I install gym=0.9.3

If I do pip install gym==0.9.3 it install 0.10.3

ravijo, 2018-03-02:
@ahsteven: strange! Please make sure you are using Python 2.7. I made a conda environment with Python 2.7 and then installed gym 0.9.3

ahsteven, 2018-03-02:
Ok I was able to get this installed with py 2.7 and import mujoco_py. 

One question, how do I set the version when installing from the github? I installed using pip install gym==0.9.3 but I would like to install from the github with 'pip install -e .' or something similar.

My goal in installing this is to be able to run the openai gym baselines. Which I am getting an invalid syntax error right now so I am thinking the gym baselines might work with py 35. 

  File "baselines/common/dataset.py", line 50
    def iterbatches(arrays, *, num_batches=None, batch_size=None, shuffle=True, include_final_partial_batch=True):
                             ^
SyntaxError: invalid syntax

I will try this again with python 352. 

This forum has been very helpful.

eliasdc, 2018-03-05:
@ahsteven: checkout the correct tag or commit and than use pip install -e .

SiddhardhaSaran, 2018-03-09:
So the only workaround is to revert to an older version?. Has any one successfully installed mujoco-py v1.5?

ravijo, 2018-03-09:
@SiddhardhaSaran: I think it is possible to install. However, in my case, I have CUDA installed which is creating issues. You can try on your system without CUDA.

SiddhardhaSaran, 2018-03-09:
running `sudo apt-get install libglew-dev` 
and installing patchelf. I was able to install mujoco v1.5.

lbhdc, 2018-03-09:
@SiddhardhaSaran I am having the same / similar problem as @ravijo. I had previously installed `libglew-dev` but did not have `patchelf`. The installation is successful, but it fails on import.

I found this python wrapper works with my GPU (nvidia) and mjpro150: [`mujoco`](https://github.com/lobachevzky/mujoco)

kirk86, 2018-03-14:
I have both `libglew-dev` and `patchelf` installed in my systems and still during installation I see the missing `GLEW.h` error:

```
/tmp/pip-build-7uw627tn/mujoco-py/mujoco_py/gl/eglshim.c:4:21: fatal error: GL/glew.h: No such file or directory
  compilation terminated.
  error: command '/usr/bin/gcc-5' failed with exit status 1

  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /home/user/miniconda3/envs/tf/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-7uw627tn/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-n5yr5kcp-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'mujoco_py.cymj' extension
    /tmp/pip-build-7uw627tn/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.6/tmp/pip-build-7uw627tn/mujoco-py/mujoco_py/cymj.o up to date, skipping.
    /usr/bin/gcc-5 -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-build-7uw627tn/mujoco-py/mujoco_py -I/home/user/.mujoco/mjpro150/include -I/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/numpy/core/include -I/tmp/pip-build-7uw627tn/mujoco-py/mujoco_py/vendor/egl -I/home/user/miniconda3/envs/tf/include/python3.6m -c /tmp/pip-build-7uw627tn/mujoco-py/mujoco_py/gl/eglshim.c -o /tmp/pip-build-7uw627tn/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.6/tmp/pip-build-7uw627tn/mujoco-py/mujoco_py/gl/eglshim.o -fopenmp -w
    /tmp/pip-build-7uw627tn/mujoco-py/mujoco_py/gl/eglshim.c:4:21: fatal error: GL/glew.h: No such file or directory
    compilation terminated.
    error: command '/usr/bin/gcc-5' failed with exit status 1

    ----------------------------------------
Command "/home/user/miniconda3/envs/tf/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-7uw627tn/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-n5yr5kcp-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-7uw627tn/mujoco-py/
```
Everything that I've seen so far here is just a hack to make `mujoco-py` we need true support of python 3.6.x

ravijo, 2018-03-15:
> Everything that I've seen so far here is just a hack to make mujoco-py work, we need true support of python 3.6.x
@kirk86 : Please make sure you are using Python 3.5.2. Support for Python 3.6 is [planned](https://github.com/openai/mujoco-py/issues/52)

YaqiXie, 2018-03-20:
I'm using python 3.5.2, but still got exactly the same error. (Ubuntu)

ravijo, 2018-03-21:
@YaqiXie: I would suggest to setup following environment:
* python v2.7.14
* gym v 0.9.3
* mujoco v131
* mujoco-py v0.5.7

Check out this for more info https://github.com/openai/mujoco-py/issues/198#issuecomment-366200529

azmosavi, 2018-04-04:
I have similar issue with installing mujoco as well.
For me every things is installed and located under the python virtual environment.
The pip install mujoco-py==0.5.7 works very well. But when is the turn for the 
pip3 install -U 'mujoco-py<1.50.2,>=1.50.1' 
I get the error of:
 Exception: Please add path to mujoco library to your .bashrc:
  export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/Users/user/.mujoco/mjpro150/bin
  export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/user.mujoco/mjpro150/bin
While I have in my bashrc the path to the virtual py:
export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/Users/user/.pyenv/versions/3.6.1/envs/py_env_azi/lib/python3.6/site-packages/gym/envs/mujoco/mjpro150/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/user/.pyenv/versions/3.6.1/envs/py_env_azi/lib/python3.6/site-packages/gym/envs/mujoco/mjpro150/bin

So, I dont know why it can not figure it out the location of binary files although I have the correct path under my bashrc!

ravijo, 2018-04-05:
@azmosavi: It seems that you are on Mac. Unfortunately, I haven't tried MuJoCo on mac. A quick glance at the documentation of [mujoco-py](https://github.com/openai/mujoco-py#install-mujoco), it says following about MuJoCo installation:

> Unzip the downloaded mjpro150 directory into ~/.mujoco/mjpro150

Hence, here in my Ubuntu 14.04 LTS, I place MuJoCo binaries in `$HOME`. I would suggest you do the same on your system.

azmosavi, 2018-04-05:
Thanks @ravijo. I solved my problem. I actually followed this link:
 https://github.com/openai/mujoco-py/issues/74
I did:
mkdir ~/.mujoco
copied binary files, key and also git repo. Then simply:
cd mujoco-py && python setup.py install

damvantai, 2018-11-22:
Worked!!!
## In terminal:
sudo apt-get install -y python-numpy cmake zlib1g-dev libjpeg-dev libboost-all-dev gcc libsdl2-dev wget unzip

## In forder mujoco_py:
sudo pip3 install -e . --no-cache

## In hw1
sudo pip3 install openai gym tensorflow

## In .mujoco/ add key.txt and mujoco unzip

## Export path to ~/.bashrc
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/(name_user)/.mujoco/mjpro150/bin

machinaut, 2018-11-28:
Closing this issue for now because the build system has changed since.

If any new users encounter something similar, please open a new issue!  There's an issue template which helps us in getting the information we need.

Thanks~

closed 2018-11-28 by machinaut

kalki123123, 2019-05-14:
in my case installing libglew-dev did the trick!

yxh505613923, 2019-08-03:
No such file or directory: 'patchelf': 'patchelf'

download patchelf 0.9（https://nixos.org/releases/patchelf/patchelf-0.9/patchelf-0.9.tar.gz），unzip 
 then run ./configure, make, sudo make install，patchelf --version to check if it's ok

karanbirchahal-bc, 2020-06-21:
For me a combination of these commands were able to fix this issue:

```
sudo apt-get install libglew-dev
sudo apt-get install patchelf
sudo ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1 /usr/lib/x86_64-linux-gnu/libGL.so
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so
```

seyeeet, 2020-08-25:
@karanbirchahal-bc what does the last two commands does?

YujieLu10, 2021-05-22:
conda install -c menpo osmesa

HACLANG, 2021-11-13:
just need run:
`sudo apt-get install patchelf`