openai/mujoco-py #200: Corrupted installation: error code 7

kghelwegen, 2018-02-21:
After getting the mujoco license and binaries, I tried installing with `python3 -m pip install --user mujoco-py`. Installation seemed to freeze (twice). I now see this is more common, I should install from source instead.

However, it seems I corrupted something in my first attempt. I now receive an error code -7 when trying to install (also if I install from source). Additionally, if I try to import the gym module in a python script or console, I receive an error code 7: SIGEMT. According to google, this has something to do with an emulator trap, but I'm not sure what that means. 

I tried restarting, clearing my /tmp/ folder, and reinstalling gym via pip, but I keep getting this error. Any clue how I resolve this?

kghelwegen, 2018-02-21:
I have now figured out how to install gym and the latest version of mujoco-py (from source) in a virtual environment. So I'm starting to see why virtual environments are considered best practice ;) If anyone has a clue how to fix the gym module on my main python path, please let me know

koenhelwegen, 2019-06-20:
Closing as this was probably a problem in my local setup.

closed 2019-06-20 by koenhelwegen