openai/mujoco-py #209: mujoco-py==0.5.7 has AttributeError

timelong, 2018-03-06:
Hello!
when I install gym, I know that I must install mujoco_py first. So I used pip install mujoco-py==0.5.7, and
got license and mjpro150_linux. Then, I modified the setup.py in gym like this:
        'mujoco': ['mujoco_py==0.5.7', 'imageio'],
        'robotics': ['mujoco_py==0.5.7', 'imageio']
after that, I installed gym successfully. I can successfully import gym, and import mujoco_py. But, **mujoco_py.load_model_from_path(fullpath)** has attributeError,like this:
       **AttributeError:'module' object has no attribute 'load_model_from_path'**
I alse tested **mujoco_py.MjSim()**, which had the same error:
      **AttributeError: 'module' object has no attribute 'MjSim'**

How can I solve the problem? Please! ! !

HuiwenZhang, 2018-03-15:
I think that if you use mujoco-py=0.5.7，you   might use an older version of gym.

kirk86, 2018-03-15:
@HuiwenZhang I did I used 0.9.x but still couldn't get it to work properly.