openai/mujoco-py #210: Can't find mjmodel.h when install mujoco-py

Luminosite, 2018-03-08:
I installed mujoco-py with this command: pip3 install -U mujoco-py
and got following error:

Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /Library/Frameworks/Python.framework/Versions/3.5/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-2qv983rn-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    Compiling /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/cymj.pyx because it changed.
    [1/1] Cythonizing /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/cymj.pyx
    running build_ext
    building 'mujoco_py.cymj' extension
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/kr
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py
    creating /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/gl
    /usr/local/bin/gcc-7 -fno-strict-aliasing -Wsign-compare -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch i386 -arch x86_64 -g -DONMAC -Imujoco_py -I/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py -I/Users/luminosite/.mujoco/mjpro150/include -I/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/numpy/core/include -I/Library/Frameworks/Python.framework/Versions/3.5/include/python3.5m -c /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    gcc-7: warning: x86_64 conflicts with i386 (arch flags ignored)
    /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/mujoco_py/cymj.c:573:10: fatal error: mjmodel.h: No such file or directory
     #include "mjmodel.h"
              ^~~~~~~~~~~
    compilation terminated.
    error: command '/usr/local/bin/gcc-7' failed with exit status 1

    ----------------------------------------
Command "/Library/Frameworks/Python.framework/Versions/3.5/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-2qv983rn-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/kr/1v5wym8133j98hsyjwsbs8wc3kfvvf/T/pip-build-od9nnf1q/mujoco-py/

I searched problem, and find the needed mjmodel.h should be in /Users/luminosite/.mujoco/mjpro150/include, there's no such folder at all.

kyleliang919, 2018-03-13:
.folders are reserved for system, so you can not see it.
go to your home directory, firstly do
"sudo mkdir .mujoco"
then go to the folder where you extract the mjpro150, type
"mv mjpro150 ~/.mujoco/"
to move the folder into the destination
Hope this will help.

HuiwenZhang, 2018-03-14:
Well, it's strange. I do put the mujoco file in the right folder and I can find mjmodel.h in the include folder. But I still get this error

lutein, 2018-09-07:
Hi, meet the same issue. Have you solved this?

NishanthVAnand, 2018-10-01:
facing the same isuue

Luminosite, 2018-11-19:
Still didn't find solution.

machinaut, 2018-11-28:
@Luminosite can you re-try with the most recent version of mujoco-py and share the error message?

The build system now will look for the mujoco library to be installed to the correct location, and if it is not there, tell you the location that it was expecting to find it.