openai/mujoco-py #212: FATAL:/opt/local/bin/../libexec/as/x86_64/as: I don't understand 'm' flag!     error: command '/usr/local/bin/gcc-6' failed with exit status 1

scarlettDiMa, 2018-03-09:
Hi,

I was installing with sudo pip3 install -U 'mujoco-py<1.50.2,>=1.50.1' and there is an issue saying:

ExtensionBuilder/temp.macosx-10.6-intel-3.6/private/tmp/pip-build-gcusra91/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    **gcc-6: warning: x86_64 conflicts with i386 (arch flags ignored)
    FATAL:/opt/local/bin/../libexec/as/x86_64/as: I don't understand 'm' flag!
    error: command '/usr/local/bin/gcc-6' failed with exit status 1**
    
    ----------------------------------------
Command "/Library/Frameworks/Python.framework/Versions/3.6/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-build-gcusra91/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-bsdadb9x-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/tmp/pip-build-gcusra91/mujoco-py/

I have tried reinstall gcc6 and 7, remove mjpro150 and re-download it. However none of them works.

I use a Mac OS Sierra 10.12.5, python version 3.6.4, gcc 7, mujoco version: mjpro150 osx

Any help is appreciated

planetceres, 2018-03-22:
Same here, but using Mac High Sierra 10.13.3 and python 3.5.2

planetceres, 2018-03-23:
Looks like I was able to fix this with `export PATH=/usr/bin:$PATH` as described here:

https://github.com/dmlc/xgboost/issues/1945#issuecomment-290606179