openai/mujoco-py #214: ImportError: libOSMesa.so.8: cannot open shared object file: No such file or directory

sandipan1, 2018-03-09:
I am running on Ubuntu 14.0.1 x86_64 GNU/Linux. I get the following import error. 
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/sandipan/.mujoco/mjpro150/mujoco-py-1.50.1.1/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/sandipan/.mujoco/mjpro150/mujoco-py-1.50.1.1/mujoco_py/builder.py", line 296, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/sandipan/.mujoco/mjpro150/mujoco-py-1.50.1.1/mujoco_py/builder.py", line 57, in load_cython_ext
    mod = imp.load_dynamic("cymj", cext_so_path)
  File "/home/sandipan/miniconda3/envs/res/lib/python3.5/imp.py", line 343, in load_dynamic
    return _load(spec)
ImportError: libOSMesa.so.8: cannot open shared object file: No such file or directory
>>>

backpropper, 2018-09-25:
Any updates on this issue?

ShivaSoleimany, 2020-07-13:
No.