openai/mujoco-py #216: Run model from URDF

XuanLin, 2018-03-11:
Hello,

I have a robot urdf file that I want to use mujoco_py to simulate, so I did the following steps:

1. I add 
```
  <mujoco>
        <compiler meshdir="../meshes/"/>
  </mujoco>
```
  to my URDF file, then use 
`./compile model.urdf model.mjb`
 to convert it to mjb

2. I use the given example as a trySim.py file

```
#!/usr/bin/env python3

from mujoco_py import load_model_from_path, MjSim, MjViewer
from mujoco_py.modder import TextureModder
import os

model = load_model_from_path("model.mjb")
sim = MjSim(model)
viewer = MjViewer(sim)
modder = TextureModder(sim)

t = 0
while True:
    for name in sim.model.geom_names:
        modder.rand_all(name)

    viewer.render()
    t += 1
    if t > 100 and os.getenv('TESTING') is not None:
        break
```

3. I run it by

`sudo python3 trySim.py `

Then the screen blinks once and I get the following error:
```
Creating window glfw
Traceback (most recent call last):
  File "trySim.py", line 10, in <module>
    modder = TextureModder(sim)
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.42-py3.5.egg/mujoco_py/modder.py", line 223, in __init__
    self._cache_checker_matrices()
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.42-py3.5.egg/mujoco_py/modder.py", line 408, in _cache_checker_matrices
    tex_id = self.model.mat_texid[mat_id]
TypeError: 'NoneType' object is not subscriptable
```

I confirm that my model.mjb works in mujoco if I run ./simulate and just drag and drop, I also confirm that my trySim.py works for the given humanoid.xml example. So both my model and python code should be fine...

Does anyone know why it happens?
Thank you!

XuanLin, 2018-03-15:
Solved, it is because of mjcf file issue.

HuiwenZhang, 2018-03-15:
hi，I am planing to do the similar thing. I think that I might have the same problem. So could you make it more clear about what is wrong there and how to fix that？

XuanLin, 2018-03-28:
Hey, below is the code that I got it to work. I also attach the raw xml file generates by ./compile which doesn't work 
[RAW_Silvia.docx](https://github.com/openai/mujoco-py/files/1855084/RAW_Silvia.docx). 
There are multiple issues in the raw xml file. Upon trying, almost each line that the working code has where the raw xml file doesn't have is needed. Here are what I found:

1. The raw file doesn't have `<light>` and `<geom>` under `<worldbody>`, and doesn't have `<camera>` under `<torso>`.
2. The raw file doesn't specify joint range.
3. The raw file doesn't specify `<actuator>`.
4. The stuff in `<default>` needs to be there, I am not exactly sure what role they serve for.
5. I also manually put in `coordinate="local"`, `inertiafromgeom="true"`, `<option integrator="RK4"  timestep="0.01"/>`.

```
<mujoco model="six-legged">
    <compiler angle="radian" coordinate="local" inertiafromgeom="true" meshdir="../meshes/"/>
    <option integrator="RK4" timestep="0.01"/>
    <default>
        <joint armature="1" damping="1" limited="true"/>
        <geom conaffinity="0" condim="3" density="5.0" friction="1 0.5 0.5" margin="0.01" rgba="0.8 0.6 0.4 1"/>
    </default>
    <asset>
        <texture builtin="gradient" height="100" rgb1="1 1 1" rgb2="0 0 0" type="skybox" width="100"/>
        <texture builtin="flat" height="1278" mark="cross" markrgb="1 1 1" name="texgeom" random="0.01" rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" type="cube" width="127"/>
        <texture builtin="checker" height="100" name="texplane" rgb1="0 0 0" rgb2="0.8 0.8 0.8" type="2d" width="100"/>
        <material name="MatPlane" reflectance="0.5" shininess="1" specular="1" texrepeat="60 60" texture="texplane"/>
        <material name="geom" texture="texgeom" texuniform="true"/>
        <mesh name="torso" file="torso.STL" />
        <mesh name="RF_Coxa" file="RF_Coxa.STL" />
        <mesh name="RF_Femur" file="RF_Femur.STL" />
        <mesh name="RF_Tibia" file="RF_Tibia.STL" />
        <mesh name="RM_Coxa" file="RM_Coxa.STL" />
        <mesh name="RM_Femur" file="RM_Femur.STL" />
        <mesh name="RM_Tibia" file="RM_Tibia.STL" />
        <mesh name="RR_Coxa" file="RR_Coxa.STL" />
        <mesh name="RR_Femur" file="RR_Femur.STL" />
        <mesh name="RR_Tibia" file="RR_Tibia.STL" />
        <mesh name="LR_Coxa" file="LR_Coxa.STL" />
        <mesh name="LR_Femur" file="LR_Femur.STL" />
        <mesh name="LR_Tibia" file="LR_Tibia.STL" />
        <mesh name="LM_Coxa" file="LM_Coxa.STL" />
        <mesh name="LM_Femur" file="LM_Femur.STL" />
        <mesh name="LM_Tibia" file="LM_Tibia.STL" />
        <mesh name="LF_Coxa" file="LF_Coxa.STL" />
        <mesh name="LF_Femur" file="LF_Femur.STL" />
        <mesh name="LF_Tibia" file="LF_Tibia.STL" />
    </asset>

    <worldbody>
        <light cutoff="100" diffuse="1 1 1" dir="-0 0 -1.3" directional="true" exponent="1" pos="0 0 1.3" specular=".1 .1 .1"/>
        <geom conaffinity="1" condim="3" material="MatPlane" name="floor" pos="0 0 0" rgba="0.8 0.9 0.8 1" size="40 40 40" type="plane"/>

        <body name="torso" pos="0 0 0">
            <camera name="track" mode="trackcom" pos="0 -3 0.3" xyaxes="1 0 0 0 0 1"/>
            <inertial pos="-1.04106e-10 -0.000333377 -0.0356225" quat="-4.96459e-05 0.707107 -6.11763e-05 0.707107" mass="1.94003" diaginertia="0.0628118 0.0329645 0.032962" />
            <geom type="mesh" rgba="0.752941 0.752941 0.752941 1" mesh="torso" />
            <joint armature="0" damping="0" limited="false" margin="0.01" name="root" pos="0 0 0" type="free"/>

            <body name="RF_Coxa" pos="0.1104 0.19121 -0.0411" quat="0.866025 0 0 0.500001">
                <inertial pos="0.0285 1.4128e-09 -6.5463e-10" quat="0.92388 0.382683 0 0" mass="0.06196" diaginertia="8.8248e-05 6.265e-05 6.265e-05" />
                <joint name="RF_Coxa" pos="0 0 0" axis="0 0 1" range="-180 180" type="hinge"/>
                <geom type="mesh" rgba="0.75294 0.75294 0.75294 1" mesh="RF_Coxa" />
                <body name="RF_Femur" pos="0.057 0.0005 0" quat="0.707105 0.707108 0 0">
                    <inertial pos="0.10825 -0.0059319 -4.2472e-07" quat="0.514707 0.514685 0.484871 0.484845" mass="0.36917" diaginertia="0.00311354 0.0029635 0.000293065" />
                    <joint name="RF_Femur" pos="0 0 0" axis="0 0 1" range="-180 180" type="hinge"/>
                    <geom type="mesh" rgba="0.75294 0.75294 0.75294 1" mesh="RF_Femur" />
                    <body name="RF_Tibia" pos="0.1975 -0.0265 -0.0005">
                        <inertial pos="2.4863e-08 -0.16811 8.3345e-09" quat="0.707107 0.707107 0 0" mass="0.11462" diaginertia="0.0022169 0.0021771 5.0466e-05" />
                        <joint name="RF_Tibia" pos="0 0 0" axis="0 0 1" range="-180 180" type="hinge"/>
                        <geom type="mesh" rgba="0.75294 0.75294 0.75294 1" mesh="RF_Tibia" />
                    </body>
                </body>
            </body>
[... 81 lines cut ...]
        </body>
    </worldbody>

    <actuator>
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="RF_Coxa" gear="150"/>
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="RF_Femur" gear="150"/>
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="RF_Tibia" gear="150"/>
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="RM_Coxa" gear="150"/>
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="RM_Femur" gear="150"/>
[... cut ...]
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="LM_Coxa" gear="150"/>
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="LM_Femur" gear="150"/>
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="LM_Tibia" gear="150"/>    
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="LF_Coxa" gear="150"/>
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="LF_Femur" gear="150"/>
      <motor ctrllimited="true" ctrlrange="-1.0 1.0" joint="LF_Tibia" gear="150"/>  
    </actuator> 
</mujoco>
```

gabriel80808, 2018-08-01:
Thanks! This was helpful!

jinparksj, 2018-11-11:
Thank! This is really helpful!

ilinojkovic, 2019-04-18:
Hi @XuanLin thanks for the answer, however I find it a bit confusing. In the first post you had a problem with a urdf file, however in the following explanation you presented xml files. Is there a chance you can post the urdf as well?

XuanLin, 2019-04-23:
> Hi @XuanLin thanks for the answer, however I find it a bit confusing. In the first post you had a problem with a urdf file, however in the following explanation you presented xml files. Is there a chance you can post the urdf as well?

Here it is:
[urdf_SiLVIA.docx](https://github.com/openai/mujoco-py/files/3105981/urdf_SiLVIA.docx)
It is a bit long because this robot has 6 legs.

ilinojkovic, 2019-05-21:
> Here it is:
> [urdf_SiLVIA.docx](https://github.com/openai/mujoco-py/files/3105981/urdf_SiLVIA.docx)
> It is a bit long because this robot has 6 legs.

I see you have the `actuator` tag inside the xml file, but I cannot seem to find the equivalent inside your urdf. Did you add this manually after compilation?

XuanLin, 2019-05-28:
> I see you have the `actuator` tag inside the xml file, but I cannot seem to find the equivalent inside your urdf. Did you add this manually after compilation?

Yes, I did it manually. ./compile doesn't give me the <actuator> part.

tuonan, 2019-06-11:
@XuanLin @jinparksj @gabriel80808  what's "./compile"?

jinparksj, 2019-06-11:
> @XuanLin @jinparksj @gabriel80808 what's "./compile"?

It is the command in terminal for compiling URDF file. The compile execution file may be in the mujoco folder. You can find 'compile' execution file.

tuonan, 2019-06-14:
@XuanLin I cann't use compile.cpp，so i can not use ,/compile, can you help me?

YanhuaZhang516, 2019-11-24:
HI, I am new to mujoco. When I use "./compile model.urdf model.mjb", but the mjb file I got is empty, do you have the same question before? thanks a lot. So i still need to add the contents to  the XML file by myself?

XuanLin, 2019-12-04:
> HI, I am new to mujoco. When I use "./compile model.urdf model.mjb", but the mjb file I got is empty, do you have the same question before? thanks a lot. So i still need to add the contents to the XML file by myself?

Sorry for the late reply. Is the problem solved? I didn't encounter this issue before. Have you add 

`  <mujoco>
        <compiler meshdir="../meshes/"/>
  </mujoco>`

to your urdf file? Also you want to make sure you have the writing permission to your files in that location.

If you don't mind you can share your file and I can try it on my computer.

YanhuaZhang516, 2019-12-18:
> Sorry for the late reply. Is the problem solved? I didn't encounter this issue before. Have you add
>
> ```
>   <mujoco>
>         <compiler meshdir="../meshes/"/>
>   </mujoco>
> ```
>
> to your urdf file? Also [...]

Thanks for your reply! I solved it . And by the way, could you use xml.eletree.parse() to parse your urdf file? I meet a  new problem with urdf file :(

XuanLin, 2020-01-01:
> Thanks for your reply! I solved it . And by the way, could you use xml.eletree.parse() to parse your urdf file? I meet a new problem with urdf file :(

Sorry I haven't used xml.eletree.parse() in my project. I haven't been working on mujoco for a while so it will probably take me some time to pick up the stuff I left aside ...