openai/mujoco-py #218: compilation error on importing mujoco_py

kirk86, 2018-03-14:
Hi, after installing `mujoco_py`, upon firring up python interpreter and executing `import mujoco_py` I get the following error:

```
/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/cymj.c:609:21: fatal error: mjmodel.h: No such file or directory
compilation terminated.
Traceback (most recent call last):
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command '/usr/bin/gcc-5' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/builder.py", line 502, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/builder.py", line 81, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/builder.py", line 238, in build
    built_so_file_path = self._build_impl()
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/builder.py", line 310, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/builder.py", line 261, in _build_impl
    dist.run_commands()
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/builder.py", line 157, in build_extensions
    build_ext.build_extensions(self)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/builder.py", line 127, in caching_compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/bin/gcc-5' failed with exit status 1
```

kirk86, 2018-03-15:
Things that I've done to resolve some of the issues. I'm on a headless ubuntu 16.04 server with normal user privileges (not root access).

Some packages were missing from the system `glew, patchelf`

`conda install -c conda-forge glew, patchelf` if you have sudo access then `sudo apt-get install libglew-dev patchelf`

`git clone https://github.com/openai/mujoco-py.git && cd mujoco-py`

At this point I was getting error upon `python setup.py install` about missing `glew.h` file.

`edit mujoco-py/gl/eglshim.c` and change `#include <GL/glew.h> to </home/user/miniconda/include/GL/glew.h>` then again execute `python setup.py install` now this time hopefully things will go according to plan and no errors will occur.

`pwd`
`mujoco-py`
`cd .. && rm -rf mujoco-py`

fire up `python` and `import mujoco_py` the first time it will still compile some stuff and put the the necessary folders.

Next install gym `pip install 'gym[all]'` make sure first you have installed all the dependencies and requirements!

Now `cd .mujoco/mjpro150/sample && make` then `echo "export LD_LIBRARY_PATH="$HOME/.mujoco/mjpro150/bin:$LD_LIBRARY_PATH"" >> ~/.bashrc` 

Now when I try the following:
```
import gym
env = gym.make('Humanoid-v2')
env.reset()
env.render()
```
I get the following error:

```
GLFW error (code %d): %s 65544 b'X11: RandR gamma ramp support seems broken'
GLFW error (code %d): %s 65542 b'EGL: Failed to get EGL display: Success'
Creating window glfw
ERROR: OpenGL version 1.5 or higher required
```
which I don't know yet how to solve, I've tried using this `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so.1.13.0:/usr/lib/nvidia-387/libGL.so.1 python` to fire up python in case it solves the issue but nothing so far.

kirk86, 2018-09-04:
@RchalYang yup, 

make sure you have mujoco bin folder in your `LD_LIBRARY_PATH`, as well as the cuda lib folder, the extras/CUPTI/lib64, and the nvidia drivers `/usr/lib/nvidia-387`

next put make sure you have the following:
```
export LD_PRELOAD="/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-387/libGL.so
```
change it accordingly to point to your nvidia-version-drivers.

Also put this as well:

`export LIBGL_ALWAYS_INDIRECT=1`

phaniram-sayapaneni, 2018-09-12:
I have same issue, looked in to comments for all similar issues couldn't find a fix that works.

It breaks down at:
---> 11     import mujoco_py
ends with:
command '/usr/local/bin/gcc-6' failed with exit status 1

any help from @openai. I need to fix this as soon as possible. Thanks!

zhangchuheng123, 2018-11-30:
@kirk86 @RchalYang I have the same issue. I checked my environment variables

```bash
/home1/zhangchuheng/.mujoco/mjpro150/bin:/usr/lib/nvidia-384:/home1/zhangchuheng/.mujoco/mjpro150/bin:/usr/lib/nvidia-384:/usr/local/cuda/lib64
zhangchuheng@user-ubuntu:~/projects/RL/notebook$ echo $LD_PRELOAD
/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-384/libGL.so
zhangchuheng@user-ubuntu:~/projects/RL/notebook$ echo $LIBGL_ALWAYS_INDIRECT
1
zhangchuheng@user-ubuntu:~/projects/RL/notebook$
```

I wrote a simple program

```python
import gym
from PIL import Image
env = gym.make('Hopper-v2')
env.reset()
ima = Image.fromarray(env.render(mode='rgb_array'))
print('done')
```

It finished running although the errors are shown.

```bash
zhangchuheng@user-ubuntu:~/projects/RL/notebook$ python test.py
GLFW error (code %d): %s 65544 b'X11: RandR gamma ramp support seems broken'
GLFW error (code %d): %s 65544 b'X11: RandR monitor support seems broken'
Creating window glfw
done
```

It even output a image with only the black-and-white grids and text like "FPS & Solver iterations" but there's no hopper on it !!!

So strange!!!

Needs help. Thanks!