openai/mujoco-py #220: Segmentation fault (core dumped)

JoshTao, 2018-03-17:
It show me the error when I directly use the command `python`

`running build_ext
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
Creating window glfw
Segmentation fault (core dumped)`

I have used the following method
`gdb python`
`run test.py`

and I get the following information:

`Starting program: /usr/local/bin/python3.5 test.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
running build_ext
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
Creating window glfw

Program received signal SIGSEGV, Segmentation fault.
0x00007fffede0928e in glGetString ()
   from /usr/lib/x86_64-linux-gnu/libOSMesa.so.6`

However, it works well when I execute `./simulate` in the `./mujoco/mjpro150/bin`
It seems to be the problem of mujoco-py
#44 doesn't fix my issue

bionick87, 2018-04-26:
same for me any solution?

Robin970822, 2018-05-17:
Same problem as yours.

tianheyu927, 2018-07-26:
Have you figured out how to resolve this?

wdjpng, 2018-08-17:
Same problem :(

csingh27sewts, 2021-01-29:
restart your system