openai/mujoco-py #224: Unable to build example

arunavanag591, 2018-03-23:
Hello I am using mujoco-py the latest and mjpro150, ubuntu 16 and virtualenv.

I am trying to build the simple example: 

```
#!/usr/bin/python

import gym
from mujoco_py import load_model_from_path, MjSim
from os.path import dirname
env = gym.make('Copy-v0')
env.reset()
env.render()
model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +"/xmls/claw.xml")
sim = mujoco_py.MjSim(model)
```

and it's throwing a huge error:

```

python3 gym_test.py 
running build_ext
building 'mujoco_py.cymj' extension
/home/rosi/an_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/rosi/an_ws/mujoco-py/mujoco_py/cymj.o up to date, skipping.
/home/rosi/an_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/rosi/an_ws/mujoco-py/mujoco_py/gl/eglshim.o up to date, skipping.
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /home/rosi/an_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/rosi/an_ws/mujoco-py/mujoco_py/cymj.o /home/rosi/an_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/rosi/an_ws/mujoco-py/mujoco_py/gl/eglshim.o -L/home/rosi/.mujoco/mjpro150/bin -Wl,--enable-new-dtags,-R/home/rosi/.mujoco/mjpro150/bin -lmujoco150 -lOpenGL -lEGL -lGL -lglewegl -o /home/rosi/an_ws/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
/usr/bin/ld: cannot find -lOpenGL
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/usr/lib/python3.5/distutils/unixccompiler.py", line 207, in link
    self.spawn(linker + ld_args)
  File "/usr/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/usr/lib/python3.5/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/usr/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "gym_test.py", line 4, in <module>
    from mujoco_py import load_model_from_path, MjSim
  File "/home/rosi/an_ws/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/rosi/an_ws/mujoco-py/mujoco_py/builder.py", line 502, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/rosi/an_ws/mujoco-py/mujoco_py/builder.py", line 81, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/rosi/an_ws/mujoco-py/mujoco_py/builder.py", line 238, in build
    built_so_file_path = self._build_impl()
  File "/home/rosi/an_ws/mujoco-py/mujoco_py/builder.py", line 310, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/rosi/an_ws/mujoco-py/mujoco_py/builder.py", line 261, in _build_impl
    dist.run_commands()
  File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/rosi/.local/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/usr/lib/python3.5/distutils/command/build_ext.py", line 338, in run
    self.build_extensions()
  File "/home/rosi/an_ws/mujoco-py/mujoco_py/builder.py", line 157, in build_extensions
    build_ext.build_extensions(self)
  File "/home/rosi/.local/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/usr/lib/python3.5/distutils/command/build_ext.py", line 557, in build_extension
    target_lang=language)
  File "/usr/lib/python3.5/distutils/ccompiler.py", line 717, in link_shared_object
    extra_preargs, extra_postargs, build_temp, target_lang)
  File "/usr/lib/python3.5/distutils/unixccompiler.py", line 209, in link
    raise LinkError(msg)
distutils.errors.LinkError: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

sungsulim, 2018-04-26:
I have a similar error when installing on my linux machine, but with a different library.
`/usr/bin/ld: cannot find -lOSMesa`

Kmeco, 2018-07-31:
@arunavanag591 any luck with resolving the issue? I'm getting exactly the same error. Thanks!