openai/mujoco-py #226: Unable to install Mujoco. Stat: no such file or directory

Namlich, 2018-03-26:
Hi everyone,

I just can't seem to get this working. I've had a few different errors, that i did manage to get through, but this one i just can't seem to fix. Any help is very appreciated!

Im still a bit hopeless regarding Ubuntu and using the terminal, so if you are so kind to try to help me, please be very specific :)

Let me know if i should post any other information.

Thanks in advance.

Kindly,
Benjamin

` Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Successfully built gym
Failed to build mujoco-py
Installing collected packages: numpy, chardet, certifi, idna, urllib3, requests, six, future, pyglet, glfw, Cython, Pillow, imageio, pycparser, cffi, mujoco-py, Box2D-kengz, PyOpenGL, atari-py, gym
  Running setup.py install for mujoco-py ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-3t2as6ha/mujoco-py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-v315n455-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    running build_ext
    stat: No such file or directory
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-3t2as6ha/mujoco-py/setup.py", line 52, in <module>
        cmdclass={'build': Build},
      File "/home/benjamin/.local/lib/python3.5/site-packages/setuptools/__init__.py", line 129, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.5/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/home/benjamin/.local/lib/python3.5/site-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.5/distutils/command/install.py", line 583, in run
        self.run_command('build')
      File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-build-3t2as6ha/mujoco-py/setup.py", line 19, in run
        spec.loader.exec_module(builder)
      File "<frozen importlib._bootstrap_external>", line 665, in exec_module
      File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
      File "/tmp/pip-build-3t2as6ha/mujoco-py/mujoco_py/builder.py", line 22, in <module>
        from mujoco_py.utils import discover_mujoco
      File "/tmp/pip-build-3t2as6ha/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
        from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
      File "/tmp/pip-build-3t2as6ha/mujoco-py/mujoco_py/builder.py", line 502, in <module>
        cymj = load_cython_ext(mjpro_path)
      File "/tmp/pip-build-3t2as6ha/mujoco-py/mujoco_py/builder.py", line 81, in load_cython_ext
        cext_so_path = builder.build()
      File "/tmp/pip-build-3t2as6ha/mujoco-py/mujoco_py/builder.py", line 238, in build
        built_so_file_path = self._build_impl()
      File "/tmp/pip-build-3t2as6ha/mujoco-py/mujoco_py/builder.py", line 312, in _build_impl
        join(get_nvidia_lib_dir(), 'libOpenGL.so.0'))
      File "/tmp/pip-build-3t2as6ha/mujoco-py/mujoco_py/builder.py", line 173, in fix_shared_library
        so_file])
      File "/usr/lib/python3.5/subprocess.py", line 581, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['patchelf', '--add-needed', '/usr/lib/nvidia-384/libOpenGL.so.0', '/tmp/pip-build-3t2as6ha/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so']' returned non-zero exit status 1
`

tiw51, 2018-07-11:
Hi, I am having the same problem here

ZitongBo, 2020-08-07:
Same,  is there any solutions?

qsa-fox, 2022-09-14:
`pip install patchelf` works for me.