openai/mujoco-py #228: Unable to import mujoco-py

novatig, 2018-03-29:
Hi,

I've been using mujoco-py for a while and I'm now trying installing it on a new system.
I've installed it from scratch, on a fresh python 3.5.2 and 3.6.1. 
When I import it for the first time I get this error:

```
>>> import mujoco_py
Compiling /cluster/home/novatig/mujoco-py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /cluster/home/novatig/mujoco-py/mujoco_py/cymj.pyx
warning: mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
```
More warning related to redeclared types, then `cymj.cpython-36m-x86_64-linux-gnu.so` is compiled.  
After compilation terminates, and any additional time I try `import mujoco_py`, I get this error:
```
Traceback (most recent call last):  
  File "<stdin>", line 1, in <module>  
  File "/cluster/home/novatig/mujoco-py/mujoco_py/__init__.py", line 1, in <module>  
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException  
  File "/cluster/home/novatig/mujoco-py/mujoco_py/builder.py", line 502, in <module>  
    cymj = load_cython_ext(mjpro_path)  
  File "/cluster/home/novatig/mujoco-py/mujoco_py/builder.py", line 82, in load_cython_ext  
    return load_dynamic_ext('cymj', cext_so_path)  
  File "/cluster/home/novatig/mujoco-py/mujoco_py/builder.py", line 88, in load_dynamic_ext  
    return loader.load_module()  
ImportError: /cluster/home/novatig/mujoco-py/mujoco_py/generated  /cymj_linuxcpuextensionbuilder.so: undefined symbol: _ZN4llvm19RTDyldMemoryManagerD2Ev
```
If I run `pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'`, the same error prevents me from installing.

Thank you in advance for any suggestion.

Foobanana, 2018-04-01:
The proposed solution of PyCharm is to, quote, Try to run this command from the system terminal. Make sure that you use the correct version of 'pip' installed for your Python interpreter located at 'C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\Scripts\python.exe'. , end quote.

When I do this, I get the error: Command "python setup.py egg_info" failed with error code 1 in C:\Users\frogs\AppData\Local\Temp\pip-build-vn3mou05\mujoco-py\

It also tells me: You are using pip version 9.0.1, however version 9.0.3 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

Then when I do so: 
C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\Scripts>python -m pip install --upgrade pip
Requirement already up-to-date: pip in c:\users\frogs\program files\python35\mujoco_tests\lib\site-packages

Foobanana, 2018-04-01:
(sorry for accidentally commenting on your thing earlier and giving you false hope)

wojzaremba, 2018-04-10:
Try out again with python 3.6.

novatig, 2018-04-10:
Thank you for the reply.
Unfortunately, same error with python 3.6.0.

wojzaremba, 2018-04-11:
Dockerfile describes full, reproducable environment. Check if you can run it there.

ebimor, 2018-04-30:
@novatig I have the same problem! Were you able to resolve it?

ebimor, 2018-04-30:
I resolved by following this thread

https://github.com/openai/mujoco-py/issues/80

install both mujoco-py and gym from source!

git clone https://github.com/openai/gym
cd gym 
pip install -e .[all]

instal mujoco-py from source:

git clone https://github.com/openai/mujoco-py.git
cd mujoco-py
python setup.py install

You might have to install dependencies

sudo apt-get install xvfb libav-tools xorg-dev libsdl2-dev swig cmake 

You should be able to have this example working:

import gym
env = gym.make('Hopper-v2')
print(env.observation_space)
print(env.action_space)
env.reset()
for _ in range(100):
    env.render()
    env.step(env.action_space.sample()) # take a random action