openai/mujoco-py #229: distutils.errors.DistutilsExecError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\link.exe' failed with exit status 1158

Foobanana, 2018-03-31:
I've been trying to get the mujoco-py package working for about fifteen hours now, and will detail what I've done below.

My username on my computer is frogsnakeman, abrreviated frogs, in case I copy any paths.

My system is Windows 10, 64 bit. 
I have Python 3.5.2, and I'm *pretty* sure that my IDE is using that.
My IDE is newly updated: JetBrains PyCharm Edu 2018.1 x64

I've downloaded the mjpro150 files, unzipped them, and they are sitting happily in C:\Users\frogs\.mujoco . I've written and tested some XML files in the MuJoCo environment and they work fine.
I've placed my mjkey.txt and LICENSE.txt files in there and in C:\Users\frogs\.mujoco\mjpro150\bin because I wasn't sure which was the correct one.
I've downloaded the MS Visual Build Tools from this site:
https://www.visualstudio.com/downloads/#build-tools-for-visual-studio-2017
I think I downloaded them from somewhere else too, but I can't find the link. I *think* they're updated to what they need to be (but I'm less sure of this).
I've tried going through these things in the appveyor file: https://github.com/openai/mujoco-py/blob/master/.appveyor.yml#L16-L32 , but I don't think they all worked (see the very end of this). 
I think after about 10 hours I finally got mujoco-py on my computer in some sense, because it finally showed up as a usable package on PyCharm instead of saying "no module named mujoco_py". 
I've got pip 9.0.1 for sure, and when I try to install 9.0.3 I get the error AttributeError: 'NoneType' object has no attribute 'bytes'
I get the error distutils.errors.DistutilsExecError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\link.exe' failed with exit status 1158 when trying to install the package mujoco-py ; I don't know if this is connected to the pip 9.0.3 install error.

The only thing PyCharm says is to try running pip install pip==9.0.3 in the command window, but I don't know if they mean the python command window, the Windows command window, and whether or not I have to be in a particular path or directory when I do this.

I don't know how to use the command window.
I've only programmed in Python five or six times, though I have a bunch of MATLAB experience.
I've only installed one package in Python ever, numpy.
I've never used MuJoCo before, or any other virtual environment software.
I need help.

Foobanana, 2018-04-01:
I'll put the full command output here:

> Collecting mujoco-py
  Using cached mujoco-py-1.50.1.43.tar.gz
    Complete output from command python setup.py egg_info:
    _configtest.c
    warning: build_py: byte-compiling is disabled, skipping.
    
    _cffi_backend.c
    c/_cffi_backend.c(368): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data
    c/_cffi_backend.c(369): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data
    c/_cffi_backend.c(1050): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1059): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1084): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1107): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1126): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1140): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1282): warning C4244: '=': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1284): warning C4244: '=': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1586): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1587): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1589): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1601): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1602): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1605): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1624): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1668): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1733): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1735): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1755): warning C4244: '=': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(1758): warning C4244: '=': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(2114): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(2119): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(2123): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(2138): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(2194): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(2298): warning C4244: 'return': conversion from 'Py_hash_t' to 'long', possible loss of data
    c/_cffi_backend.c(2292): warning C4244: 'initializing': conversion from 'Py_hash_t' to 'long', possible loss of data
    c/_cffi_backend.c(3090): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(3689): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(3789): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(3905): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(3946): warning C4244: 'function': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(4455): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
    c/_cffi_backend.c(4473): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
    c/_cffi_backend.c(4617): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
    c/_cffi_backend.c(4608): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data
    c/_cffi_backend.c(4629): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data
    c/_cffi_backend.c(4983): warning C4244: '=': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(5024): warning C4244: '=': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(5032): warning C4244: '=': conversion from 'Py_ssize_t' to 'int', possible loss of data
    c/_cffi_backend.c(5248): warning C4244: '=': conversion from 'Py_ssize_t' to 'unsigned short', possible loss of data
    c/_cffi_backend.c(5377): warning C4267: 'function': conversion from 'size_t' to 'int', possible loss of data
    c/_cffi_backend.c(5381): warning C4244: '=': conversion from 'Py_ssize_t' to 'int', possible loss of data
[... cut ...]
    distutils.errors.DistutilsError: Setup script exited with error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\link.exe' failed with exit status 1158
    
    ----------------------------------------

Command "python setup.py egg_info" failed with error code 1 in C:\Users\frogs\AppData\Local\Temp\pycharm-packaging\mujoco-py\
You are using pip version 9.0.1, however version 9.0.3 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

Foobanana, 2018-04-01:
The proposed solution of PyCharm is to, quote, Try to run this command from the system terminal. Make sure that you use the correct version of 'pip' installed for your Python interpreter located at 'C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\Scripts\python.exe'. , end quote.

When I do this, I get the error: Command "python setup.py egg_info" failed with error code 1 in C:\Users\frogs\AppData\Local\Temp\pip-build-vn3mou05\mujoco-py\

It also tells me: You are using pip version 9.0.1, however version 9.0.3 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

Then when I do so: 
C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\Scripts>python -m pip install --upgrade pip
Requirement already up-to-date: pip in c:\users\frogs\program files\python35\mujoco_tests\lib\site-packages 

Similar to #189

Foobanana, 2018-04-01:
Ran: pip install --upgrade setuptools
Then I again ran: python -m pip install
Got a slightly different reason for the same error.

I then ran, to forcibly update pip,: easy_install -U pip
Some error with link.exe failing, with scary red letters saying:

Command ""c:\users\frogs\program files\python35\mujoco_tests\scripts\python.exe" -u -c "import setuptools, tokenize;__file__='C:\\Users\\frogs\\AppData\\Local\\Temp\\pip-install-0n7dd2m7\\mujoco-py\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\frogs\AppData\Local\Temp\pip-record-mnn8b9l1\install-record.txt --single-version-externally-managed --compile --install-headers "c:\users\frogs\program files\python35\mujoco_tests\include\site\python3.5\mujoco-py"" failed with error code 1 in C:\Users\frogs\AppData\Local\Temp\pip-install-0n7dd2m7\mujoco-py\

Foobanana, 2018-04-01:
Ran: easy_install -U mujoco-py
It actually seemed to update and the package is now added to the interpreter (if that makes any sense)

Now I get the following error: 

>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
creating C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg
Traceback (most recent call last):
  File "C:\Program Files\Python35\lib\distutils\dir_util.py", line 70, in mkpath
    os.mkdir(head, mode)
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\frogs\\Program Files\\Python35\\MuJoCo_Tests\\lib\\site-packages\\mujoco_py-1.50.1.43-py3.5.egg\\mujoco_py\\generated\\_pyxbld_WindowsExtensionBuilder\\temp.win-amd64-3.5\\Release\\Users\\frogs\\Program Files\\Python35\\MuJoCo_Tests\\lib\\site-packages\\mujoco_py-1.50.1.43-py3.5.egg'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg\mujoco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg\mujoco_py\builder.py", line 502, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg\mujoco_py\builder.py", line 81, in load_cython_ext
    cext_so_path = builder.build()
  File "C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg\mujoco_py\builder.py", line 238, in build
    built_so_file_path = self._build_impl()
  File "C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg\mujoco_py\builder.py", line 261, in _build_impl
    dist.run_commands()
  File "C:\Program Files\Python35\lib\distutils\dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "C:\Program Files\Python35\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\Cython\Distutils\old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "C:\Program Files\Python35\lib\distutils\command\build_ext.py", line 338, in run
    self.build_extensions()
  File "C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg\mujoco_py\builder.py", line 157, in build_extensions
    build_ext.build_extensions(self)
  File "C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\Cython\Distutils\old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "C:\Program Files\Python35\lib\distutils\command\build_ext.py", line 532, in build_extension
    depends=ext.depends)
  File "C:\Program Files\Python35\lib\distutils\_msvccompiler.py", line 308, in compile
    sources, depends, extra_postargs)
  File "C:\Program Files\Python35\lib\distutils\ccompiler.py", line 348, in _setup_compile
    self.mkpath(os.path.dirname(obj))
  File "C:\Program Files\Python35\lib\distutils\ccompiler.py", line 915, in mkpath
    mkpath(name, mode, dry_run=self.dry_run)
  File "C:\Program Files\Python35\lib\distutils\dir_util.py", line 74, in mkpath
    "could not create '%s': %s" % (head, exc.args[-1]))
distutils.errors.DistutilsFileError: could not create 'C:\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\frogs\Program Files\Python35\MuJoCo_Tests\lib\site-packages\mujoco_py-1.50.1.43-py3.5.egg': The system cannot find the path specified

OscarLiXi, 2019-01-31:
The problem might be the VS2015 mess with other version of Windows SDK, see:  https://github.com/conan-io/conan/issues/1402#issuecomment-327949466

You can check if you have installed multiple version of visual studio, e.g., VS2017.
For me, I have installed both VS2015 and VS2017, and to compile Mujoco using VS2017 developer Command Prompt is working for me (VS2015, Anaconda command prompt is not working!)

Hope it helps.

kingismybrother, 2019-02-14:
1.Run cmd  as administrator 
2. set PATH=C:\Users\xxx\.mujoco\mjpro150\bin;%PATH% 
hope it helps  >_<

HORIZONNN, 2020-07-04:
1. install the [windows 8.1 SDK](https://developer.microsoft.com/zh-cn/windows/downloads/sdk-archive/
) ;
2. add "C:\Program Files (x86)\Wexit()indows Kits\8.1\bin\x86" which include "rc.exe" to the environment path;
3. restart the cmd and input "import mujoco_py" for testing.