openai/mujoco-py #235: ERROR importing mujoco_py

ganyeshprasanna, 2018-04-06:
I created a conda environment (python 3.5.2) and installed 
I followed the instructions in the Readme file and was able to install mujoco-py
When I check the packages installed in the conda environment this is what I see:
```
# packages in environment at /home/ganesh/anaconda2/envs/mj:
#
# Name                    Version                   Build  Channel
ca-certificates           2018.03.07                    0  
certifi                   2018.1.18                py35_0  
cffi                      1.11.4                  py35h9745a5d_0  
cython                    0.27.3           py35h6cdc64b_0  
freetype                  2.8                  hab7d2ae_1  
glfw                      1.6.0                     <pip>
imageio                   2.2.0            py35hd0a6de2_0  
intel-openmp              2018.0.0                      8  
jpeg                      9b                   h024ee3a_2  
libffi                    3.2.1                hd88cf55_4  
libgcc-ng                 7.2.0                hdf63c60_3  
libgfortran-ng            7.2.0                hdf63c60_3  
libpng                    1.6.34               hb9fc6fc_0  
libstdcxx-ng              7.2.0                hdf63c60_3  
libtiff                   4.0.9                h28f6b97_0  
mkl                       2018.0.2                      1  
mkl_fft                   1.0.1            py35h3010b51_0  
mkl_random                1.0.1            py35h629b387_0  
mujoco-py                 1.50.1.52                 <pip>
numpy                     1.14.2           py35hdbf6ddf_1  
olefile                   0.45.1                   py35_0  
openssl                   1.0.2o               h20670df_0  
osmesa                    12.2.2.dev                    0    menpo
pillow                    4.2.1            py35h03abc04_0  
pip                       9.0.3                    py35_0  
pycparser                 2.18             py35h61b3040_1  
python                    3.5.2                         0  
readline                  6.2                           2  
setuptools                39.0.1                   py35_0  
sqlite                    3.13.0                        0  
tk                        8.5.18                        0  
wheel                     0.31.0                   py35_0  
xz                        5.2.3                h55aa19d_2  
zlib                      1.2.11               ha838bed_2  
```
Mujoco-py is installed 
But when I try to import mujoco_py inside python I get an error that mjmodel.h does not exist
```
import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py -I/home/ganesh/.mujoco/mjpro150/include -I/home/ganesh/.local/lib/python3.5/site-packages/numpy/core/include -I/home/ganesh/anaconda2/envs/mj/include/python3.5m -c /home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/cymj.c -o /home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.52_35_linuxcpuextensionbuilder/temp.linux-x86_64-3.5/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/cymj.c:610:21: fatal error: mjmodel.h: No such file or directory
compilation terminated.
Traceback (most recent call last):
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/builder.py", line 452, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/builder.py", line 85, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/builder.py", line 188, in build
    built_so_file_path = self._build_impl()
  File "/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/builder.py", line 242, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/builder.py", line 211, in _build_impl
    dist.run_commands()
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/ganesh/.local/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/command/build_ext.py", line 338, in run
    self.build_extensions()
  File "/home/ganesh/Documents/.mujoco/mujoco-py/mujoco_py/builder.py", line 121, in build_extensions
    build_ext.build_extensions(self)
  File "/home/ganesh/.local/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/command/build_ext.py", line 532, in build_extension
    depends=ext.depends)
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/ganesh/anaconda2/envs/mj/lib/python3.5/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'gcc' failed with exit status 1
```
Can someone help me with this?

ganyeshprasanna, 2018-04-09:
No worries. Solved it

closed 2018-04-09 by ganyeshprasanna

geweihgg, 2018-04-17:
Hi,
    I've encountered this problem too, but I can't find any solutions for this. Could you tell me how to solve it?
    Thank you very much.
                                                                                                                              GeWei

ganyeshprasanna, 2018-04-17:
What were the steps you followed?
Do you want to use it with Openaigym?

geweihgg, 2018-04-17:
Hi, 
    I want to install mujoco based on python3.6, then use it in within gym.
    I followed these steps:
        1). git clone mujoco-py
        2). cd mujoco-py
        3). pip install -e .
        4). import mujoco-py
    Then it threw out error almost the same as yours.
    By the way, I did these steps using miniconda.
    Thank you again.
                                                                                         GeWei

ganyeshprasanna, 2018-04-17:
I am not sure if it works with Python 3.6

ganyeshprasanna, 2018-04-17:
https://www.linkedin.com/pulse/setting-up-mujoco-ganesh-prasanna/

geweihgg, 2018-04-17:
Ok，^_^

geweihgg, 2018-04-18:
Hello~
  I've tried to install the mujoco-py based on miniconda + python3.5.4, and it works.
  Now I can 'import mujoco_py' successfully, but when I run the code below, it threw out some errors:
`>>> import mujoco_py
>>> env = gym.make('FetchPush-v1')
>>> env.reset()
{'observation': array([ 1.34193113e+00,  7.48903354e-01,  4.13631365e-01,  1.21495161e+00,
        6.41324519e-01,  4.24702091e-01, -1.26979513e-01, -1.07578836e-01,
        1.10707256e-02, -2.06467383e-06,  1.87039390e-03, -8.82449686e-08,
        1.35761490e-07,  3.97194063e-15, -2.96711930e-07, -5.50441164e-05,
        4.69134018e-05,  5.03907166e-08, -7.75241793e-08,  2.51278755e-18,
        2.94776838e-07,  5.50428586e-05, -8.56360696e-08,  5.26545712e-07,
        6.00978493e-05]), 'achieved_goal': array([1.21495161, 0.64132452, 0.42470209]), 'desired_goal': array([1.46108154, 0.67958309, 0.42469975])}
>>> for _ in range(1000):
...   env.render()
...   env.step(env.action_space.sample())
... 
Creating window glfw
ERROR: GLEW initalization error: Missing GL version
`
  I thought this error occurred because I didn't follow your instruction completely. 
  I googled this, and found that it's still on progress. And tartavull provided a solution that we could write our own viewer.
  Have you ever encountered the problem that 'env.render()' can't work well?
  Looking forward to your reply.
  Thank you!
  Gewei

geweihgg, 2018-04-18:
Hi, 
  I followed your instruction "export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-384/libGL.so ", and it works well.
  Thank you very very much~
  Gewei

ganyeshprasanna, 2018-04-18:
Hi,
Add it to your bash. You should be all set. 
No problem
-Ganesh

Baichenjia, 2018-05-08:
In the blog "https://www.linkedin.com/pulse/setting-up-mujoco-ganesh-prasanna/"
Say "PS: you should always be inside ~/mujoco-py folder to run the simulation"
Can you fix this path problem? Can you run simulaiton in any folder?

Baichenjia, 2018-05-08:
I have a solution, add the path in the begining of your code, like this
`import sys
sys.path.append("/home/chenjia/mujoco-py")
`
This work for me. But do you have other solution?

geweihgg, 2018-05-08:
@Baichenjia 
Maybe you can add it the the PYTHONPATH.

GeWei

Baichenjia, 2018-05-08:
@geweihgg 
it works~ thank you very much !

geweihgg, 2018-05-08:
@Baichenjia 
I'm very happy to hear that. ^-^

GeWei

zhouweiqing-star, 2022-10-21:
> Hi, I am not sure if it works with Python 3.6. There have been several issues regarding this. I used python 3.5.2. I have made a documentation as to how I installed it. Wait let me pull it out for you. Check this out. This worked for me and a bunch of my friends: https://www.linkedin.com/pulse/setting-up-mujoco-ganesh-prasanna/ -Ganesh

您好，您这个链接我尝试打开，但显示不存在，我最近也遇到了相似的问题：distutils.errors.CompileError: command 'C:\\Users\\86178\\vs2019\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

zhouweiqing-star, 2022-10-21:
> @白晨佳听到这个消息我很高兴。-
> 
> 葛伟

Could you please give me some guidance? I had a similar problem recently: distutils.errors.CompileError: command 'C:\\Users\\86178\\vs2019\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2, but I couldn't open the link you gave me. I operate under the window system

zhouweiqing-star, 2022-10-21:
> 别担心。解决了

Hello, you this link I tried to open, but the display does not exist, I have also encountered a similar problem recently：distutils.errors.CompileError: command 'C:\\Users\\86178\\vs2019\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

zhouweiqing-star, 2022-10-21:
> Could you please give me some guidance? I had a similar problem recently: distutils.errors.CompileError: command 'C:\Users\86178\vs2019\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe' failed with exit status 2, but I couldn't open the link you gave me. I operate under the window system

Could you please give me some guidance? I had a similar problem recently: distutils.errors.CompileError: command 'C:\Users\86178\vs2019\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe' failed with exit status 2, but I couldn't open the link you gave me. I operate under the window system

> @geweihgg很管用~非常感谢！

Could you please give me some guidance? I had a similar problem recently: distutils.errors.CompileError: command 'C:\Users\86178\vs2019\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe' failed with exit status 2, but I couldn't open the link you gave me. I operate under the window system