openai/mujoco-py #236: install  shell error sh: 2: [  messy code

zdx3578, 2018-04-07:
Downloading https://pypi.tuna.tsinghua.edu.cn/packages/07/a0/575ff0206a7fb5b18b662606ffbab46b72ea081b72c241b76be7ca83414e/mujoco-py-1.50.1.45.tar.gz (6.4MB)
    100% |████████████████████████████████| 6.4MB 3.3MB/s 
Requirement already satisfied: glfw>=1.4.0 in ./herenv35/lib/python3.5/site-packages (from mujoco-py) (1.6.0)
Requirement already satisfied: numpy>=1.11 in ./herenv35/lib/python3.5/site-packages (from mujoco-py) (1.14.2)
Requirement already satisfied: Cython>=0.27.2 in ./herenv35/lib/python3.5/site-packages (from mujoco-py) (0.28.1)
Requirement already satisfied: imageio>=2.1.2 in ./herenv35/lib/python3.5/site-packages (from mujoco-py) (2.3.0)
Requirement already satisfied: cffi>=1.10 in ./herenv35/lib/python3.5/site-packages (from mujoco-py) (1.11.5)
Requirement already satisfied: pillow in ./herenv35/lib/python3.5/site-packages (from imageio>=2.1.2->mujoco-py) (5.0.0)
Requirement already satisfied: pycparser in ./herenv35/lib/python3.5/site-packages (from cffi>=1.10->mujoco-py) (2.18)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /home/liu/git/baselines/herenv35/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-c8244cqn/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-rhd0s53a --python-tag cp35:
  running bdist_wheel
  running build
  [  [V: not found
  sh: 2: [
            : not found
  [  [V: not found
  sh: 2: [
            : not found
  [  [V: not found
  sh: 2: Syntax error: "(" unexpected
  [  [V: not found
  sh: 2: Syntax error: "(" unexpected
  ERROR: Invalid activation key
  
  Press Enter to exit ...
  
  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /home/liu/git/baselines/herenv35/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-c8244cqn/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-se44zoj4/install-record.txt --single-version-externally-managed --compile --install-headers /home/liu/git/baselines/herenv35/include/site/python3.5/mujoco-py:
    running install
    running build
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    ERROR: Invalid activation key
    
    Press Enter to exit ...
    
    ----------------------------------------
Command "/home/liu/git/baselines/herenv35/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-c8244cqn/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-se44zoj4/install-record.txt --single-version-externally-managed --compile --install-headers /home/liu/git/baselines/herenv35/include/site/python3.5/mujoco-py" failed with error code 1 in /tmp/pip-install-c8244cqn/mujoco-py/

--------------------------------------------------------------------------------------------------------------------------------------

  Running setup.py bdist_wheel for PyOpenGL ... done
  Stored in directory: /home/liu/.cache/pip/wheels/4c/d8/2c/fa93d519b6438a056cb8b764159b94f303a21a9f2100a2ee28
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /home/liu/git/baselines/herenv36/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-dqea8a02/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-kw914k4t --python-tag cp36:
  running bdist_wheel
  running build
  /home/liu/git/baselines/herenv36/lib/python3.6/importlib/_bootstrap.py:205: RuntimeWarning: compiletime version 3.5 of module 'mujoco_py.cymj' does not match runtime version 3.6
    return f(*args, **kwds)
  
  WARNING: Existing glfw python module detected!
  
  MuJoCo comes with its own version of GLFW, so it's preferable to use that one.
  
  The easy solution is to `import mujoco_py` _before_ `import glfw`.
  
  [  [V: not found
  sh: 2: [
            : not found
  [  [V: not found
  sh: 2: [
            : not found
  [  [V: not found
  sh: 2: Syntax error: "(" unexpected
  [  [V: not found
  sh: 2: Syntax error: "(" unexpected
  ERROR: Invalid activation key
  
  Press Enter to exit ...
  
  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
  Running setup.py bdist_wheel for gast ... done
  Stored in directory: /home/liu/.cache/pip/wheels/17/0a/dc/bb6d7b129029482a8d55901d66b65e756a681f6a1da7297a9b
  Running setup.py bdist_wheel for termcolor ... done
  Stored in directory: /home/liu/.cache/pip/wheels/65/c8/98/8361afe9bafba434b7acf14c08627560d63018272226ff3e10
  Running setup.py bdist_wheel for absl-py ... done
  Stored in directory: /home/liu/.cache/pip/wheels/a8/1c/ab/b14e1d334eb48e7d8f846d18fb55a2f636271830d0f8055d9b
  Running setup.py bdist_wheel for future ... done
  Stored in directory: /home/liu/.cache/pip/wheels/e9/52/ba/50c91c0a1a7e2ca55d109ad3083ca093553759b2cf0092267e
  Running setup.py bdist_wheel for glfw ... done
  Stored in directory: /home/liu/.cache/pip/wheels/3a/9c/bf/10928fd437e9cc10df1dd12b67d3d0c4730152c87755cb6a0d
  Running setup.py bdist_wheel for html5lib ... done
  Stored in directory: /home/liu/.cache/pip/wheels/e4/68/0c/85c19b1fbe67607477168d154bd1c20ba586fc23988e03818a
Successfully built gym zmq dill mpi4py atari-py PyOpenGL gast termcolor absl-py future glfw html5lib
Failed to build mujoco-py
Installing collected packages: numpy, certifi, urllib3, idna, chardet, requests, six, future, pyglet, atari-py, Pillow, PyOpenGL, glfw, Cython, imageio, pycparser, cffi, mujoco-py, gym, scipy, tqdm, joblib, pyzmq, zmq, dill, python-utils, progressbar2, mpi4py, cloudpickle, astor, protobuf, grpcio, gast, markdown, html5lib, bleach, werkzeug, tensorboard, termcolor, absl-py, tensorflow, click, baselines
  Running setup.py install for mujoco-py ... error
    Complete output from command /home/liu/git/baselines/herenv36/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-dqea8a02/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-w72lgh0x/install-record.txt --single-version-externally-managed --compile --install-headers /home/liu/git/baselines/herenv36/include/site/python3.6/mujoco-py:
    running install
    running build
    /home/liu/git/baselines/herenv36/lib/python3.6/importlib/_bootstrap.py:205: RuntimeWarning: compiletime version 3.5 of module 'mujoco_py.cymj' does not match runtime version 3.6
      return f(*args, **kwds)
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    ERROR: Invalid activation key
    
    Press Enter to exit ...
    
    ----------------------------------------
Command "/home/liu/git/baselines/herenv36/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-dqea8a02/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-w72lgh0x/install-record.txt --single-version-externally-managed --compile --install-headers /home/liu/git/baselines/herenv36/include/site/python3.6/mujoco-py" failed with error code 1 in /tmp/pip-install-dqea8a02/mujoco-py/
(herenv36) liu@kangaroo:~/git/baselines$ which python

agakshat, 2018-04-07:
I was getting the same error on doing a `pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'`
As a successful workaround, I cloned the `mujoco-py` repo, `cd` into it, and did a `python setup.py install`
Then opened the python3 interpreter, and ran `import mujoco_py`

The first time you run this, it will compile a lot of files, but should be successful. Next time onwards, it won't have to compile those files at all, and will import smoothly.

zdx3578, 2018-04-09:
yes , when using pip3 install -U 'mujoco-py<1.50.2,>=1.50.1' got that error,but clone mujoco-py repo, cd into it, and did a python setup.py install , it ok.  thanks @agakshat

closed 2018-04-09 by zdx3578