openai/mujoco-py #237: Headless GPU rendering with Docker image

amandlek, 2018-04-07:
I've been trying to get GPU rendering working by using the most recent Dockerfile on this repository, with no luck. I've attached a bash script (mujoco_py_docker.sh) and a test script (test1.py). The bash script is used to install some further dependencies (an NVIDIA driver, for example) and the test script is used to try and retrieve frames from a gym environment. 

I keep running into this issue:

[image]

The number of GPUs is completely screwed up, and OpenGL initialization fails. However, `nvidia-smi` works fine inside the docker. 

Steps to reproduce:

1. Clone the repository, copy mjkey.txt (the MuJoCo license file) into the directory, and run `nvidia-docker build -t dockjoco .`
2. Run `nvidia-docker run -it dockjoco /bin/bash`.
3. In a new terminal, transfer mujoco_py_docker.sh and test1.py into /root in the docker container. 
4. In the docker, run `bash mujoco_py_docker.sh`.
5. Run `echo 'export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-390' >> ~/.bashrc` followed by `source ~/.bashrc`.
6. Finally, run `python test1.py`.

ascane, 2018-12-19:
Same problem here. Have you found any solutions?
PS. I can't see your attachments.

amandlek, 2018-12-19:
I've managed to get headless GPU rendering working on Ubuntu 16.04, but not on the docker unfortunately. The trick for getting it to work on Ubuntu was to make sure mujoco-py could find the nvidia driver and also making sure that patchelf was installed.

richardrl, 2019-01-12:
@amandlek Why do they not include the nvidia driver in the sample docker? The driver is necessary for standard torch/keras GPU acceleration right?

BTickell, 2019-03-16:
What worked for me is to simply mount the host machine `nvidia-XXX` folder at `/usr/lib/nvidia-XXX` in the docker image and add it's location to `$LD_LIBRARY_PATH$`. Doing this enables headless rendering inside of the docker image. Note you will need to rebuild your mujoco_py package.

@richardrl it seems like nvidia-docker links the nvidia-drivers in a way that is not apparently accessible from inside a running container although perhaps someone has details on this.