openai/mujoco-py #238: import mujoco_py error

xionghhcs, 2018-04-09:
I have install mujoco-py in ubuntu16.04,the package is as follow:
``` shell
(rllab) zju@amax:~/.mujoco/mjpro150/bin$ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
absl-py (0.1.13)
astor (0.6.2)
atari-py (0.1.1)
baselines (0.1.5, /data2/home/zju/workspace/xionghaihui/rl/baselines)
bleach (1.5.0)
certifi (2018.1.18)
cffi (1.11.5)
chardet (3.0.4)
click (6.7)
cloudpickle (0.5.2)
Cython (0.28.1)
dill (0.2.7.1)
future (0.16.0)
gast (0.2.0)
glfw (1.6.0)
grpcio (1.10.1)
gym (0.10.5)
html5lib (0.9999999)
idna (2.6)
imageio (2.3.0)
joblib (0.11)
Markdown (2.6.11)
mpi4py (3.0.0)
mujoco-py (1.50.1.52, /data2/home/zju/.mujoco/mujoco-py)
numpy (1.14.2)
opencv-python (3.4.0.12)
Pillow (5.1.0)
pip (9.0.1)
progressbar2 (3.37.0)
protobuf (3.5.2.post1)
pycparser (2.18)
pyglet (1.3.1)
PyOpenGL (3.1.0)
python-utils (2.3.0)
pyzmq (17.0.0)
requests (2.18.4)
scipy (1.0.1)
setuptools (39.0.1)
six (1.11.0)
tensorboard (1.7.0)
tensorflow (1.7.0)
termcolor (1.1.0)
tqdm (4.21.0)
urllib3 (1.22)
Werkzeug (0.14.1)
wheel (0.29.0)
zmq (0.0.0)

```
But when I import the model, it shows the following error:
```
rllab) zju@amax:~/.mujoco/mjpro150/bin$ python
Python 3.5.4 |Continuum Analytics, Inc.| (default, Aug 14 2017, 13:26:58) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data2/home/zju/.mujoco/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/data2/home/zju/.mujoco/mujoco-py/mujoco_py/builder.py", line 452, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/data2/home/zju/.mujoco/mujoco-py/mujoco_py/builder.py", line 68, in load_cython_ext
    if os.getenv('MUJOCO_PY_FORCE_CPU') is None and get_nvidia_lib_dir() is not None:
  File "/data2/home/zju/.mujoco/mujoco-py/mujoco_py/builder.py", line 35, in get_nvidia_lib_dir
    "installed in the system. Found several: %s" % str(paths)
AssertionError: Expected only one version of nvidia to be installed in the system. Found several: ['/usr/lib/nvidia-375', '/usr/lib/nvidia-384']
>>> 

```
It seem that there is two drivers in my computer?how can I fix this problem??

wojzaremba, 2018-04-09:
please check the most recent version : 1.50.1.53

xionghhcs, 2018-04-11:
Yes.After I update to the latest version(1.50.1.53),this problem have beed solved.Thank you for @wojzaremba