openai/mujoco-py #244: I just upgrade the gym and mujoco, then my code can not use

wsgdrfz, 2018-04-14:
self.model = mujoco_py.load_model_from_path(fullpath)
AttributeError: 'module' object has no attribute 'load_model_from_path'

gym: 0.10.5
mujoco-py:0.5.7

wsgdrfz, 2018-04-14:
If I change .mujoco/mjpro131 to mjpro150, the Error is:

mujoco_py.error.MujocoDependencyError: Found your MuJoCo license key but not binaries. Please put your binaries into ~/.mujoco/mjpro131 or set MUJOCO_PY_MJPRO_PATH. Follow the instructions on https://github.com/openai/mujoco-py for setup.

how to set the  MUJOCO_PY_MJPRO_PATH?

wsgdrfz, 2018-04-14:
Okay, now I change the env.
Now:
 python 3.6, mujoco 0.5.7 with the mjpro131 package installed and gym 0.9.1

It works, but I still don't know why it doesn't work if I upgrade the gym and mujoco-py.

henrywood2018, 2018-05-31:
Hello, how to install gym 0.9.1? I mean, it will automatically install the newest version with 
```
git clone https://github.com/openai/gym.git
cd gym
pip install -e .
```

IcarusWizard, 2018-06-09:
@henrywood2018 
Hi, you can simply get gym 0.9.1 with pip. Try`pip install gym==0.9.1`

15980959042, 2018-07-05:
liduan@ubuntu18:~$ python3
Python 3.6.5 (default, Apr  1 2018, 05:46:30) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
>>> from os.path import dirname
>>> model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +"/xmls/claw.xml")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'mujoco_py' has no attribute 'load_model_from_path'
Hello～The problem is still here!
gym:0.9.1
mujoco:mjpro131
mujoco_py:0.5.7
Who can tell me how to solve it !....

Derekabc, 2019-04-15:
@LiDuan1111   Hi, do you know how to solve this problem?

Liliduan, 2019-04-16:
> @LiDuan1111   Hi, do you know how to solve this problem?

I changed the version of gym and mujoco_py since the CPU of my computer was old. Specific configuration is as below:

python: 3.6
Tensorflow:1.5.0
Gym: 0.9.1
mujoco: 131
mujoco_py:0.5.7

Good luck~

Derekabc, 2019-04-16:
By long term trail, I solved the problem.  By using gym=0.10.5, mujoco-py=2.0.2.1, tensorflow =1.13.1, and mujoco200. My Path looks like this, 

```
#export LD_LIBRARY_PATH=~/.mujoco/mjpro131/bin${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
export LD_LIBRARY_PATH=~/.mujoco/mjpro200/bin${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/dong/.mujoco/mujoco200/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-384
export MUJOCO_KEY_PATH=~/.mujoco${MUJOCO_KEY_PATH}
```

Using low version of gym, may cause prolem like: 
`ImportError cannot import name FlattenDictWrapper`, When I upgrade it from 0.9.1 to 0.10.5, problems gone.

Another Problem is `"AttributeError:module 'mujoco_py' has no attribute 'load_model_from_path''`, I solved it by using a higher version of mujoco-py and mujoco200 rather than mujoco131 and mujoco-py=0.5.7, and added the PATH. 

Hope helpful!

closed 2019-12-26 by wsgdrfz

han-x, 2020-08-30:
I got the same problem, and I just added mjpro131 in ~/.mujoco as well, since there could be different versions of mujoco at the same time under ~/.mujoco
Specific configuration is as below:

python: 3.6
Tensorflow:1.4.0
Gym: 0.10.6
mujoco: 131
mujoco_py:2.0.~