openai/mujoco-py #245: New robotics environment error

lpsantao, 2018-04-19:
I've tried running this code:
`import gym
env = gym.make('HandManipulateBlock-v0')
env.reset()
env.render()`

and the following error is generated:
```
File "wrappers.pxi", line 2528, in mujoco_py.cymj.PyMjData.get_joint_qpos
TypeError: 'numpy.int32' object is not iterable
```
this happens only with every new robotics environment. Any suggestions?
I'm using python 3.6.5 and windows 10

gabielchong, 2018-06-28:
same with me!

dotd, 2018-09-25:
Hello,

The same is with me!

Thanks,

Dotan

LiDuan1111, 2018-10-03:
The same with me,I am using python3.6.5 and windows7

aaronsnoswell, 2019-01-29:
+1 - getting the same error message, but when I run `MjSim.data.set_joint_qvel`. Error log below:

```
C:\Users\uqasnosw\AppData\Local\Continuum\Miniconda3\envs\python36\python.exe "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.1\helpers\pydev\pydevd.py" --multiproc --qt-support=auto --client 127.0.0.1 --port 63564 --file E:/Development/jitterbug/experiment.py
pydev debugger: process 18164 is connecting

Connected to pydev debugger (build 183.4886.43)
Creating window glfw
<built-in method set_joint_qvel of mujoco_py.cymj.PyMjData object at 0x0000024DEC8D9098>
Backend Qt5Agg is interactive backend. Turning interactive mode on.
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.1\helpers\pydev\pydevd.py", line 1741, in <module>
    main()
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.1\helpers\pydev\pydevd.py", line 1735, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.1\helpers\pydev\pydevd.py", line 1135, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.1\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "E:/Development/jitterbug/experiment.py", line 199, in <module>
    experiment.run()
  File "E:/Development/jitterbug/experiment.py", line 26, in run
    self.sim.data.set_joint_qvel('jointMass', 0)
  File "mujoco_py\generated/wrappers.pxi", line 2566, in mujoco_py.cymj.PyMjData.set_joint_qvel
    start_i, end_i = addr
TypeError: 'numpy.int32' object is not iterable
```

My platform is Windows 10, MuJoCo 1.5, mujoco-py 1.68. NB: I am able to install the library, compile correctly, and run the demo code on the README no problems, it is only when I call `set_joint_qvel` I get this error.

aaronsnoswell, 2019-01-29:
Possibly a duplicate of #324 - the fix there worked for me.

gkcgvn, 2019-03-13:
Hi, I modified the humanoidstandup-v2, and I would like to stop the episode when some of the conditions (like stepping) are satisfied. For this reason, I placed a done condition instead of built-in done= False inside step function. When the step is taken, the environment is giving assert not done (from mujoco_env.py) error. When I switched only done condition to false as in v2 everything is working. Stopping the episode when the step is taken is crucial for me. How can I modify done condition without assertion error? When I remove assert not done from mujoco_py the env is not working.