openai/mujoco-py #249: Access camera data from mujoco model

adarshsehgal, 2018-05-02:
I have a model of AUBO i5 in mujoco and I am using mujoco py with gym AI to implement reinforcement learning algorithm. I have put camera on the robot and I want to get images from the camera and access those in mujoco py so that I can process them. 

Any help would be highly appreciated

Thank you,
Adarsh Sehgal

gisely, 2018-08-30:
I would also like to know how to do this!

Thanks,
Guy Isely

kracon7, 2019-04-01:
Hi, @adarshsehgal @gisely 
have you guys figured this out yet? I ran into the same problem. 
Thanks in advance!

adarshsehgal, 2019-04-01:
Unfortunately, I was not able to figure this out.

kracon7, 2019-04-01:
@adarshsehgal 
I figured it out. 

```
model = load_model_from_path("test.xml")
sim = MjSim(model)

 ## a is a tuple if depth is True and a numpy array if depth is False ##
a = sim.render(width=200, height=200, camera_name='rgb', depth=True)
rgb_img = a[0]
depth_img = a[1]
```
It's in their api reference:
https://openai.github.io/mujoco-py/build/html/reference.html#mjsim-basic-simulation

adarshsehgal, 2019-04-01:
Great! Thank you so much!

quinnliu, 2021-01-12:
'rgb' is not a valid camera but you can use `('frontview', 'birdview', 'agentview', 'sideview', 'robot0_robotview', 'robot0_eye_in_hand')`

Engineering-Geek, 2022-06-09:
> 'rgb' is not a valid camera but you can use `('frontview', 'birdview', 'agentview', 'sideview', 'robot0_robotview', 'robot0_eye_in_hand')`

If you open the XML file for the robot being rendered, you can see the different camera names being used. In your case, you don't have a camera named 'rgb', but you have the names 'frontview', 'birdview', 'agentview', 'sideview', 'robot0_robotview', 'robot0_eye_in_hand'. Take for example my little xml file:
`<camera name="camera" pos="0 0 0" quat="0 0 0 1"/>`
In my case, the camera name is literally 'camera' itself. Check your XML file to see what it is