openai/mujoco-py #253: What is the step by step process to install mujoco-py in Windows 10 without installing Anaconda?

akshay-sophos, 2018-05-12:


iandanforth, 2018-07-28:
Here's what worked for me.

## Background

Mujoco must be installed, working and with a valid key. I'll assume you've done that bit.

Mujoco-py uses cythonize to compile code. For that to work you need a working set of build tools. You might already have these if you have Visual Studio installed. 

If you don't have visual studio installed (and don't want to install the giant IDE) you can [download just the build tools](https://visualstudio.microsoft.com/downloads/#build-tools-for-visual-studio-2017) and install them. (Note it will default to also installing the Windows SDK, but I didn't need that and it saved several GB of disk space by not including it during the install process)

Once that is installed you should have a new terminal installed called `x64 Native Tools Command Prompt for VS 2017`. You should be able to find and launch that from the windows menu.

Perform the rest of these commands in that window.

## Install

Clone this repository and move into the directory.

```
git clone git@github.com:openai/mujoco-py.git
cd mujoco-py
```

Make sure your setup tools are up to date

```
python -m pip install --upgrade setuptools
```

Install both regular and dev dependencies for mujoco-py

```
pip install -r requirements.txt
pip install -r requirements.dev.txt
```

Now install mujoco-py

```
python setup.py install
```

If this completes successfully you should be able to run the examples

```
python examples\body_interaction.py
```

## Troubleshooting

I had an old partially compiled version of mujoco_py in my pythonpath that was very confusing. I uninstalled mujoco-py and then had to *manually delete* the mujoco-py directory from `c:\users\<USER>\appdata\local\programs\python\python36\lib\site-packages`

I was then able to follow the above.

Mis-configured versions of Visual Studio may also cause problems as would an out-of-date setuptools.

jesdefer, 2018-08-22:
@iandanforth  here you can find here the command window output:
`   
C:\Users\USUARIO\Dropbox\TFM\Mujoco\mujoco-py>python examples\body_interaction.py
Traceback (most recent call last):
  File "examples\body_interaction.py", line 10, in <module>
    from mujoco_py import load_model_from_xml, MjSim, MjViewer
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, Mujoc
oException
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 78, in load_cython_ext
    "set %s=%s;%%%s%%" % (var, lib_path, var))
Exception: Please add mujoco library to your PATH:
set PATH=C:\Users\USUARIO\.mujoco\mjpro150\bin;%PATH%

C:\Users\USUARIO\Dropbox\TFM\Mujoco\mujoco-py>SET PATH=%userprofile%\.mujoco\mjp
ro150\bin;%PATH%;

C:\Users\USUARIO\Dropbox\TFM\Mujoco\mujoco-py>python examples\body_interaction.p
y
running build_ext
building 'mujoco_py.cymj' extension
C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.
15.26726\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Python\
Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mujoco_py -IC:\Users
\USUARIO\.mujoco\mjpro150\include -IC:\Python\Python36-32\lib\site-packages\nump
y\core\include -IC:\Python\Python36-32\include -IC:\Python\Python36-32\include "
-IC:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\1
4.15.26726\include" /TcC:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.
56-py3.6.egg\mujoco_py\cymj.c /FoC:\Python\Python36-32\lib\site-packages\mujoco_
py-1.50.1.56-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.56_36_windowsextension
builder\temp.win32-3.6\Release\Python\Python36-32\lib\site-packages\mujoco_py-1.
50.1.56-py3.6.egg\mujoco_py\cymj.obj -fopenmp -w
cl : Línea de comandos warning D9025 : invalidando '/W3' con '/w'
cl : Línea de comandos warning D9002 : se omite la opción desconocida '-fopenmp'

cymj.c
c:\python\python36-32\include\pyconfig.h(59): fatal error C1083: No se puede abr
ir el archivo incluir: 'io.h': No such file or directory
Traceback (most recent call last):
  File "C:\Python\Python36-32\lib\distutils\_msvccompiler.py", line 423, in comp
ile
    self.spawn(args)
  File "C:\Python\Python36-32\lib\distutils\_msvccompiler.py", line 542, in spaw
n
    return super().spawn(cmd)
  File "C:\Python\Python36-32\lib\distutils\ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "C:\Python\Python36-32\lib\distutils\spawn.py", line 38, in spawn
    _spawn_nt(cmd, search_path, dry_run=dry_run)
  File "C:\Python\Python36-32\lib\distutils\spawn.py", line 81, in _spawn_nt
    "command %r failed with exit status %d" % (cmd, rc))
distutils.errors.DistutilsExecError: command 'C:\\Program Files (x86)\\Microsoft
 Visual Studio\\2017\\BuildTools\\VC\\Tools\\MSVC\\14.15.26726\\bin\\HostX86\\x8
6\\cl.exe' failed with exit status 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples\body_interaction.py", line 10, in <module>
    from mujoco_py import load_model_from_xml, MjSim, MjViewer
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, Mujoc
oException
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 90, in load_cython_ext
    cext_so_path = builder.build()
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 202, in build
    built_so_file_path = self._build_impl()
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 225, in _build_impl
    dist.run_commands()
  File "C:\Python\Python36-32\lib\distutils\dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "C:\Python\Python36-32\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "C:\Python\Python36-32\lib\site-packages\Cython\Distutils\old_build_ext.p
y", line 186, in run
    _build_ext.build_ext.run(self)
  File "C:\Python\Python36-32\lib\distutils\command\build_ext.py", line 339, in
run
    self.build_extensions()
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 125, in build_extensions
    build_ext.build_extensions(self)
  File "C:\Python\Python36-32\lib\site-packages\Cython\Distutils\old_build_ext.p
y", line 194, in build_extensions
    self.build_extension(ext)
  File "C:\Python\Python36-32\lib\distutils\command\build_ext.py", line 533, in
build_extension
    depends=ext.depends)
  File "C:\Python\Python36-32\lib\distutils\_msvccompiler.py", line 425, in comp
ile
    raise CompileError(msg)
distutils.errors.CompileError: command 'C:\\Program Files (x86)\\Microsoft Visua
l Studio\\2017\\BuildTools\\VC\\Tools\\MSVC\\14.15.26726\\bin\\HostX86\\x86\\cl.
exe' failed with exit status 2
`

RedTachyon, 2018-10-09:
I'm having the same problem. Did you find a way to fix it?

wotmd5731, 2018-10-16:
I'm having that problem too!!!

gtatiya, 2018-12-11:
Follow these steps to install on Windows:

`mkdir   %userprofile%\.mujoco`

`extract   mjpro150_win64.zip in %userprofile%\.mujoco`

`Copy mjkey.txt   %userprofile%\.mujoco\mjkey.txt`
`Copy mjkey.txt   %userprofile%\.mujoco\mjpro150\bin\mjkey.txt`

Activate your Python env in which you want to install.

`SET   PATH=%userprofile%\.mujoco\mjpro150\bin;%PATH%;`

Install Microsoft   Visual C++ Build Tools 2015: https://download.microsoft.com/download/5/f/7/5f7acaeb-8363-451f-9425-68a90f98b238/visualcppbuildtools_full.exe?fixForIE=.exe

Check Windows 8.1   SDK and Windows 10 SDK options

Use Visual C++ 2015   x64 Native Build Tools Command Prompt in C:\Program Files (x86)\Microsoft   Visual C++ Build Tools - https://stackoverflow.com/a/45918595/7448860

`cd   "C:\Users\USER\Downloads\Compressed\mujoco-py-master"`

`pip install -r   requirements.txt`
`pip install -r   requirements.dev.txt`
`python setup.py   install`
  
`import mujoco_py`

Enjoy!! :)

vaisakm, 2019-11-06:
> Follow these steps to install on Windows:
> 
Hi I have followed this method But when I run import mujoco_py, I get the following error.
>>> import mujoco_py

You appear to be missing MuJoCo.  We expected to find the file here: C:\Users\Vaisak Mohan\.mujoco\mujoco200

This package only provides python bindings, the library must be installed separately.

Please follow the instructions on the README to install MuJoCo

    https://github.com/openai/mujoco-py#install-mujoco

Which can be downloaded from the website

    https://www.roboti.us/index.html

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\............\OpenAI Simulation\mujoco setup\mujoco-py-master\mujoco_py\__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "C:\Users\...............\OpenAI Simulation\mujoco setup\mujoco-py-master\mujoco_py\builder.py", line 505, in <module>
    mujoco_path, key_path = discover_mujoco()
  File "C:\Users\.................\OpenAI Simulation\mujoco setup\mujoco-py-master\mujoco_py\utils.py", line 93, in discover_mujoco
    raise Exception(message)
Exception:
You appear to be missing MuJoCo.  We expected to find the file here: C:\Users\Vaisak Mohan\.mujoco\mujoco200

vaisakm, 2019-11-06:
Two things I did different:
Activate your Python env in which you want to install.   => I'm using my main environment. Its okay for me.

SET PATH=%userprofile%\.mujoco\mjpro150\bin;%PATH%; => I have added this directory in the windows environment variable PATH.

Rub22, 2020-05-11:
> @iandanforth here you can find here the command window output:
>
> ```
> C:\Users\USUARIO\Dropbox\TFM\Mujoco\mujoco-py>python examples\body_interaction.py
> Traceback (most recent call last):
>   File [...]

I have the same error. I've tried the process many times, including the ones indicated in the comments. Does anybody nknow how can I solve it?

ppan99, 2020-12-20:
> Here's what worked for me.
>
> ## Background
> Mujoco must be installed, working and with a valid key. I'll assume you've done that bit.
>
> Mujoco-py uses cythonize to compile code. For that to work you [...]

hi! when i type "pip install -r requirements.txt" this command, it appears "ERROR: Could not open requirements file:". so i find another approach. "pip freeze > requirements.txt" should be first typed. however, the another error "refuse to access" appears.
i've changed the permission of the directory.
Does anybody know how can I solve it? need help

kapilrathore42, 2021-03-30:
I am  having the same problem.

kapilrathore42, 2021-04-14:
i am having same problem

kentwhf, 2021-05-22:
> I have the same error. I've tried the process many times, including the ones indicated in the comments. Does anybody nknow how can I solve it?

Hello did you figure that out?

reda2097, 2021-06-07:
Hello, while trying to execute "python examples\body_interaction.py" i have the following problem:
raise DistutilsFileError(
distutils.errors.DistutilsFileError: could not create 'C:\Users\dz_re\Anaconda3\envs\DQN\lib\site-packages\mujoco_py-1.50.1.68-py3.8.egg\mujoco_py\generated\_pyxbld_1.50.1.68_38_windowsextensionbuilder\temp.win-amd64-3.8\Release\Users\dz_re\Anaconda3\envs\DQN\lib\site-packages\mujoco_py-1.50.1.68-py3.8.egg': file name or extention too long

please do you have any idea how to solve it

csyhy1986, 2021-09-15:
> Hello, while trying to execute "python examples\body_interaction.py" i have the following problem:
> raise DistutilsFileError(
> distutils.errors.DistutilsFileError: could not create [...]

It is still a problem for me and have not been solved for now yet.

MotorCityCobra, 2021-11-12:
> git clone git@github.com:openai/mujoco-py.git

This is not how you clone this repo.

JohannesBA2019, 2022-01-25:
> It is still a problem for me and have not been solved for now yet.

Have you been able to find a solution. I am stuck here aswell.

csyhy1986, 2022-01-26:
> Have you been able to find a solution. I am stuck here aswell.

No, it is still a problem.

aymenboudhina, 2022-03-25:
I had a problem "ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'" with
 pip install -r requirements.txt
pip install -r requirements.dev.txt
Can someone please tell me my mistake?

djh-1997, 2022-04-09:
> n find here the command window output:

You can copy the mujoco_py file(mine is under C:\Users\19240\.mujoco\mujoco-py) to the site-packages(mine is C:\Users\19240\Anaconda3\envs\mujoco\Lib\site-packages), it is useful in my computer, I hope it can help you.

djh-1997, 2022-04-09:
> Have you been able to find a solution. I am stuck here aswell.
You can copy the mujoco_py file(mine is under C:\Users\19240\.mujoco\mujoco-py) to the site-packages(mine is C:\Users\19240\Anaconda3\envs\mujoco\Lib\site-packages), it is useful in my computer, I hope it can help you.