openai/mujoco-py #254: Mujoco_py support for latest GCC compiler

JulianYG, 2018-05-12:
In mujoco_py/builder.py  line 302, in _build_impl 'Could not find GCC 6 or GCC 7 executable.\n\n'
RuntimeError: Could not find GCC 6 or GCC 7 executable.
Since I've updated my gcc to 8.1.0.

TCtobychen, 2018-05-28:
same issue

cassanolucas, 2018-05-29:
Same issue. Any known solutions?

TeaPearce, 2018-05-30:
snap

JulianYG, 2018-06-04:
I solved it by copying out the error commands and manually forcing it to use another gcc 7 compiler. However, it would be great if we can have gcc 8 support, thanks!

iandanforth, 2018-07-11:
Another work around on OSX is to use brew to install another version of GCC

`brew install gcc@6`

This worked for me with other versions installed as well. Note gcc@7 did not work immediately so I went with 6.

arpit15, 2018-08-14:
To add support for gcc 8 change [this](url
https://github.com/openai/mujoco-py/blob/master/mujoco_py/builder.py#L294) line to
```python
c_compilers = ['/usr/local/bin/gcc-6', '/usr/local/bin/gcc-7','/usr/local/bin/gcc-8']
``` 
@JulianYG @TCtobychen @cassanolucas @TeaPearce

pavyedav, 2018-08-23:
At @arpit15 , what were the steps you took after adding gcc-8 to c_compilers? I still get the error: 

`>>> import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/usr/local/lib/python3.6/site-packages/mujoco_py/builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/usr/local/lib/python3.6/site-packages/mujoco_py/builder.py", line 90, in load_cython_ext
    cext_so_path = builder.build()
  File "/usr/local/lib/python3.6/site-packages/mujoco_py/builder.py", line 202, in build
    built_so_file_path = self._build_impl()
  File "/usr/local/lib/python3.6/site-packages/mujoco_py/builder.py", line 302, in _build_impl
    'Could not find GCC 6 or GCC 7 executable.\n\n'
RuntimeError: Could not find GCC 6 or GCC 7 executable.`

Seems like I need to do something (like a compile) after saving that change.

phaniram-sayapaneni, 2018-09-12:
brew install gcc@6
or
brew install gcc@7

pip uninstall mujoco_py
pip install mujoco_py

Did not work for me, any suggestions please? I get the following error:
istutils.errors.CompileError: command '/usr/local/bin/gcc-6' failed with exit status 1

phaniram-sayapaneni, 2018-09-12:
Also tried to set the compilers to gcc-8
Still getting the error:
CompileError: command '/usr/local/bin/gcc-8' failed with exit status 1

machinaut, 2018-10-24:
Merged a fix to this, latest mujoco-py works with latest homebrew GCC!

closed 2018-10-24 by machinaut

phaniram-sayapaneni, 2018-10-24:
Great!

Thanks a lot!!

ali-chaudhry8, 2019-05-18:
Hi, I am getting this error: 
"distutils.errors.CompileError: command '/usr/local/bin/gcc-6' failed with exit status 1"
Tried re installing Mujoco, installed GCC7 and 8, but this issue persists. Please help...

marjanin, 2019-06-17:
> Hi, I am getting this error:
> "distutils.errors.CompileError: command '/usr/local/bin/gcc-6' failed with exit status 1"
> Tried re installing Mujoco, installed GCC7 and 8, but this issue persists. Please help...

Hi,
I am getting the same error on Mac. Did you find a solution?

marjanin, 2019-06-17:
It worked once I installed the pkg mentioned here:
https://stackoverflow.com/questions/51274905/brew-install-on-mojave

dmalpica, 2019-08-08:
It's August 2019. Had this issue with gcc-9

brew install gcc@8 worked for me

newuhe, 2019-11-22:
> To add support for gcc 8 change [this](https://github.com/openai/mujoco-py/blob/master/mujoco_py/builder.py#L294) line to
> 
> ```python
> c_compilers = ['/usr/local/bin/gcc-6', '/usr/local/bin/gcc-7','/usr/local/bin/gcc-8']
> ```
> 
> @JulianYG @TCtobychen @cassanolucas @TeaPearce

this works for me, gcc9

emimarz, 2019-12-17:
same issue gcc 9
RuntimeError: Could not find GCC executable.

imrancharania, 2020-07-30:
Using gcc 8 seems to have resolved this issue for me.

wookayin, 2021-12-07:
Another workaround (perhaps most the easiest solution) is to use `CC` environment variable. (see #651 as well)

```
export CC=/usr/local/bin/gcc-11 
python -c "import mujoco_py"     # to initiate Cython build_ext
```