openai/mujoco-py #255: What does attribute 'xvelp' stand for?

liuyuezhangadam, 2018-05-13:
*Currently I'm using mujoco-py to build a robot simulation environment for Baxter. I have some problem in understanding the attribute 'xvelp' and failed to solve it in documentation. 

*For example, if I'd like to get the velocity of the site 'right_gripper', I may write like this:

'''python
def get_grip_vel(sim):
    dt = sim.nsubsteps * sim.model.opt.timestep
    grip_velp = sim.data.get_site_xvelp('right_gripper') * dt
    return grip_velp
'''

So what does the 'xvelp' stand for? Why you have to times it with 'dt'? Does it stand for acceleration?

I'm new to mujoco_py and OpenAI Gym. Thanks for your answers.

JoyChopra1298, 2018-08-08:
@liuyuezhangadam Did you decode it?

JoyChopra1298, 2018-08-08:
https://github.com/openai/mujoco-py/blob/cd247d5fae39f21e913104350e26ac7254fa4b93/mujoco_py/tests/test_cymj.py#L442

According to comment on this line it means positional(cartesian) velocity in world frame whereas xvelr is rotational velocity

Don't know why multiplication by dt is needed

liuyuezhangadam, 2018-08-11:
Sorry for a late reply. 

No, I didn't decode it and I move to work on Atari game this summer. Thank you for providing me a great clue! 

I shall continue to work on the project in this autumn. I'll  keep in track.

valeriechen, 2018-10-14:
Hi,

Where did the get_grip_vel function come from? Also do you mind sharing what versions you have for mujoco_py and gym? I am trying to get Baxter set up as well.

YeeCY, 2019-12-30:
Just a guess (may not be correct): 

`nsubsteps > 1` means mujoco takes more than one step when you call the step function in the env. Therefore, we need to scale the positional velocity in the world frame returned by mujoco to get corresponding velocity in the env. 

By the way, the `test_xvelp` function test exactly the velocity in mujoco.

closed 2020-01-04 by liuyuezhang

wangjunyi9999, 2023-03-10:
Well..., so the issue answer is what @YeeCY  said before?