openai/mujoco-py #256: Ubuntu16.04 LTS gets stuck in login interface after installing mujoco-py

supershania, 2018-05-17:
My mouse and keyboard didn't response when entering the log in interface.

My laptop is ThinkPad T470, Ubuntu 16.04 LTS (with windows), got the kernel 4.13.0-41-generic,4.13.0-36-generic, Graphics card Nividia GeForce 940 MX.

After installed ubuntu 16.04, I installed Cuda 9.2 following the official installation guidance, and the example works well. After that, I installed the Anaconda3, also I tried the example successfully. After this, I created the conda create --name gymlab pthon=3.5. Then, I installed mujoco 150pro and run the example successfully. My laptop can reboot successfully every time so far by now. After that, I installed the mujoco_py and run the example as well, it works well. But when I reboot this time, the ubuntu just freezed at the login interface, and my mouse and keyboard doesn't work at all. I can't even try ctrl+alt+F1.

How can I fix this? I want to install the openAI gym and mujoco.

yangyou95, 2018-08-20:
I have encountered the same problem and can't resolve it, I think it's related to the "patchelf" or some others packages installation which will cause a conflict with the nvidia driver.

yangyou95, 2018-08-20:
I fixed it! Just don't install Xorg-server-dev and xorg-dev.

WoodenJin, 2018-11-24:
Sir, I did the same thing, apt-get remove xorn-dev, and apt-get remove Xorg-server-dev, But it doesn't work. I think the package name is wrong?

machinaut, 2018-11-28:
closing this for now

closed 2018-11-28 by machinaut

fjh1997, 2020-03-29:
sudo apt install xserver-xorg-input-all ，I fix it by this.