openai/mujoco-py #261: module 'mujoco_py' has no attribute 'load_model_from_path'

henrywood2018, 2018-05-31:
Hello everyone,
have you ever met the problem " module 'mujoco_py' has no attribute 'load_model_from_path' ".
When I installed the **MuJoCo 131**, **mujoco_py 0.5.7** and **gym** and tried to test the installation, 

```
$ python3
import mujoco_py
from os.path import dirname
model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +"/xmls/claw.xml")
sim = mujoco_py.MjSim(model)
```
I got the prompt:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'mujoco_py' has no attribute 'load_model_from_path'
```
My python version is 3.5.5.
How can I solve the problem? Thanks!

dineshresearch, 2018-06-06:
same problem

IcarusWizard, 2018-06-09:
Hi guys, I think our problem can be solved with install gym 0.9.1. 
See #244

15980959042, 2018-07-06:
@IcarusWizard  I have the same problem,and I have changed the version of gym to 0.9.1,but the problem is still here....

15980959042, 2018-07-09:
I found that the following configuration can make it work:
mujoco：131
mujoco_py:0.5.7
tensorflow:1.5.0
gym:0.9.1
python3.6
but you may need to use the order version of environment models in mujoco platform.
(eg. you have to use 'InvertedPendulum-v1' in stead of the higher version'InvertedPendulum-v2')
Good luck~

jp18813100494, 2018-07-17:
It works when I install gym==0.9.1!But some problems occur that some gym environments are not in use.

machinaut, 2018-10-18:
@hangwudy i think the issue you have is that mujoco-py is out of date.

Do you get the same error with the latest mujoco-py?

HareshMiriyala, 2020-05-20:
installing gym 0.9.1 fixes this issue

```python -m pip install gym==0.9.1```