openai/mujoco-py #263: Any plan to support temporary vfs?

llan-ml, 2018-06-16:
Currently, we have to load model from xml file on the disk. Any plan to support load model from xml string in the memory, like https://github.com/deepmind/dm_control/blob/master/dm_control/mujoco/wrapper/core.py#L228?

machinaut, 2018-10-18:
I am curious for the use cases for this.

Right now we have found using a temporary file for XML is both useful for debugging and fast enough for us.

llan-ml, 2018-10-18:
@machinaut During the training of RL algorithms, we may want to change the environment dynamically. Take a maze for example, we need to generate a new maze randomly in each iteration or every several iterations.

Currently, the solution is to first open a template xml file, modify the configuration, and write it into the disk. However, after a lot of iterations, the system will raise an error that a single process cannot open more files.

machinaut, 2018-10-18:
I think a pull request adding VFS support would be welcome.

do you have an example file which reproduces the file error?  I have not yet seen it (and we also have many machines which make many environments) but maybe the error would be more informative.