openai/mujoco-py #265: Import mujoco_py fails after installation - issue with cymj

LiterallyUniqueLogin, 2018-06-28:
```
$ python>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py -I/home/jonathan/.mujoco/mjpro150/include -I/home/jonathan/pythonEnvs/MLpy36/lib/python3.6/site-packages/numpy/core/include -I/home/jonathan/pythonEnvs/MLpy36/include -I/usr/include/python3.6m -c /home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/cymj.c -o /home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/cymj.c:52:20: fatal error: Python.h: No such file or directory
compilation terminated.
Traceback (most recent call last):
  File "/usr/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/usr/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/usr/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/usr/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/builder.py", line 90, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/builder.py", line 202, in build
    built_so_file_path = self._build_impl()
  File "/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/builder.py", line 256, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/builder.py", line 225, in _build_impl
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/jonathan/pythonEnvs/MLpy36/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/jonathan/pythonEnvs/MLpy36/temp/mujoco-py/mujoco_py/builder.py", line 125, in build_extensions
    build_ext.build_extensions(self)
  File "/home/jonathan/pythonEnvs/MLpy36/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/usr/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/usr/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

Not sure why this is happening. Any advice?

Environment details:
Running on Ubuntu 16.04
$ python --version
Python 3.6.6
$ pip --version
pip 10.0.1 from /home/jonathan/pythonEnvs/MLpy36/lib/python3.6/site-packages/pip (python 3.6)

I have tried installing mujoco_py in each of these ways, doesn't seem to make a difference:
pip install mujoco_py
Checkout mujoco_py repository and pip install -e .
Checkout mujoco_py repository and python setup.py install

Mujoco works fine on my machine.
- Install located in the ~/.mujoco directory
- ./simulate ../model/humanoid.xml works fine

I've looked at all the other github issues, none seem to resolve this problem.

Any help appreciated. Thanks!

Jonathan

LiterallyUniqueLogin, 2018-06-29:
Hey all, fixed the problem.

I hadn't installed python3.6-dev, which meant importing Python.h didn't work. I did have python3-dev installed, but since that was python3.5 and I was using a python3.6 virtual environment, that didn't matter.

To the mujoco_py team - can you please call out the dependency on python3.6-dev in an explicit manner? And can you make it clear when these compilations fail what's actually gone wrong (compiling code upon module import instead of upon first installation, and the double stack traces alternating between python code and executables and c code, make debugging unnecessarily difficult).

Thanks!

closed 2018-06-29 by LiterallyUniqueLogin

KonkimallaChandraPrakash, 2018-11-11:
running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder
  creating /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6
  creating /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/tmp
  creating /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-pnrq0k29
  creating /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-pnrq0k29/mujoco-py
  creating /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py
  creating /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/gl
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py -I/home/chandu/.mujoco/mjpro150/include -I/home/chandu/Desktop/venvs/openai/lib/python3.6/site-packages/numpy/core/include -I/tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/vendor/egl -I/usr/include/python3.6m -I/home/chandu/Desktop/venvs/openai/include/python3.6m -c /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/cymj.c:52:20: fatal error: Python.h: No such file or directory
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /home/chandu/Desktop/venvs/openai/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-pnrq0k29/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-3g4vpaa9/install-record.txt --single-version-externally-managed --compile --install-headers /home/chandu/Desktop/venvs/openai/include/site/python3.6/mujoco-py:
    running install
    running build
    Removing old mujoco_py cext /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/cymj_1.50.1.68_36_linuxgpuextensionbuilder_36.so
    running build_ext
    building 'mujoco_py.cymj' extension
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py -I/home/chandu/.mujoco/mjpro150/include -I/home/chandu/Desktop/venvs/openai/lib/python3.6/site-packages/numpy/core/include -I/tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/vendor/egl -I/usr/include/python3.6m -I/home/chandu/Desktop/venvs/openai/include/python3.6m -c /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    /tmp/pip-install-pnrq0k29/mujoco-py/mujoco_py/cymj.c:52:20: fatal error: Python.h: No such file or directory
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/home/chandu/Desktop/venvs/openai/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-pnrq0k29/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-3g4vpaa9/install-record.txt --single-version-externally-managed --compile --install-headers /home/chandu/Desktop/venvs/openai/include/site/python3.6/mujoco-py" failed with error code 1 in /tmp/pip-install-pnrq0k29/mujoco-py/

The above way didn't solve the problem for me