openai/mujoco-py #267: LD_LIBRARY_PATH Error

shubhomb, 2018-07-03:
Hello.
I repeatedly get the following error installing Mujoco on Ubuntu 18.04 LTS (Bionic Beaver): 

Please add following line to .bashrc
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/shubhom/.mujoco/mjpro150/bin

however, my .bashrc file is updated with that and i f I echo $LD_LIBRARY_PATH I receive 
:/home/shubhom/.mujoco/mjpro50/bin

zhujiapeng, 2018-07-04:
same question

huangjiancong1, 2018-07-15:
me too, i have add the true path of `mjpro150/bin` in the `.bashrc` flie, but when i `import mujoco_py`, it output error 

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.56-py3.5.egg/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.56-py3.5.egg/mujoco_py/builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.56-py3.5.egg/mujoco_py/builder.py", line 68, in load_cython_ext
    _ensure_set_env_var("LD_LIBRARY_PATH", lib_path)
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.56-py3.5.egg/mujoco_py/builder.py", line 101, in _ensure_set_env_var
    var_name, var_name, lib_path))
Exception: 
Missing path to your environment variable. 
Current values LD_LIBRARY_PATH=
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/huang/.mujoco/mjpro150/bin
```

zwfcrazy, 2018-07-25:
Did you guys run python with sudo? If this is true, sudo will change your current user settings to root settings, including environment variables.

shubhomb, 2018-07-27:
I did that and it did not change the error. Any other suggestions?

zwfcrazy, 2018-07-27:
What I meant was that you should not use sudo.
If you look at the source code, it is actually this line of code failed to get the correct LD_LIBRARY_PATH.
```python
os.environ.get("LD_LIBRARY_PATH", "")
```
You may run this code elsewhere to see if it gives you the correct LD_LIBRARY_PATH.

What I found is the following,
When you add "export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/shubhom/.mujoco/mjpro150/bin" to .bashrc, this environment variable is only valid to the current user. However, if you run the python script using sudo, the user is changed to the root user and the LD_LIBRARY_PATH of the root user is empty.

morison1, 2018-07-31:
Same issue... someone for the rescue ?

iSarmad, 2018-08-19:
same issue, anyone who found the fix?

iSarmad, 2018-08-19:
So here is something that worked for me:
1. Do not use Sudo while opening your Python/ IDE, as LIBRARY_LD_PATH is empty for sudo user.
2. Instead always open your Python editor (PyCharm in my case) with ./pycharm.sh command without sudo. 
3. Here is the key, even after updating bashrc, always close all terminals/ IDE and open again before running your IDE again. This should be done as even after updating bashrc, the IDE reads the old bashrc file which does not even exist anymore.

shubhomb, 2018-08-22:
It's still not working for me. Anybody have another idea?

coolKeen, 2018-09-03:
I solve it by installing mujoco_py without using sudo, try this:
sudo pip3 uninstall mujoco_py
pip3 install mujoco-py==1.50.1.56

LiuWenzhang, 2018-09-23:
I have got the same problem when I did "import mujoco_py" in pycharm. I solved this problem by reinstall my Pycharm on linux.

ssghost, 2018-10-29:
same question, no solution yet.

machinaut, 2018-10-29:
Here's an example dockerfile installing mujoco-py and gym on ubuntu 18.04 https://gist.github.com/machinaut/b5a7c6719a30e9fbfd95c13d5e963b61

Without more information I don't know how to help with this error.

The python process you import mujoco-py in needs to have the environment variable (LD_LIBRARY_PATH) set to include the folder with the mujoco binaries.

In order to avoid crashes with difficult-to-understand bugs, we explicitly check for it before importing, and then print a warning with instructions useful to beginner users on how to fix it.

smallguoan, 2018-11-26:
same problem, but I solve it by opening IDE(Pycharm) in terminal.
In terminal:
pycharm2018.3/bin/pycharm.sh  

But I still have error by opening my IDE with icon on desktop

zhaolongkzz, 2018-11-27:
I think the answer of LiuWenzhang is right, maybe pycharm can not recognize the pip from the 3rd party without installing on pycharm itself.

machinaut, 2018-11-28:
This issue has a bunch of different concerns, many from old versions.

Closing for now.  If you see this error, please open a new issue!  We now have a template that helps us get the information we need!

Thanks,
~Alex

closed 2018-11-28 by machinaut

ZhengLiangliang1996, 2019-02-19:
Same issue here using spyder, and I solved it by open it through Terminal.

ZhengLiangliang1996, 2019-02-19:
Same issue here using spyder, and I solved it by opening it through Terminal.

hzm2016, 2019-11-05:
I solve this issue, by adding "source ~/.bashrc" to ".profile"

NoobHua, 2019-12-16:
1. this problem arise when you run it on terminal. 
you should check your ~/.bashrc

export LD_LIBRARY_PATH=~/.mujoco/mjpro150/bin${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
export MUJOCO_KEY_PATH=~/.mujoco${MUJOCO_KEY_PATH}

and "source ~/.bashrc"
2. this problem arise when you run it on PyCharm.
you should check if it can run successfully on terminal at first.
if ok on terminal, then:
 click Run -> Edit Configurations -> Environment Variables
Add 
LD_LIBRARY_PATH  /path/to/.mujoco/mjpro150/bin
MUJOCO_KEY_PATH /path/to/.mujoco

seyeeet, 2020-08-27:
@hzm2016  > I solve this issue, by adding "source ~/.bashrc" to ".profile"

how

rfali, 2021-08-04:
@seyeeet by 

> and "source ~/.bashrc"
@NoobHua meant to run this on Terminal (command line)

See [this](https://stackoverflow.com/questions/49083789/how-to-add-new-line-in-bashrc-file-in-ubuntu)

>

rfali, 2021-08-04:
SpinningUp Installation Handbook:
OS: Ubuntu 18.04.5 LTS

From https://spinningup.openai.com/en/latest/user/installation.html

Create conda env and activate
conda create -n spinningup python=3.6
conda activate spinningup

Install OpenMPI
sudo apt-get update && sudo apt-get install libopenmpi-dev

Install Spinning Up
git clone https://github.com/openai/spinningup.git
cd spinningup
pip install -e .

Check Your Install
https://spinningup.openai.com/en/latest/user/installation.html#check-your-install 
	
Installing MuJoCo

Obtain License from https://www.roboti.us/license.html For Aug-Oct 2021, there is free Activation Key (mkjey.txt) available on the above mentioned link.

If you don’t get a free activation, then first you need to register for a Free Trial (30-days) or a Student License (1-year). You will receive an Account Number in the email. Go to https://www.roboti.us/license.html and generate a Computer ID by downloading getid executable pertaining to your OS (Win32, Win64, Linux, OSX). Change the permissions for getid_linux.exe and run it to generate the Computer ID. Enter the Account Number and Computer ID on the same page and hit “Register Computer”. You should get an activation key (mjkey.txt) in your email.

Important: SpinningUp requires Mujoco 1.5 and not 2.0. SpinningUp Instructions asks you to go to https://github.com/openai/mujoco-py to follow instructions, which in turn ask to install the latest 2.0. Here is what you do instead:

Download the zip mjpro150_linux.zip from https://www.roboti.us/index.html

Unzip/Extract this folder to home/username/.mujoco using
unzip mjpro150_linux.zip ~/.mujoco
This will create a folder like home/username/.mujoco/mjpro15

Place the mjkey.txt in the top level folder called .mujoco

Now, run
pip install gym[mujoco,robotics]
This will most likely complain
Exception: 
Missing path to your environment variable. 
Current values LD_LIBRARY_PATH=
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/user/.mujoco/mjpro150/bin

The following is based on:
https://github.com/openai/mujoco-py/issues/267
https://stackoverflow.com/questions/49083789/how-to-add-new-line-in-bashrc-file-in-ubuntu

In order to solve above, open the bashrc file by
nano ~/.bashrc

Copy the following line to the end while replacing {user} with your linux login username.
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/user/.mujoco/mjpro150/bin

Run on the Terminal the following line to update bashrc
source ~/.profile

Now, try to run 
pip install gym[mujoco,robotics]

If you get an error about patchelf, do the following to install patchelf:
conda install patchelf
Taken from https://github.com/openai/mujoco-py/issues/147

Try again
pip install gym[mujoco,robotics]

If the packages are installed successfully without any error, then you are good to proceed.

Run the Mujoco test from Spinning Up by
python -m spinup.run ppo --hid "[32,32]" --env Walker2d-v2 --exp_name mujocotest

If the test runs fine and gets completed, try running the video. If the video does not pop up, and gives a GLEW error, then do the following:

sudo apt-get update -y
sudo apt-get install -y libglew-dev
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so

Now run the Mujoco test again and you should see a Walker.
Above is taken from
https://github.com/openai/mujoco-py/issues/268 
https://www.programmersought.com/article/70476125019/