openai/mujoco-py #268: GLEW initalization error: Missing GL version

LiterallyUniqueLogin, 2018-07-05:
I'm getting this issue when running 
import gym
env = gym.make('Ant-v2')

Mujoco works fine on my machine, I can get the simulations up and running. I've seen [this thread](https://github.com/openai/mujoco-py/issues/44) but none of the suggestions seem to help.

Any debugging suggestions? What .so files should I expect to have? Where should I expect to find them? What packages should I have installed via apt-get? I'm wondering if one of my installations went wrong somehow.

I'm running on Ubuntu 16.04 with an NVIDIA GeForce GTX 980M graphics card and graphics drivers installed, though I'm not sure that's relevant to this problem.

LiterallyUniqueLogin, 2018-07-05:
The solution at the bottom of this [thread](https://github.com/openai/mujoco-py/pull/145) on the openaigym repo seems to work. (The key part being `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so`)

closed 2018-07-05 by LiterallyUniqueLogin

Isaac009, 2020-03-05:
> The solution at the bottom of this [thread](https://github.com/openai/mujoco-py/pull/145) on the openaigym repo seems to work. (The key part being `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so`)

If you still have some issue try this out https://zoomadmin.com/HowToInstall/UbuntuPackage/libglew-dev

and then  (The key part being `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so`)

monacv, 2021-06-03:
`sudo reboot` worked for me in Ubuntu 20.04

HesNobi, 2021-11-14:
The solutions explained here may work for some installations. However, I am using different simulation environments and can't employ the methods mentioned here. I came up with a quick fix by initializing a dummy GL window before rendering! it is patchy but gets the job done.

I added this to the GYM's own observation wrapper `PixelObservationWrapper`:

```
import OpenGL.GLUT as gl
def __init__(....
          # Creating a dummy GL Windows to bupas the Error: "GLEW initalization error: Missing GL version"
          gl.glutInit()
          gl.glutInitWindowSize(10, 10)
          gl.glutCreateWindow('GLEW Testing')
....
```

JamesKCS, 2022-05-06:
This initially fixed the problem for me.  Something changed in the months since then (no idea what), and I started getting the error again.  The solution was to _remove_ the `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so` line from my .bashrc/from my PyCharm env variables.  Hopefully this hint saves someone else the hours it took me to figure this out :)