openai/mujoco-py #269: error while importing mujoco_py into python

avinash5519, 2018-07-10:
Hello, 
i am trying to install mujoco_py following the instructions:
mujoco pro 150
mujoco py version - 1.50

pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
when i am doing " import mujoco_py" it is giving me this error 

running build_ext
building 'mujoco_py.cymj' extension

6_64-3.6/mujoco_py/cymj.cpython-36m-x86_64-linux-gnu.so -fopenmp
**/home/avi/miniconda3/compiler_compat/ld: cannot find -lmujoco150
/home/avi/miniconda3/compiler_compat/ld: cannot find -lglewosmesa**
collect2: error: ld returned 1 exit status

Can anyone help me with this ?

Thanks.

wushichatong, 2018-08-09:
That is same with me, don't know what is the matter.
Systems: Ubuntu 18.04
Python 3.6.5

import mujoco_py
Import error. Trying to rebuild mujoco_py.
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/instellar/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py -I/home/instellar/.mujoco/mjpro150/include -I/home/instellar/anaconda3/lib/python3.6/site-packages/numpy/core/include -I/home/instellar/anaconda3/include/python3.6m -c /home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/cymj.c -o /home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/instellar/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py -I/home/instellar/.mujoco/mjpro150/include -I/home/instellar/anaconda3/lib/python3.6/site-packages/numpy/core/include -I/home/instellar/anaconda3/include/python3.6m -c /home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/gl/osmesashim.c -o /home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/gl/osmesashim.o -fopenmp -w
gcc -pthread -shared -B /home/instellar/anaconda3/compiler_compat -L/home/instellar/anaconda3/lib -Wl,-rpath=/home/instellar/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/cymj.o /home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/gl/osmesashim.o -L/home/instellar/.mujoco/mjpro150/bin -Wl,-R/home/instellar/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -o /home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxcpuextensionbuilder/lib.linux-x86_64-3.6/mujoco_py/cymj.cpython-36m-x86_64-linux-gnu.so -fopenmp
Traceback (most recent call last):
  File "/home/instellar/anaconda3/lib/python3.6/site-packages/IPython/core/interactiveshell.py", line 2963, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-6-85c6ae7fd0aa>", line 1, in <module>
    import mujoco_py
  File "/snap/pycharm-professional/80/helpers/pydev/_pydev_bundle/pydev_import_hook.py", line 20, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/snap/pycharm-professional/80/helpers/pydev/_pydev_bundle/pydev_import_hook.py", line 20, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/builder.py", line 91, in load_cython_ext
    return load_dynamic_ext('cymj', cext_so_path)
  File "/home/instellar/anaconda3/lib/python3.6/site-packages/mujoco_py/builder.py", line 106, in load_dynamic_ext
    return loader.load_module()
ImportError: libglewosmesa.so: cannot open shared object file: No such file or directory

lutein, 2018-09-10:
the same.

ZackPashkin, 2018-09-20:
I'm having the same problem

wweichn, 2018-10-16:
I have done this, created .mujoco file ,but there is still the same problem

avinash5519, 2018-10-16:
what is ur bashrc path for bin directory ? can u post it please

wweichn, 2018-10-16:
I have deleted the added path about mujoco in my bashrc.
So there is nothing about bin path of mujoco in bashrc.

wweichn, 2018-10-16:
the error is 
`import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I.local/lib/python3.6/site-packages/mujoco_py -I/home/ww/.local/lib/python3.6/site-packages/mujoco_py -I/home/ww/.mujoco/mjpro150/include -I/home/ww/.local/lib/python3.6/site-packages/numpy/core/include -I/home/ww/.local/lib/python3.6/site-packages/mujoco_py/vendor/egl -I/usr/include/python3.6m -c /home/ww/.local/lib/python3.6/site-packages/mujoco_py/cymj.c -o /home/ww/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.59_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/ww/.local/lib/python3.6/site-packages/mujoco_py/cymj.o -fopenmp -w
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I.local/lib/python3.6/site-packages/mujoco_py -I/home/ww/.local/lib/python3.6/site-packages/mujoco_py -I/home/ww/.mujoco/mjpro150/include -I/home/ww/.local/lib/python3.6/site-packages/numpy/core/include -I/home/ww/.local/lib/python3.6/site-packages/mujoco_py/vendor/egl -I/usr/include/python3.6m -c /home/ww/.local/lib/python3.6/site-packages/mujoco_py/gl/eglshim.c -o /home/ww/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.59_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/ww/.local/lib/python3.6/site-packages/mujoco_py/gl/eglshim.o -fopenmp -w
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /home/ww/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.59_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/ww/.local/lib/python3.6/site-packages/mujoco_py/cymj.o /home/ww/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.59_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/ww/.local/lib/python3.6/site-packages/mujoco_py/gl/eglshim.o -L/home/ww/.mujoco/mjpro150/bin -Wl,--enable-new-dtags,-R/home/ww/.mujoco/mjpro150/bin -lmujoco150 -lglewegl -o /home/ww/.local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_1.50.1.59_36_linuxgpuextensionbuilder/lib.linux-x86_64-3.6/mujoco_py/cymj.cpython-36m-x86_64-linux-gnu.so -fopenmp
/usr/bin/ld: cannot find -lmujoco150
/usr/bin/ld: cannot find -lglewegl
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/usr/lib/python3.6/distutils/unixccompiler.py", line 207, in link
    self.spawn(linker + ld_args)
  File "/usr/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/usr/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/usr/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ww/.local/lib/python3.6/site-packages/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/ww/.local/lib/python3.6/site-packages/mujoco_py/builder.py", line 476, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/ww/.local/lib/python3.6/site-packages/mujoco_py/builder.py", line 97, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/ww/.local/lib/python3.6/site-packages/mujoco_py/builder.py", line 210, in build
    built_so_file_path = self._build_impl()
  File "/home/ww/.local/lib/python3.6/site-packages/mujoco_py/builder.py", line 282, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/ww/.local/lib/python3.6/site-packages/mujoco_py/builder.py", line 233, in _build_impl
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/ww/.local/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/ww/.local/lib/python3.6/site-packages/mujoco_py/builder.py", line 133, in build_extensions
    build_ext.build_extensions(self)
  File "/home/ww/.local/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 558, in build_extension
    target_lang=language)
  File "/usr/lib/python3.6/distutils/ccompiler.py", line 717, in link_shared_object
    extra_preargs, extra_postargs, build_temp, target_lang)
  File "/usr/lib/python3.6/distutils/unixccompiler.py", line 209, in link
    raise LinkError(msg)
distutils.errors.LinkError: command 'x86_64-linux-gnu-gcc' failed with exit status 1`

avinash5519, 2018-10-16:
please add the path of bin directory of mujoco like this in bashrc
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/shubhom/.mujoco/mjpro150/bin
Adjust this according to your computerpath

wweichn, 2018-10-16:
After I add this in bashrc,I met another problem/hint.
`Please add following line to .bashrc:`
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-384`
And after I add this in bashrc, I met the problem  like in the beginning.

machinaut, 2018-10-17:
@avinash5519 do you have the mujoco and osmesa libraries installed?

It looks like the compiler is failing to find the libraries it needs to link against.

alexmaraval, 2020-05-16:
I had the same problem as @wushichatong when I was running `import mujoco_py` in the python console of PyCharm.

It worked in the terminal and I found the only difference between `os.environ` from the terminal and from my PyCharm console was that `LD_LIBRARY_PATH` was not inherited form the `~/.profile` or `~/.bashrc` which caused the compiler to fail looking for the libraries in the right places as mentioned by @machinaut .

I solved this issue at least for PyCharm by adding `LD_LIBRARY_PATH=:/home/path/to/.mujoco/mujoco200/bin` to the environment variables in the Python Console settings (see [here](https://www.jetbrains.com/help/pycharm/console-python-console.html)).

TachikakaMin, 2020-06-04:
cp -r ~/.mujoco/mujoco200/bin/* /usr/lib/

Su-Lemon, 2020-09-03:
I found a solution to other similar problems. When using the new version of Mujoco, its folder name should be like "mujoco200"; but when using the previous version, the folder name should be like "mjpro131"