openai/mujoco-py #270: when mujoco131 changed to mujoco150,gym.make("envname") break down,command 'gcc' failed with exit status 1

ZEUP330, 2018-07-20:
I have a problem. Before that, I used ubuntu16.04, python3.5.5, gym=10.0.1 mujoco_py=0.5.7, mujoco150, my simulation is normal, and the make of gym is working normally.
Now I want fetch in gym to train my robot. According to the requirements, I need to change to the new mujoco150, using mujoco_py=1.50.1.56, but the following error occurs in gym.make("envname"):
`
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/ljt/anaconda3/envs/py35/compiler_compat -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py -I/home/ljt/.mujoco/mjpro150/include -I/home/ljt/.local/lib/python3.5/site-packages/numpy/core/include -I/home/ljt/anaconda3/envs/py35/include/python3.5m -c /home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/cymj.c -o /home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/generated/_pyxbld_1.50.1.56_35_linuxcpuextensionbuilder/temp.linux-x86_64-3.5/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/ljt/anaconda3/envs/py35/compiler_compat -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py -I/home/ljt/.mujoco/mjpro150/include -I/home/ljt/.local/lib/python3.5/site-packages/numpy/core/include -I/home/ljt/anaconda3/envs/py35/include/python3.5m -c /home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/gl/osmesashim.c -o /home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/generated/_pyxbld_1.50.1.56_35_linuxcpuextensionbuilder/temp.linux-x86_64-3.5/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/gl/osmesashim.o -fopenmp -w
/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/gl/osmesashim.c:1:23: fatal error: GL/osmesa.h: No such file or directory
compilation terminated.
Traceback (most recent call last):
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/unixccompiler.py", line 118, in _compile
extra_postargs)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
spawn(cmd, dry_run=self.dry_run)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/spawn.py", line 36, in spawn
_spawn_posix(cmd, search_path, dry_run=dry_run)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
% (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/ljt/try_test/gym_test.py", line 3, in 
env = gym.make('Ant-v2') #创造环境
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/registration.py", line 167, in make
return registry.make(id)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/registration.py", line 119, in make
env = spec.make()
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/registration.py", line 85, in make
cls = load(self._entry_point)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/registration.py", line 14, in load
result = entry_point.load(False)
File "/home/ljt/.local/lib/python3.5/site-packages/pkg_resources/init.py", line 2322, in load
return self.resolve()
File "/home/ljt/.local/lib/python3.5/site-packages/pkg_resources/init.py", line 2328, in resolve
module = import(self.module_name, fromlist=['name'], level=0)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/mujoco/init.py", line 1,in 
from gym.envs.mujoco.mujoco_env import MujocoEnv
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/gym/envs/mujoco/mujoco_env.py", line 11, in 
import mujoco_py
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/init.py", line 1, in 
from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/builder.py", line 468, in 
cymj = load_cython_ext(mjpro_path)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/builder.py", line 90, in load_cython_ext
cext_so_path = builder.build()
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/builder.py", line 202, in build
built_so_file_path = self._build_impl()
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/builder.py", line 256, in _build_impl
so_file_path = super()._build_impl()
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/builder.py", line 225, in _build_impl
dist.run_commands()
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/dist.py", line 955, in run_commands
self.run_command(cmd)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/dist.py", line 974, in run_command
cmd_obj.run()
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
_build_ext.build_ext.run(self)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/command/build_ext.py", line 339, in run
self.build_extensions()
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/mujoco_py/builder.py", line 125, in build_extensions
build_ext.build_extensions(self)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
self.build_extension(ext)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/command/build_ext.py", line 533, in build_extension
depends=ext.depends)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/ccompiler.py", line 574, in compile
self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
File "/home/ljt/anaconda3/envs/py35/lib/python3.5/distutils/unixccompiler.py", line 120, in _compile
raise CompileError(msg)
distutils.errors.CompileError: command 'gcc' failed with exit status 1`

zuoxingdong, 2018-07-23:
Same error here, Ubuntu 16.04, with default gcc

ZEUP330, 2018-07-24:
You mean that GCC that comes with Ubuntu16.04 cannot run mujoco-py.@zuoxingdong

zuoxingdong, 2018-07-26:
@ZEUP330 The problem get resolved by upgrading GCC to `gcc-7`. Not sure but it might be okay for Ubuntu 18.04 where gcc is newer version by default.

wushichatong, 2018-08-09:
@zuoxingdong Same errors for Ubuntu 18.04 with 7.3.0 GCC

iSarmad, 2018-08-19:
also try this, as it worked for me
sudo apt-get install libglew-dev

GraceYYJ, 2018-10-08:
Same error here, Ubuntu 16.04, with default gcc

machinaut, 2018-10-17:
The critical error is here
```
error: GL/osmesa.h: No such file or directory
```

Do you have the mesa library package installed?  If you have it installed then the compiler is failing to find the headers.

jp18813100494, 2018-12-02:
> @ZEUP330 The problem get resolved by upgrading GCC to `gcc-7`. Not sure but it might be okay for Ubuntu 18.04 where gcc is newer version by default.

how to update GCC to 'gcc-7'?thanks, i met the same problem!

jangirrishabh, 2019-05-06:
Hello @machinaut my 18.04 machine has the mesa lib package installed but compiler doesn't find it.
There's a file in "/usr/include/GL/osmesa.h" and also I have tried adding
"export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGL.so" but it still doesn't find the package.
Any help would be appreciated ! Thanks