openai/mujoco-py #271: How to get camera matrix in Mujoco

shinshiner, 2018-07-24:
Hi,
I wanted to convert the depth image to 3D point cloud in mujoco. I was trying to find camera intrinsic matrix for the camera body that I attached to the robot, but I couldn't find any information regarding this. Is there any way to attach this information ?

shinshiner, 2018-07-25:
Maybe there is no camera matrix in Mujoco ? If so, how to implement such conversion?

InverseNova, 2018-09-25:
I haven't found camera matrix in mujoco, however, I could construct it.
I assumed that the principal point is not shifted and pixels are of square form. You need to know frame size (width, height) and focus distance (in pixels), which can be computed as following:
`fovy = sim.model.cam_fovy[cam_id]`
`f = 0.5 * height / math.tan(fovy * math.pi / 360)`
Then you get camera matrix as `np.array(((f, 0, width / 2), (0, f, height / 2), (0, 0, 1)))`
It worked for me.
For more info about camera matrix look [here](https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html), for more detailed understanding you may have a look on [this book](http://cvrs.whu.edu.cn/downloads/ebooks/Multiple%20View%20Geometry%20in%20Computer%20Vision%20(Second%20Edition).pdf)

tkelestemur, 2019-11-03:
@beingkartik According to the MuJoCo API [documentation](http://www.mujoco.org/book/XMLreference.html#camera), the horizontal field of view (fovx) is calculated based on the horizontal field of view and the windows size (width, height). If you use, a square image they be equal.

kracon7, 2020-03-27:
Hi @shinshiner  and @InverseNova 
I'm also working on a 3d reconstrction project in MuJoCo. But I had a problem about how to extract the real depth value from the depth image rendered by sim.render(). So basically, the depth image we get from the simulation is normalized to [0,1]. Could you guys let me know how to convert it back to the real depth value?
I also submitted an issue #520 with description of my problem in detail. 

Thanks in advance!

kracon7, 2020-03-27:
[Anwser](https://github.com/openai/mujoco-py/issues/520#issuecomment-605335819) to the depth conversion problem.

ehu-ai, 2020-12-24:
I tried using this camera matrix for projecting 3d coordinates into 2d pixel coordinates. But I found the horizontal pixel dimension to be flipped. The problem seems to be with the camera matrix, where the first element is **-f**, not **f**. Indeed, if you check deepmind's camera matrix, they also negate the first **f** in the diagonal.

https://github.com/deepmind/dm_control/blob/87e046bfeab1d6c1ffb40f9ee2a7459a38778c74/dm_control/mujoco/engine.py#L717