openai/mujoco-py #272: Segmentation fault when creating window glfw

tianheyu927, 2018-07-26:
When I run `python -c "import gym; env = gym.make('Humanoid-v2'); env.render()"` with mujoco 1.50, it shows the following error:
```
Creating window glfw
Segmentation fault (core dumped)
```

I use gdb to debug this and get the following messages:
```
Creating window glfw

Program received signal SIGSEGV, Segmentation fault.
0x00007fffefd4228e in glGetString () from /usr/lib/x86_64-linux-gnu/libOSMesa.so.6
```

However, it does works well when I try ./simulate in the ./mujoco/mjpro150/bin
Does anyone know how to fix this issue? I have Ubuntu 14.04, python 3.5.2 and have updated the latest OpenGL library.

closed 2018-07-26 by tianheyu927

yxnyxnyxnyxn, 2018-09-12:
i have the same problem, have you solved that?

tianheyu927, 2018-09-12:
This [thread](https://github.com/openai/mujoco-py/pull/145) could be helpful.