openai/mujoco-py #278: Mujoco-py on Windows 64 bits

jesdefer, 2018-08-22:
I have fully followed all the steps indicated in your github. All is properly installed, but when I run the example _body_interaction.py_ I get the following error:
`cl.exe failed with exit status 2`
Why does this error appear?

jesdefer, 2018-08-22:
A more detailed view of the command window output can be found here:

```
C:\Users\USUARIO\Dropbox\TFM\Mujoco\mujoco-py>python examples\body_interaction.py
Traceback (most recent call last):
  File "examples\body_interaction.py", line 10, in <module>
    from mujoco_py import load_model_from_xml, MjSim, MjViewer
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, Mujoc
oException
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 78, in load_cython_ext
    "set %s=%s;%%%s%%" % (var, lib_path, var))
Exception: Please add mujoco library to your PATH:
set PATH=C:\Users\USUARIO\.mujoco\mjpro150\bin;%PATH%
```

```
C:\Users\USUARIO\Dropbox\TFM\Mujoco\mujoco-py>SET PATH=%userprofile%\.mujoco\mjp
ro150\bin;%PATH%;
```

```
C:\Users\USUARIO\Dropbox\TFM\Mujoco\mujoco-py>python examples\body_interaction.p
y
running build_ext
building 'mujoco_py.cymj' extension
C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.
15.26726\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Python\
Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mujoco_py -IC:\Users
\USUARIO\.mujoco\mjpro150\include -IC:\Python\Python36-32\lib\site-packages\nump
y\core\include -IC:\Python\Python36-32\include -IC:\Python\Python36-32\include "
-IC:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\1
4.15.26726\include" /TcC:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.
56-py3.6.egg\mujoco_py\cymj.c /FoC:\Python\Python36-32\lib\site-packages\mujoco_
py-1.50.1.56-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.56_36_windowsextension
builder\temp.win32-3.6\Release\Python\Python36-32\lib\site-packages\mujoco_py-1.
50.1.56-py3.6.egg\mujoco_py\cymj.obj -fopenmp -w
cl : Línea de comandos warning D9025 : invalidando '/W3' con '/w'
cl : Línea de comandos warning D9002 : se omite la opción desconocida '-fopenmp'

cymj.c
c:\python\python36-32\include\pyconfig.h(59): fatal error C1083: No se puede abr
ir el archivo incluir: 'io.h': No such file or directory
Traceback (most recent call last):
  File "C:\Python\Python36-32\lib\distutils\_msvccompiler.py", line 423, in comp
ile
    self.spawn(args)
  File "C:\Python\Python36-32\lib\distutils\_msvccompiler.py", line 542, in spaw
n
    return super().spawn(cmd)
  File "C:\Python\Python36-32\lib\distutils\ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "C:\Python\Python36-32\lib\distutils\spawn.py", line 38, in spawn
    _spawn_nt(cmd, search_path, dry_run=dry_run)
  File "C:\Python\Python36-32\lib\distutils\spawn.py", line 81, in _spawn_nt
    "command %r failed with exit status %d" % (cmd, rc))
distutils.errors.DistutilsExecError: command 'C:\\Program Files (x86)\\Microsoft
 Visual Studio\\2017\\BuildTools\\VC\\Tools\\MSVC\\14.15.26726\\bin\\HostX86\\x8
6\\cl.exe' failed with exit status 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples\body_interaction.py", line 10, in <module>
    from mujoco_py import load_model_from_xml, MjSim, MjViewer
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, Mujoc
oException
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 90, in load_cython_ext
    cext_so_path = builder.build()
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 202, in build
    built_so_file_path = self._build_impl()
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 225, in _build_impl
    dist.run_commands()
  File "C:\Python\Python36-32\lib\distutils\dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "C:\Python\Python36-32\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "C:\Python\Python36-32\lib\site-packages\Cython\Distutils\old_build_ext.p
y", line 186, in run
    _build_ext.build_ext.run(self)
  File "C:\Python\Python36-32\lib\distutils\command\build_ext.py", line 339, in
run
    self.build_extensions()
  File "C:\Python\Python36-32\lib\site-packages\mujoco_py-1.50.1.56-py3.6.egg\mu
joco_py\builder.py", line 125, in build_extensions
    build_ext.build_extensions(self)
  File "C:\Python\Python36-32\lib\site-packages\Cython\Distutils\old_build_ext.p
y", line 194, in build_extensions
    self.build_extension(ext)
  File "C:\Python\Python36-32\lib\distutils\command\build_ext.py", line 533, in
build_extension
    depends=ext.depends)
  File "C:\Python\Python36-32\lib\distutils\_msvccompiler.py", line 425, in comp
ile
    raise CompileError(msg)
distutils.errors.CompileError: command 'C:\\Program Files (x86)\\Microsoft Visua
l Studio\\2017\\BuildTools\\VC\\Tools\\MSVC\\14.15.26726\\bin\\HostX86\\x86\\cl.
exe' failed with exit status 2
```

machinaut, 2018-10-17:
It looks like the package is failing to compile the extension.

I think the relevant error line is here:
```
c:\python\python36-32\include\pyconfig.h(59): fatal error C1083: No se puede abr
ir el archivo incluir: 'io.h': No such file or directory
```

I've never seen this before -- `io.h` is not a header file we need, but maybe it is needed by the lower level python files (either python compiled extensions, or cython).

Do you know where this header is supposed to come from?

ChenZiYiLOVE, 2019-10-18:
have someone figure it out?