openai/mujoco-py #279: How to capture specific figures from Mujoco

JackieTseng, 2018-08-25:
Hey, I want to take some screenshot in mujoco_env under condition of:
* removing the grid in mujoco env
* assigning a specific and nice pose in the simulator, if possible

An example like this:
[image]
Any ideas? THX.

BrunoBSM, 2018-09-01:
When you say "mujoco env" do you mean the gym env or your own simulation? like a MjSim()

machinaut, 2018-10-17:
We have a couple small examples in the repository here, a few of which do render (so you can see how those files render images)

This repository is just python wrapper to the Mujoco library, if you want to understand how the mujoco library works then I would recommend looking at the documentation here http://www.mujoco.org/book/index.html

The pose of everything in the simulator is controlled by the `qpos` variable -- if you set that, then you need to make sure its forward consistent, so following that by `mj_forward` will make sure variables are correct.  The `render` function is possibly what you want in order to render an image of the scene.

closed 2018-10-17 by machinaut