openai/mujoco-py #280: Error during 'import mujoco_py': ‘for’ loop initial declarations are only allowed in C99 mode

Rainer-K, 2018-08-27:
I installed mujoco_py via pip as written in the readme. Installation completed without an error. However, when trying to 'import mujoco_py', I encountered the following error: (I am running Python 3.6.4 with anaconda (located in [home]/python/anaconda3). I replaced my home directory with [home] to make paths a bit shorter.)

    Python 3.6.4 |Anaconda, Inc.| (default, Jan 16 2018, 18:10:19) 
    [GCC 7.2.0] on linux
    Type "help", "copyright", "credits" or "license" for more information.

    >>> import mujoco_py
    running build_ext
    building 'mujoco_py.cymj' extension
    gcc -pthread -B [home]/python/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py -I[home]/.mujoco/mjpro150/include -I[home]/python/anaconda3/lib/python3.6/site-packages/numpy/core/include -I[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/vendor/egl -I[home]/python/anaconda3/include/python3.6m -c [home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/cymj.c -o [home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/cymj.o -fopenmp -w
    gcc -pthread -B [home]/python/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py -I[home]/.mujoco/mjpro150/include -I[home]/python/anaconda3/lib/python3.6/site-packages/numpy/core/include -I[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/vendor/egl -I[home]/python/anaconda3/include/python3.6m -c [home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/gl/eglshim.c -o [home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/gl/eglshim.o -fopenmp -w
    [home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/gl/eglshim.c: In function ‘closeOpenGL’:
    [home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/gl/eglshim.c:159:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
         for (int device_id=0; device_id<MAX_DEVICES; device_id++) {
         ^
    [home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/gl/eglshim.c:159:5: note: use option -std=c99 or -std=gnu99 to compile your code
    Traceback (most recent call last):
      File "[home]/python/anaconda3/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
        extra_postargs)
      File "[home]/python/anaconda3/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
        spawn(cmd, dry_run=self.dry_run)
      File "[home]/python/anaconda3/lib/python3.6/distutils/spawn.py", line 36, in spawn
        _spawn_posix(cmd, search_path, dry_run=dry_run)
      File "[home]/python/anaconda3/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
        % (cmd, exit_status))
    distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/__init__.py", line 1, in <module>
        from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
      File "[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/builder.py", line 468, in <module>
        cymj = load_cython_ext(mjpro_path)
      File "[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/builder.py", line 90, in load_cython_ext
        cext_so_path = builder.build()
      File "[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/builder.py", line 202, in build
        built_so_file_path = self._build_impl()
      File "[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/builder.py", line 274, in _build_impl
        so_file_path = super()._build_impl()
      File "[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/builder.py", line 225, in _build_impl
        dist.run_commands()
      File "[home]/python/anaconda3/lib/python3.6/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "[home]/python/anaconda3/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "[home]/python/anaconda3/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
        _build_ext.build_ext.run(self)
      File "[home]/python/anaconda3/lib/python3.6/distutils/command/build_ext.py", line 339, in run
        self.build_extensions()
      File "[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/builder.py", line 125, in build_extensions
        build_ext.build_extensions(self)
      File "[home]/python/anaconda3/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
        self.build_extension(ext)
      File "[home]/python/anaconda3/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
        depends=ext.depends)
      File "[home]/python/anaconda3/lib/python3.6/distutils/ccompiler.py", line 574, in compile
        self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
      File "[home]/python/anaconda3/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
        raise CompileError(msg)
    distutils.errors.CompileError: command 'gcc' failed with exit status 1

I went into the mentioned file `[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/mujoco_py/gl/eglshim.c` and changed the for loop signature in line 159 from

    for (int device_id=0; device_id<MAX_DEVICES; device_id++) {

to 

    int device_id = 0;
    for (device_id=0; device_id<MAX_DEVICES; device_id++) {

Now I am able to import mujoco_py. I would suggest setting the c99 flag or changing the loop signature.

On a side note, I tried the example from the readme file, but the file `/xmls/claw.xml` does not exist. (It is looking in `[home]/python/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.56-py3.6.egg/`, but there is no directory `/xmls`.) But this is probably another issue.

Rainer-K, 2018-08-27:
I uninstalled the pip version and installed from source:

    pip uninstall mujoco_py
    git clone https://github.com/openai/mujoco-py.git
    cd mujoco_py
    python setup.py install

When importing the, same error (for loop signature) occured, and it was again fixed by changing the respective file (`mujoco-py/mujoco_py/gl/eglshim.c`). But now the `/xmls/claw.xml` exists and I can run the example.

Rainer-K, 2018-08-27:
Would be fixed by #252

machinaut, 2018-10-18:
@Rainer-K thanks, separately these issues are fixed on master now and should be in the next release!

closed 2018-10-18 by machinaut